earlephilhower/arduino-pico #2412: Regression: incorrect numbering for GPIO6 on Adafruit Feather RP2040

sillycowvalley, 2024-09-04:
I believe there was a mistake made in the Adafruit Feather pins_arduino.h in this commit which causes code using GPIO6 / D6 to now fail:
https://github.com/earlephilhower/arduino-pico/commit/a584d10321e28ecdae0396701db4a66efb61c290#diff-946482728f153bf0ff3529fb3d122116240ce5d7a64753afd7ee6916260cebbbR47

I've tested for each GPIO from the Adafruit Pinout PDF and I believe:
- __PIN_D4 line should be removed (there is no GPIO4 / D4)
- __PIN_D6 value should be 6u (not 8u)

earlephilhower, 2024-09-04:
Err, looking at https://cdn-learn.adafruit.com/assets/assets/000/107/203/original/adafruit_products_feather-rp2040-pins.png?1639162603 D6 == GPIO8 and D4 == GPIO6

sillycowvalley, 2024-09-04:
Apologies. I stand corrected.

I was relying on the original 1:1 relationship between the Dx and the physical pin numbers to deduce the GPx, rather than the labels on the sheet. This fix corrected that issue which broke my code generator (by convention, I use the GPx in Hopper, not the Dx).

closed 2024-09-04 by sillycowvalley