earlephilhower/arduino-pico #2414: SerialUSB Issue when Rebooting with rp2040.reboot()

peternlewis, 2024-09-04:
There seems to be an issue with SerialUSB when using rp2040.reboot().

After the reboot, there are issues with the resulting output in Arduino. 

I am using Arduino IE 2.3.2 running on macOS Sonoma 14.6.1.

I can easily duplicate the issue with the following relatively trivial code.

It may have nothing to do with SerialUSB and may be simply that there are issues with cleanly rebooting with `rp2040.reboot()`.

I've checked and SerialUSB.h/cpp appears to be identical code as on github. I'm happy to do any further debugging if this is not easily reproducible for others.

Thanks.

```
uint32_t gLastTime;
uint32_t gCount = 1;

void setup() 
{
  Serial.begin(115200);
  while (!Serial) ;
  Serial.println( "Starting" );
  gLastTime = millis();
}

void loop()
{
  if ( millis() - gLastTime > 1000 ) {
    Serial.print( "Count: " );
    Serial.println( gCount );
    gCount++;
    gLastTime = millis();
    if ( gCount == 15 ) {
      delay( 1000 );
      rp2040.reboot();
    }
  }
}

```

```
18:08:56.576 -> Count: 1
18:08:57.562 -> Count: 2
18:08:58.552 -> Count: 3
18:08:59.571 -> Count: 4
18:09:00.556 -> Count: 5
18:09:01.575 -> Count: 6
18:09:02.562 -> Count: 7
18:09:03.580 -> Count: 8
18:09:04.569 -> Count: 9
18:09:05.557 -> Count: 10
18:09:06.575 -> Count: 11
18:09:07.590 -> Count: 12
18:09:08.576 -> Count: 13
18:09:09.564 -> Count: 14
18:09:11.575 -> Starting
18:09:13.548 -> Count: 2
18:09:15.550 -> Count: 4
18:09:17.555 -> Count: 6Count: 7
18:09:19.560 -> Count: 8
18:09:20.547 -> 
18:09:21.533 -> Count: 10
18:09:23.534 -> Count: 12
18:09:24.555 -> Count: 13Count: 14
18:09:28.376 -> Count: 1Count: 2
18:09:32.382 -> Count: 5
18:09:35.406 -> Count: 8
18:09:36.396 -> Count: 9
18:09:38.400 -> Count: 11
18:09:40.405 -> Count: 13
18:09:44.289 -> Count: 1
18:09:47.285 -> Count: 4
18:09:48.273 -> Count: 5
18:09:50.284 -> Count: 7
18:09:51.304 -> Count: 8
18:09:53.277 -> Count: 10
18:09:55.289 -> Count: 12
18:09:59.176 -> Starting
18:10:00.127 -> Count: 1Count: 2Count: 3
18:10:03.120 -> 
18:10:05.123 -> Count: 6Count: 7Count: 8
18:10:09.132 -> Count: 10
18:10:10.119 -> 
18:10:11.134 -> Count: 12
18:10:12.120 -> 
18:10:13.140 -> 
18:10:16.000 -> 
18:10:18.007 -> Count: Count: 4Count: 5
18:10:22.020 -> Count: 7
18:10:24.030 -> 
18:10:25.016 -> Count: 10Count: 12
18:10:28.008 -> Count: 13
18:10:33.864 -> Count: 3
18:10:34.882 -> Count: 4
18:10:36.887 -> Count: 6Count: 7
18:10:39.874 -> Count: 9
18:10:41.879 -> Count: 11
18:10:43.882 -> Count: 13
18:10:48.750 -> Count: Count: 4Count: 5Count: 6
18:10:53.754 -> 
18:10:54.742 -> 
18:10:56.748 -> Count: 10
18:10:58.755 -> 
```

earlephilhower, 2024-09-04:
USB is run by the Pico itself, not a separate chip like on the old Unos.  A reset is like an unexpected unplug and replug sequence and the OS and apps can miss the first few transfers while things are being rejiggered on the host.  This is normal and expected and not an issue with the Pico or the host

closed 2024-09-04 by earlephilhower

peternlewis, 2024-09-05:
It isn't “the first few transfers”. The situation continues indefinitely with missed and reordered data. I adjusted the `15` in the code I posted to `200`, so that is 200 seconds between rebooted, 200 lines of `Count: n` sent, and even after all that it has not regained consistency.

Here is a sample 2.5 minutes and 150 lines after the first reboot:

```
09:46:13.134 -> Count: 149
09:46:14.119 -> Count: 150
09:46:16.132 -> Count: 152
09:46:17.152 -> Count: Count: 154
09:46:20.140 -> Count: 156
09:46:21.145 -> 157Count: 158Count: 159
09:46:24.141 -> 
09:46:25.163 -> Count: 161Count: 
09:46:28.155 -> Count: 164Count: 165Count: 166
09:46:31.153 -> 
09:46:32.138 -> Count: 168Count: 169Count: 170Count: Count: 172
```

Further reboots or sending of data do not resolve the behaviour.

Are you saying it is to be expected that after the rp2040.reboot(), the USB serial will never again reliably send data?

earlephilhower, 2024-09-05:
On my system after this kind of random reboot it doesn't even come up as the same device due to the fact it's unplugging unexpectedly (in the middle of a transfer/ping from the host, possibly).  I get it alternating between `/dev/ttyACM1` and `/dev/ttyACM2`.

This is a function of the OS and the serial terminal app.  It has nothing to do with the core or the Pico, it's just like  randomly unplugged and then replugged a device over and over.  On Linux it runs perfectly fine and as soon as I pop to the newly created TTY I get a nice count from 1...14 at which point it resets and I need to swap emulated TTY.  On other OSes or other terminals, YMMV.

I suggest you shut down whatever you're running to view the USB CDC port and restart it after a reboot if this is really troublesome.