earlephilhower/arduino-pico #2416: PIN_LED not defined in Board definition causes compilation to fail

savejeff, 2024-09-05:
I noticed that with the latest version my compilation failed. 

the difference is that in `framework-arduinopico/variants/generic/common.h` the line

```

#ifdef PIN_LED
#define LED_BUILTIN PIN_LED
#endif
```

changed to 

`
#define LED_BUILTIN PIN_LED
`
causing all board to fail that do not define/have PIN_LED 
an example is 
`board = waveshare_rp2040_lcd_1_28`

I use PlatformIO with this Build Config

```

[env:VALIDATE_RP2040_BASE]
framework = arduino
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
board = waveshare_rp2040_lcd_1_28
board_build.core = earlephilhower
board_build.filesystem_size = 0.5m
lib_archive = true

```

savejeff, 2024-09-05:
okay it seems i got an older version 1.30100.0 on a clean install on a different laptop. my older laptop has 1.30800.0. 
any idea what this happens? i have never used the rp2040 on my other laptop and just today installed the core

closed 2024-09-05 by earlephilhower

savejeff, 2024-09-06:
Yeah i noticed i have an old version. 
i think there was already an old version on that other system. 

What is the recommended way to update to the latest version? i would like to avoid just deleting the old in .platformio and redownload. Do you by chance know how to trigger an update/upgrade?

savejeff, 2024-09-06:
Okay i managed to update it via `pio pkg update`

earlephilhower, 2024-09-06:
Please check out https://github.com/earlephilhower/arduino-pico/blob/master/docs/platformio.rst .  I'm not sure a PIO command will get the latest versions.