earlephilhower/arduino-pico #2419: SoftwareSerial does not work anymore in 4.0.1

jaszczurtd, 2024-09-06:
As mentioned in the description, my code works without any issues up to version 3.9.5.

I also tested whether adding "true" to the object constructor would make a difference, but unfortunately, it didn’t.

I noticed some upcoming changes related to RX/TX inversion (commit ID: 226a3188975b21d29d32fe4423a6c929766192a5). Perhaps you could take another look at the issue with this change in mind?

Please let me know if you need any further information.

[...]

#define SERIAL_RX_GPIO 22
#define SERIAL_TX_GPIO 21

static SoftwareSerial gpsSerial(SERIAL_RX_GPIO, SERIAL_TX_GPIO); <-- inserting "true" here does not change anything for 4.0.1
static TinyGPSPlus gps;

void serialTalks(void);

static bool isGPSInitialized = false;
void initGPS(void) {
  if(!isGPSInitialized) {
    attachInterrupt(SERIAL_RX_GPIO, serialTalks, FALLING);  
    gpsSerial.begin(9600, SERIAL_7N1);
    isGPSInitialized = true;
  }
}

void serialTalks(void) {
  if(gpsSerial.available() > 0) {
    gps.encode(gpsSerial.read());
  }
}

earlephilhower, 2024-09-06:
Looks like a typo.  If you're using the `git` version, can you edit
https://github.com/earlephilhower/arduino-pico/blob/226a3188975b21d29d32fe4423a6c929766192a5/cores/rp2040/SoftwareSerial.h#L40-L41 to the following and try (note the underscore in the name):
````
 setInvertTX(_invert); 
 setInvertRX(_invert); 
````

earlephilhower, 2024-09-06:
Actually, it was a little more complicated.  Please try #2423 and report back.  The TX inversion got overwritten by the PIO TX setup before.  I've verified normal and inverted on a logic analyzer just now with this new patch.

closed 2024-09-07 by earlephilhower

gandy92, 2024-12-06:
Maybe off-topic, but I'm using setInvertTX() in code that also need to compile with the vanilla rp2040 core. Is there any macro that allows to know if setInvertTX() is available?

earlephilhower, 2024-12-06:
You can check the core of this or official using https://github.com/earlephilhower/arduino-pico/blob/master/docs/contrib.rst#compiler-defines-for-porting

PR #2423 (2024-09-07): Fix SoftwareSerial/SerialPIO inversion and build
SWSerial wasn't even building due to a typo in the header, and SerialPIO needs to set the OE-invert flag after PIO initialization for transmit.

Fixes #2419
commit be2174bd (2024-09-07): Fix SoftwareSerial/SerialPIO inversion and build (#2423)

SWSerial wasn't even building due to a typo in the header, and SerialPIO
needs to set the OE-invert flag after PIO initialization for transmit.

Fixes #2419