earlephilhower/arduino-pico #2429: Pimoroni PGA2350 and PSRAMTest.ino

NollKollTroll, 2024-09-08:
Using version 4.0.1 of the board package. I have a Pimoroni PGA2350, which lacking a specific profile, use the Generic profile. Works very well for many tested things, but not the following.

Running the RP2350/PSRAMTest.ino, setting the menu option "PSRAM Size" to 8MByte and adding the following to the top of the code:
`#define PSRAM_CS 47`

The result is a compile that fails:
```
/tmp/.arduinoIDE-unsaved202488-134564-gi9kl9.1fdyb/PSRAMTest/PSRAMTest.ino: In function 'void loop()':
/tmp/.arduinoIDE-unsaved202488-134564-gi9kl9.1fdyb/PSRAMTest/PSRAMTest.ino:66:30: error: 'pmalloc' was not declared in this scope; did you mean 'pvalloc'?
   66 |   uint8_t *pmem = (uint8_t *)pmalloc(PMALLOCSIZE);
      |                              ^~~~~~~
      |                              pvalloc
```

Not defining PSRAM_CS predictably compiles OK, and does not test any PSRAM.

PontusO, 2024-09-08:
This flag needs to be defined in the pins_arduino.h file for this to work.

NollKollTroll, 2024-09-08:
Thanks PontusO, now I know where to edit this.
Just tried to directly add this in the `generic_rp2350/pins_arduino.h`, which makes the `PSRAMTest.ino` example work perfectly :dancers: 
Now I will try my own code, to see the impact compared to internal RAM.
I'll try creating a new board later, hopefully good enough for a PR.

I suggest editing the docs regarding PSRAM and Generic RP2350 that it does not work, even though the menu option suggests otherwise. Or maybe remove the menu option for boards lacking a proper profile, which is beyond my knowledge.

earlephilhower, 2024-09-08:
Oh no, the generic runs fine with PSRAM...but the core doesn't know anything about pins >31 so you couldn't set it for the B variant.  Just tested w/the iLabs board.

A user was generous enough to send me one of those PGA-like boards with the B chip but I haven't had a chance to actually wire it up since I need to break out a USB breakout board, some mini switrches for BOOTSEL and RUN, and my wire-wrap tool to actually communicate with it.  Kind of a pain to prototype but I see how for production it makes things easier...

earlephilhower, 2024-09-08:
This is being tracked in #2381 so closing here...

closed 2024-09-08 by earlephilhower