earlephilhower/arduino-pico #2430: LwIP Integration questioned

HamzaHajeir, 2024-09-08:
Hi 

I've asked a [question on Discussions](https://github.com/earlephilhower/arduino-pico/discussions/2339) while ago about an issue regarding arduino-pico integration which ended up with no conclusion. 

However, I've managed to reproduce the issue with raw LwIP APIs, exterminating any other dependency to create an MCVE to replicate the issue.

### MCVE: 

```cpp
#include <string>

#include <lwip/tcp.h>

#include <Arduino.h>
#include <WiFi.h>

#define BURST_SIZE 20
#define MSG_SIZE TCP_SND_BUF / BURST_SIZE
#define YOUR_SSID "XXXXXXXX"
#define YOUR_PWD "XXXXXXXX"

struct tcp_pcb* pcb;

IPAddress ip(192,168,100,4);
#define REMOTE_PORT 	8007
bool cnx=false;
bool issueSend = false;
uint32_t freeHeap() { return rp2040.getFreeHeap(); }
uint32_t maxBlock() { return freeHeap()/2; }
template<int I,typename... Args>
void _PRINTF(const char* fmt, Args... args) {
	Serial.printf(std::string(std::string("H4AT:% 7lu: H=%u M=%u ")+fmt).c_str(),millis(),freeHeap(),maxBlock(),args...);
}

#define PRINTF(...) _PRINTF<0>(__VA_ARGS__)

struct Data {
	uint8_t* data;
	uint16_t len;
};
std::queue<Data> rxQueue;

uint8_t msg[MSG_SIZE];

void popQueue(std::queue<Data>& Q) {
  if (Q.size()) {
	auto &d = Q.front();
	free(d.data);
	Q.pop();
  }
}
void _tcp_error(void* arg, err_t err) {
	PRINTF("err=%d\n", err);
	cnx=false;
}

err_t _tcp_recv(void *arg, struct tcp_pcb *tpcb, struct pbuf *p, err_t err) {
	PRINTF("_tcp_recv %p %p %d\n", tpcb, p, err);
	if (p){ 
		PRINTF("data %p tot_len %d\n", p->payload, p->tot_len);

		auto cpydata=static_cast<uint8_t*>(malloc(p->tot_len));
		if (cpydata) {
			pbuf_copy_partial(p,cpydata,p->tot_len,0);
			rxQueue.push({cpydata,p->tot_len});
		}

		tcp_recved(tpcb, p->tot_len);
		pbuf_free(p);

[... 436 lines cut ...]
0x1001b6ae: c0 46           	nop			@ (mov r8, r8)
0x1001b6b0: 0c 10           	asrs	r4, r1, #32
0x1001b6b2: 00 20           	movs	r0, #0
0x1001b6b4: 08 10           	asrs	r0, r1, #32
0x1001b6b6: 00 20           	movs	r0, #0
0x1001b6b8: f4 18           	adds	r4, r6, r3
0x1001b6ba: 00 20           	movs	r0, #0
0x1001b6bc: 14 10           	asrs	r4, r2, #32
0x1001b6be: 00 20           	movs	r0, #0
0x1001b6c0: 54 05           	lsls	r4, r2, #21
0x1001b6c2: 00 00           	movs	r0, r0
```
</details>

#### Analysis:

The issue presents when a memory is being allocated within `_tcp_recv` callback by `malloc` and freed on main loop, even without `pbuf_copy_partial()`.

#### Hypothesis:

The arduino integration [wraps](https://github.com/earlephilhower/arduino-pico/blob/master/cores/rp2040/lwip_wrap.cpp) the raw LwIP APIs to lock the async_context that runs lwip context. Where all of these are of calling from app down to LwIP context. 

The wrapping file does not show any other-way around wrapping, the LwIP core asserts the input calls to be processed with locked context [`tcp_input()`](https://github.com/lwip-tcpip/lwip/blob/73fcf72792a926a4e0ac8b656b29bff70552d927/src/core/tcp_in.c#L130) [`ip4_input()`](https://github.com/lwip-tcpip/lwip/blob/73fcf72792a926a4e0ac8b656b29bff70552d927/src/core/ipv4/ip4.c#L473) [`ethernet_input()`](https://github.com/lwip-tcpip/lwip/blob/73fcf72792a926a4e0ac8b656b29bff70552d927/src/netif/ethernet.c#L89) [netif_input()](https://github.com/lwip-tcpip/lwip/blob/73fcf72792a926a4e0ac8b656b29bff70552d927/src/core/netif.c#L230).

This alone does not explain the root cause, but could contribute to.

earlephilhower, 2024-09-09:
Once I fixed the small syntax error in the sketch (`std::queue<Data>` vs `std::queue<struct Data>`) and converted it to run on my currently hooked up W5100 board using IRQ-based packet processing (same code paths as the Pico W in lwipEthernet) it's been running for 35 minutes so far without incident.
````
#include <string>
#include <lwip/tcp.h>
#include <Arduino.h>
#include <WiFi.h>
#include <W5100lwIP.h>

// To use Interrupt-driven mode, pass in an SPI object and an IRQ pin like so:
Wiznet5100lwIP eth(17, SPI, 21);

#define BURST_SIZE 20
#define MSG_SIZE TCP_SND_BUF / BURST_SIZE
#define YOUR_SSID "XXXXXXXX"
#define YOUR_PWD "XXXXXXXX"

struct tcp_pcb* pcb;

IPAddress ip(192,168,1,8);
#define REMOTE_PORT   8007
bool cnx=false;
bool issueSend = false;
uint32_t freeHeap() { return rp2040.getFreeHeap(); }
uint32_t maxBlock() { return freeHeap()/2; }
template<int I,typename... Args>
void _PRINTF(const char* fmt, Args... args) {
  Serial.printf(std::string(std::string("H4AT:% 7lu: H=%u M=%u ")+fmt).c_str(),millis(),freeHeap(),maxBlock(),args...);
}

#define PRINTF(...) _PRINTF<0>(__VA_ARGS__)

typedef struct Data {
  uint8_t* data;
  uint16_t len;
} ;
std::queue<Data> rxQueue;

uint8_t msg[MSG_SIZE];

void popQueue(std::queue<struct Data>& Q) {
  if (Q.size()) {
  auto &d = Q.front();
  free(d.data);
  Q.pop();
  }
}
void _tcp_error(void* arg, err_t err) {
  PRINTF("err=%d\n", err);
  cnx=false;
}

err_t _tcp_recv(void *arg, struct tcp_pcb *tpcb, struct pbuf *p, err_t err) {
  PRINTF("_tcp_recv %p %p %d\n", tpcb, p, err);
  if (p){ 
    PRINTF("data %p tot_len %d\n", p->payload, p->tot_len);

    auto cpydata=static_cast<uint8_t*>(malloc(p->tot_len));
    if (cpydata) {
      pbuf_copy_partial(p,cpydata,p->tot_len,0);
      rxQueue.push({cpydata,p->tot_len});
    }

    tcp_recved(tpcb, p->tot_len);
    pbuf_free(p);

  } else {
    PRINTF("Closed!\n");
    cnx=false;
  }
  return ERR_OK;
}
err_t _tcp_sent(void* arg, struct tcp_pcb* p, u16_t len) {
  Serial.printf("_tcp_sent %d\n", len);
  return ERR_OK;
}

err_t _tcp_connected(void* arg, void* tpcb, err_t err){
  Serial.printf("Connected to echoserver\n");
  cnx = true;
  issueSend = true;
  tcp_nagle_enable(pcb);
  tcp_recv(pcb, _tcp_recv);
  tcp_sent(pcb, _tcp_sent);
  return ERR_OK;
}

void sendData(const uint8_t* data,size_t len,bool copy){
    size_t  bytesSent=0;
    size_t  bytesRemaining=len;

    while(bytesRemaining){
        size_t available=tcp_sndbuf(pcb);
    PRINTF("H=%lu available=%d\n", rp2040.getFreeHeap(), available);
        if(available && (tcp_sndqueuelen(pcb) < TCP_SND_QUEUELEN )){ // if there is any space at all (and space in Q!), use it
            auto chunk=std::min(bytesRemaining,available); // send as much as will fit
            auto err=tcp_write(pcb,data+bytesSent,chunk,TCP_WRITE_FLAG_COPY);
            // error handling deliberately omitted
            PRINTF("(e=%d) AVAILABLE=%d SENT=%d REMAINING=%d QL=%d\n",err,available,bytesSent,bytesRemaining,tcp_sndqueuelen(pcb));
            err=tcp_output(pcb);
            if(err) PRINTF("ERROR! tcp_output returns %d\n",err);
            else {
              bytesSent+=chunk;
              bytesRemaining-=chunk;
            }
        }
    }
}

void setup(){
  Serial.begin(921600);
  delay(5000);
  // Start the Ethernet port
  if (!eth.begin()) {
    Serial.println("No wired Ethernet hardware detected. Check pinouts, wiring.");
    while (1) {
      delay(1000);
    }
  }

  while (!eth.connected()) {
    Serial.print(".");
    delay(500);
  }  Serial.println("");
  Serial.println("Ethernet connected");
  Serial.println("IP address: ");
  Serial.println(eth.localIP());

  //WiFi.mode(WIFI_STA);
  //WiFi.beginNoBlock(YOUR_SSID, YOUR_PWD);
  //while (WiFi.waitForConnectResult() != WL_CONNECTED){
//    Serial.println("Connection Failed! Rebooting...");
    //Serial.print(".");
    //delay(5000);
    //rp2040.reboot();
  //}
  //Serial.printf("\nIP: %s\n", WiFi.localIP().toString().c_str());
  pcb=tcp_new();
  tcp_err(pcb,_tcp_error);
  Serial.printf("MSS=%d SND_BUF=%d WND=%d SND_QUEUELEN=%d\n",TCP_MSS,TCP_SND_BUF,TCP_WND,TCP_SND_QUEUELEN);

  ip_addr_t ipt;
  ip_addr_set_ip4_u32(&ipt, ip);
  err_t err = tcp_connect(pcb, &ipt, REMOTE_PORT,(tcp_connected_fn)&_tcp_connected);
}

void loop(){
  static uint32_t prevmillis = millis();
  if (millis() - prevmillis >= 500){
    issueSend = true;
    prevmillis = millis();
  }
  if (issueSend){
#define SEND_EACH_LOOP 1
#if SEND_EACH_LOOP
    static int I = 0;
    if (I<BURST_SIZE) {
      sendData(msg, MSG_SIZE, false); // 1x only has to be on main loop
      I++;
    } else {
      issueSend=false;
      I=0;
    }
#else
    for (int i = 0; i < BURST_SIZE; i++)
      sendData(msg, MSG_SIZE, false); // 1x only has to be on main loop
    issueSend = false;
#endif
  }
  if (rxQueue.size()){
    auto d = rxQueue.front();

    /* process data on loop */
    Serial.printf("Processing Data %p %d\n", d.data, d.len);

    /* Free it */
    popQueue(rxQueue);
  }
}
````

earlephilhower, 2024-09-09:
Update:  It's at 58 mins runtime without incident so far.  I'll leave it sit overnight and if it's still good try and find an OG PicoW to replicate the experiment on.  If that dies where this runs, then the issue is inside the CYW43 driver and not the infrastructure here which would be interesting.

HamzaHajeir, 2024-09-09:
I think you can get PIO to work atop of git master, but required to remove the present one under .platformio/packages DIR [framework-arduinopico], specify the git URL in platform_packages in platformio.ini as follows: 
```ini
platform_packages =
   framework-arduinopico@https://github.com/earlephilhower/arduino-pico.git#master
```
Then PIO is enforced to download it [IDK how they run the cache regarding branches...]. 

However, it's sometimes hard to get it right away, that's why I now run a version to accelerate the incident, it gets the issue here in less a minute: 

BTW I don't have an ethernet board so couldn't run the provided.

```cpp
#include <string>
#include <lwip/tcp.h>
#include <Arduino.h>
#include <WiFi.h>

#define BURST_SIZE 20
#define MSG_SIZE TCP_SND_BUF / BURST_SIZE
#define YOUR_SSID "XXXXXXXX"
#define YOUR_PWD "XXXXXXXX"

struct tcp_pcb* pcb;

IPAddress ip(192,168,100,4);
#define REMOTE_PORT 	8007
bool cnx=false;
bool issueSend = false;
uint32_t freeHeap() { return rp2040.getFreeHeap(); }
uint32_t maxBlock() { return freeHeap()/2; }
template<int I,typename... Args>
void _PRINTF(const char* fmt, Args... args) {
	Serial.printf(std::string(std::string("H4AT:% 7lu: H=%u M=%u ")+fmt).c_str(),millis(),freeHeap(),maxBlock(),args...);
}

#define PRINTF(...) _PRINTF<0>(__VA_ARGS__)

struct Data {
	uint8_t* data;
	uint16_t len;
};
std::queue<struct Data> rxQueue;

uint8_t msg[MSG_SIZE];

void popQueue(std::queue<struct Data>& Q) {
  if (Q.size()) {
	auto &d = Q.front();
	free(d.data);
	Q.pop();
  }
}
void _tcp_error(void* arg, err_t err) {
	PRINTF("err=%d\n", err);
	cnx=false;
}

err_t _tcp_recv(void *arg, struct tcp_pcb *tpcb, struct pbuf *p, err_t err) {
	PRINTF("_tcp_recv %p %p %d\n", tpcb, p, err);
	if (p){ 
		PRINTF("data %p tot_len %d\n", p->payload, p->tot_len);

		auto cpydata=static_cast<uint8_t*>(malloc(p->tot_len));
		if (cpydata) {
			// pbuf_copy_partial(p,cpydata,p->tot_len,0);
			rxQueue.push({cpydata,p->tot_len});
		}

		for (int i=0; i<10; i++) {
			auto d = static_cast<uint8_t*>(malloc(p->tot_len));
			rxQueue.push({d,p->tot_len});
			delay(10);
		}

		tcp_recved(tpcb, p->tot_len);
		pbuf_free(p);

	} else {
		PRINTF("Closed!\n");
		cnx=false;
	}
	return ERR_OK;
}
err_t _tcp_sent(void* arg, struct tcp_pcb* p, u16_t len) {
	Serial.printf("_tcp_sent %d\n", len);
	return ERR_OK;
}

err_t _tcp_connected(void* arg, void* tpcb, err_t err){
	Serial.printf("Connected to echoserver\n");
	cnx = true;
	issueSend = true;
	tcp_nagle_enable(pcb);
	tcp_recv(pcb, _tcp_recv);
	tcp_sent(pcb, _tcp_sent);
	return ERR_OK;
}

void sendData(const uint8_t* data,size_t len,bool copy){
    size_t  bytesSent=0;
    size_t  bytesRemaining=len;

    while(bytesRemaining){
        size_t available=tcp_sndbuf(pcb);
		PRINTF("H=%lu available=%d\n", rp2040.getFreeHeap(), available);
        if(available && (tcp_sndqueuelen(pcb) < TCP_SND_QUEUELEN )){ // if there is any space at all (and space in Q!), use it
            auto chunk=std::min(bytesRemaining,available); // send as much as will fit
            auto err=tcp_write(pcb,data+bytesSent,chunk,TCP_WRITE_FLAG_COPY);
            // error handling deliberately omitted
            PRINTF("(e=%d) AVAILABLE=%d SENT=%d REMAINING=%d QL=%d\n",err,available,bytesSent,bytesRemaining,tcp_sndqueuelen(pcb));
            err=tcp_output(pcb);
            if(err) PRINTF("ERROR! tcp_output returns %d\n",err);
            else {
              bytesSent+=chunk;
              bytesRemaining-=chunk;
            }
        }
    }
}

void setup(){
	Serial.begin(921600);
	WiFi.mode(WIFI_STA);
	WiFi.beginNoBlock(YOUR_SSID, YOUR_PWD);
	while (WiFi.waitForConnectResult() != WL_CONNECTED){
		Serial.println("Connection Failed! Rebooting...");
		Serial.print(".");
		delay(5000);
		rp2040.reboot();
	}
	Serial.printf("\nIP: %s\n", WiFi.localIP().toString().c_str());
	pcb=tcp_new();
	tcp_err(pcb,_tcp_error);
	Serial.printf("MSS=%d SND_BUF=%d WND=%d SND_QUEUELEN=%d\n",TCP_MSS,TCP_SND_BUF,TCP_WND,TCP_SND_QUEUELEN);

	ip_addr_t ipt;
	ip_addr_set_ip4_u32(&ipt, ip);
	err_t err = tcp_connect(pcb, &ipt, REMOTE_PORT,(tcp_connected_fn)&_tcp_connected);
}

void loop(){
	static uint32_t prevmillis = millis();
	if (millis() - prevmillis >= 500){
		issueSend = true;
		prevmillis = millis();
	}
	if (issueSend){
#define SEND_EACH_LOOP 1
#if SEND_EACH_LOOP
		static int I = 0;
		if (I<BURST_SIZE) {
			sendData(msg, MSG_SIZE, false); // 1x only has to be on main loop
			I++;
		} else {
			issueSend=false;
			I=0;
		}
#else
		for (int i = 0; i < BURST_SIZE; i++)
			sendData(msg, MSG_SIZE, false); // 1x only has to be on main loop
		issueSend = false;
#endif
	}
	if (rxQueue.size()){
		auto &d = rxQueue.front();

		/* process data on loop */
		Serial.printf("Processing Data %p %d\n", d.data, d.len);

		/* Free it */
		popQueue(rxQueue);
	}
}
```

Per `struct` in cpp, AFAIK it's technically the same to `class`, but differs in default accessibility (public for struct). 

But anyway I've used `struct Data` instead of `Data`.

earlephilhower, 2024-09-09:
The sample here ran for > 12hrs so far, so killing.  Will try w/a Pico W instead, soon.  If that fails then the issue is pinpointed to something related to the PicoSDK CYW43 implementation.

earlephilhower, 2024-09-09:
I have been running the [same example](https://github.com/earlephilhower/arduino-pico/issues/2430#issuecomment-2336995773) using a PicoW for 60 mins so far, no issues seen.  Still dumping
````
....
H4AT:3276075: H=188944 M=94472 _tcp_recv 0x2000d3f0 0x2000c7e0 0
H4AT:3276076: H=188944 M=94472 data 0x2000c826 tot_len 584
H4AT:3276074: H=189000 M=94500 H=189056 available=10512
H4AT:3276077: H=188384 M=94192 (e=0) AVAILABLE=10512 SENT=0 REMAINING=584 QL=2
_tcp_sent 584
....
````

Qualitatively, the throughput of log messages seems lower vs. the wired test but I do not have an exact characterization there.  It's running at a good rate, just that the wired connection felt like it was always writing.  With WiFi it's maybe .5s between log dumps.

I'll keep it going for now, but I would suggest trying `git master` win your site and see if you can still repro the failure.  Something in the RP2350B update/SDK 2.0 move may have unbroken a weird corner case or something...

````
#include <string>

#include <lwip/tcp.h>

#include <Arduino.h>
#include <WiFi.h>

#define BURST_SIZE 20
#define MSG_SIZE TCP_SND_BUF / BURST_SIZE
#define YOUR_SSID XXXX
#define YOUR_PWD YYYY

struct tcp_pcb* pcb;

IPAddress ip(192,168,1,8);
#define REMOTE_PORT   8007
bool cnx=false;
bool issueSend = false;
uint32_t freeHeap() { return rp2040.getFreeHeap(); }
uint32_t maxBlock() { return freeHeap()/2; }
template<int I,typename... Args>
void _PRINTF(const char* fmt, Args... args) {
  Serial.printf(std::string(std::string("H4AT:% 7lu: H=%u M=%u ")+fmt).c_str(),millis(),freeHeap(),maxBlock(),args...);
}

#define PRINTF(...) _PRINTF<0>(__VA_ARGS__)

typedef struct Data {
  uint8_t* data;
  uint16_t len;
} ;
std::queue<Data> rxQueue;

uint8_t msg[MSG_SIZE];

void popQueue(std::queue<struct Data>& Q) {
  if (Q.size()) {
  auto &d = Q.front();
  free(d.data);
  Q.pop();
  }
}
void _tcp_error(void* arg, err_t err) {
  PRINTF("err=%d\n", err);
  cnx=false;
}

err_t _tcp_recv(void *arg, struct tcp_pcb *tpcb, struct pbuf *p, err_t err) {
  PRINTF("_tcp_recv %p %p %d\n", tpcb, p, err);
  if (p){ 
    PRINTF("data %p tot_len %d\n", p->payload, p->tot_len);

    auto cpydata=static_cast<uint8_t*>(malloc(p->tot_len));
    if (cpydata) {
      pbuf_copy_partial(p,cpydata,p->tot_len,0);
      rxQueue.push({cpydata,p->tot_len});
    }

    tcp_recved(tpcb, p->tot_len);
    pbuf_free(p);

  } else {
    PRINTF("Closed!\n");
    cnx=false;
  }
  return ERR_OK;
}
err_t _tcp_sent(void* arg, struct tcp_pcb* p, u16_t len) {
  Serial.printf("_tcp_sent %d\n", len);
  return ERR_OK;
}

err_t _tcp_connected(void* arg, void* tpcb, err_t err){
  Serial.printf("Connected to echoserver\n");
  cnx = true;
  issueSend = true;
  tcp_nagle_enable(pcb);
  tcp_recv(pcb, _tcp_recv);
  tcp_sent(pcb, _tcp_sent);
  return ERR_OK;
}

void sendData(const uint8_t* data,size_t len,bool copy){
    size_t  bytesSent=0;
    size_t  bytesRemaining=len;

    while(bytesRemaining){
        size_t available=tcp_sndbuf(pcb);
    PRINTF("H=%lu available=%d\n", rp2040.getFreeHeap(), available);
        if(available && (tcp_sndqueuelen(pcb) < TCP_SND_QUEUELEN )){ // if there is any space at all (and space in Q!), use it
            auto chunk=std::min(bytesRemaining,available); // send as much as will fit
            auto err=tcp_write(pcb,data+bytesSent,chunk,TCP_WRITE_FLAG_COPY);
            // error handling deliberately omitted
            PRINTF("(e=%d) AVAILABLE=%d SENT=%d REMAINING=%d QL=%d\n",err,available,bytesSent,bytesRemaining,tcp_sndqueuelen(pcb));
            err=tcp_output(pcb);
            if(err) PRINTF("ERROR! tcp_output returns %d\n",err);
            else {
              bytesSent+=chunk;
              bytesRemaining-=chunk;
            }
        }
    }
}

void setup(){
  Serial.begin(921600);
  delay(5000);

  WiFi.mode(WIFI_STA);
  WiFi.beginNoBlock(YOUR_SSID, YOUR_PWD);
  while (WiFi.waitForConnectResult() != WL_CONNECTED){
    Serial.println("Connection Failed! Rebooting...");
    Serial.print(".");
    delay(5000);
    rp2040.reboot();
  }
  Serial.printf("\nIP: %s\n", WiFi.localIP().toString().c_str());
  pcb=tcp_new();
  tcp_err(pcb,_tcp_error);
  Serial.printf("MSS=%d SND_BUF=%d WND=%d SND_QUEUELEN=%d\n",TCP_MSS,TCP_SND_BUF,TCP_WND,TCP_SND_QUEUELEN);

  ip_addr_t ipt;
  ip_addr_set_ip4_u32(&ipt, ip);
  err_t err = tcp_connect(pcb, &ipt, REMOTE_PORT,(tcp_connected_fn)&_tcp_connected);
}

void loop(){
  static uint32_t prevmillis = millis();
  if (millis() - prevmillis >= 500){
    issueSend = true;
    prevmillis = millis();
  }
  if (issueSend){
#define SEND_EACH_LOOP 1
#if SEND_EACH_LOOP
    static int I = 0;
    if (I<BURST_SIZE) {
      sendData(msg, MSG_SIZE, false); // 1x only has to be on main loop
      I++;
    } else {
      issueSend=false;
      I=0;
    }
#else
    for (int i = 0; i < BURST_SIZE; i++)
      sendData(msg, MSG_SIZE, false); // 1x only has to be on main loop
    issueSend = false;
#endif
  }
  if (rxQueue.size()){
    auto d = rxQueue.front();

    /* process data on loop */
    Serial.printf("Processing Data %p %d\n", d.data, d.len);

    /* Free it */
    popQueue(rxQueue);
  }
}
````

HamzaHajeir, 2024-09-09:
I've just seen a bug within the acceleration code regarding memory management, Therefore I need to make sure about the incident over v4.0.1 and further test on master.

I'll feedback the conclusion soon.

earlephilhower, 2024-09-10:
The PicoW is at 3 hrs now and I need the USB port back so I'm going to stop it and replace it with a 2350.  FWIW, no issue was seen using the latest `git master`.  Will wait for your update...

earlephilhower, 2024-09-13:
Any update w/the latest releases on this, @HamzaHajeir ?

HamzaHajeir, 2024-09-13:
Hi Earle, I was just drafting this message. 

Had the final conclusion: The bug disappears with the new release 4.0.1. And the incident within 4.0.1 was an issue with memory management.

closed 2024-09-13 by HamzaHajeir

HamzaHajeir, 2024-09-13:
@earlephilhower 

While the issue was replicated in 3.9.5, it was not replicated by any correct code in 4.0.1, including the continuous stress test modified from H4AsyncTCP library.

What's worth to add that I can confirm the upstack calls (lwip->app) are locked by the async_context (at least for 4.0.1). 

Thank you for taking the issue seriously. Sometime soon I'll announce the support in Discussions.

earlephilhower, 2024-09-13:
Great, glad to hear it's running.  Thanks for the update!