earlephilhower/arduino-pico #2434: F_CPU doesn't match actual sys_clk frequency

Yveaux, 2024-09-09:
I'm testing different sys_clk frequencies on rp2350 (pico2) using platformio and latest arduino-pico.
The board frequency is configured to 125MHz in platformio.ini, which correctly shows in F_CPU macro.
However, reading the actual clock frequency at startup of the code shows sys_clk running at 150MHz.
After reconfiguring sys_clk to F_CPU at runtime, the correct frequency is reported.

This issue could be caused by platformio; I'm not sure.
Follow code easily reproduces this behavior:

platformio.ini:
```
[env]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
framework = arduino
board_build.core = earlephilhower
monitor_speed = 115200
upload_protocol = cmsis-dap
debug_tool = cmsis-dap
debug_speed = 20000

debug_build_flags =
  -O3

build_flags =
  -D DEBUG_PORT=Serial1
  -D DEBUG_BAUDRATE=115200
  -D DEBUG_RP2040_PORT=DEBUG_PORT

[env:pico2]
board = rpipico2
board_build.f_cpu = 125000000L
```
main.cpp
```
#include <Arduino.h>

void setup()
{
  DEBUG_PORT.begin(115200);
  DEBUG_PORT.print("Configured F_CPU: "); DEBUG_PORT.println(F_CPU);
  DEBUG_PORT.print("Actual F_CPU:     "); DEBUG_PORT.println(rp2040.f_cpu());

  DEBUG_PORT.println("Reconfigure sys_clk to F_CPU");
  DEBUG_PORT.flush();

  set_sys_clock_hz(F_CPU, true);

  DEBUG_PORT.begin(115200);
  DEBUG_PORT.print("Actual F_CPU:     "); DEBUG_PORT.println(rp2040.f_cpu());
}

void loop()
{
}
```
Serial output:
```
Configured F_CPU: 125000000
Actual F_CPU:     150000000
Reconfigure sys_clk to F_CPU
Actual F_CPU:     125000000
```

earlephilhower, 2024-09-09:
You ran into the *one* case where the bug would show itself on the RP2350.  Congratulations!  :laughing: 
https://github.com/earlephilhower/arduino-pico/blob/3e24400bebcbf9be5731f1f4a72a6e864be0da80/cores/rp2040/main.cpp#L82-L84

Basically that needs to be updated to include the RP2350 default of 150.

Yveaux, 2024-09-09:
> You ran into the _one_ case where the bug would show itself on the RP2350. Congratulations! 😆

Yeah, I tend to do that... All by just picking some values... Guess I'm 'blessed' 🤔 

Anyway, I also see some strange behavior with SPI frequency defaulting to 24MHz after changing sys_clk from software. I now understand why I didn't see this at 125MHz.

Yveaux, 2024-09-09:
Tested correctly working for 125 & 150MHz!
Thanks for the awesome quick fix!

```
Configured F_CPU: 125000000
Actual F_CPU:     125000000
Reconfigure sys_clk to F_CPU
Actual F_CPU:     125000000
```

```
Configured F_CPU: 150000000
Actual F_CPU:     150000000
Reconfigure sys_clk to F_CPU
Actual F_CPU:     150000000
```

PR #2436 (2024-09-09): Fix F_CPU 125MHz setting on RP2350
Fixes #2434
commit a9b39029 (2024-09-09): Fix F_CPU 125MHz setting on RP2350 (#2436)

Fixes #2434
closed 2024-09-09 by earlephilhower