earlephilhower/arduino-pico #2442: Pimorni Pico Plus 2 in list of supported boards but not available in the boards menu under the 4.01 menu

Macrosii, 2024-09-11:
Hi
Under the supported boards listing there are the following Pimoroni boards listed;
Pimoroni PGA2040
Pimoroni Pico Plus 2
Pimoroni Plasma2040
Pimoroni Tiny2040
Pimoroni Tiny2350

However under the boards menu for 4.01, it seems to be missing the two new RP2350 boards (apologies if I am missing something).

Regards

earlephilhower, 2024-09-11:
You're looking at the `git head` readme and running the `4.0.1` release.  The boards are in `git` but put in after 4.0.1.  They'll be in the next release.

closed 2024-09-11 by earlephilhower

Macrosii, 2024-09-11:
As I said apologies if being stupid!!!
No sure what you mean by "git head" but was just looking at the Boards Listing on the main page of the repository "https://github.com/earlephilhower/arduino-pico"

earlephilhower, 2024-09-11:
Yes, that's the `head` (current working point) of the source code control system, `git`.