earlephilhower/arduino-pico #2447: SerialPIO on RP235X

vslinuxdotnet, 2024-09-12:
Hello,
I test the SerialPIO on RP235X with a ESP8266 over serial and only the Transmit works, I don't receive anything over serial.

some test:
```
SerialPIO ser6(10,11);//tx/rx

void setup() {
    delay(1000);
    Serial.begin(9600);
    ser6.begin(115200);
    
    Serial.println("\n-----boot--------");
   }

void loop() {

ser6.write("AT+GMR\r\n");
delay(100);
  while (ser6.available()) {
    Serial.println(ser6.readString());
  }
delay(1000);
}

```
I connect a serial to usb module in 115200 and have data from ESP8266 :

  

> OK
>   AT+GMR
>   AT version:2.2.0.0(b097cdf - ESP8266 - Jun 17 2021 12:57:45)
>   SDK version:v3.4-22-g967752e2
>   compile time(6800286):Aug  4 2021 17:20:05
>   Bin version:2.2.0(Cytron_ESP-01S)

Can be some bug in SerialPIO on RP235X  ?

the device:
[image]

Thanks VS

earlephilhower, 2024-09-12:
Ah, very interesting!  I have triple-checked the output (which was the one using 32b bit shifts so wasn't working on GPIOS >31 on the 2350B) and it's good, and just re-verified it and it's still good.

Input, however, seems busted and only on the 2350.  If you install a loopback from GP4<-GP6> and run this modified test on a 2040
````
SerialPIO ser6(4, 6); //tx/rx
void setup() {
  delay(5000);
  Serial.begin(9600);
  ser6.begin(115200);
  Serial.println("\n-----boot--------");
}

void loop() {
  ser6.write("AT+GMR\r\n");
  delay(100);
  while (ser6.available()) {
    Serial.printf("%c", ser6.read());
  }
  Serial.printf("----\n----\n");
}
````
You get the AT+GMR reported back
````
---
----
AT+GMR
----
----
AT+GMR
----
----
AT+GMR
----
----
````

But same app on the 2350 doesn't actually notice received data.

Should be a simple pin issue because there's no special casing in SerialPIO per-chip type.

earlephilhower, 2024-09-13:
Can you try #2451, @vslinuxdotnet, and report back?  Loopback tests now work, and it was pretty obvious the issue once I looked into it.

vslinuxdotnet, 2024-09-13:
Hello, yes In two hours.
Thanks

earlephilhower, 2024-09-13:
Sorry, can't reproduce that here.  If a single character gets read, they all should be deposited into the FIFO.  I bet you're reading faster than the 8266 can write chars out and as such the `available()` goes to 0 while the 8266 is still thinking.

I suggest something simple like
````
SerialPIO ser6(4, 6); //tx/rx (whatever your board is wired to)
void setup() {
  delay(5000);
  Serial.begin(9600);
  ser6.begin(115200);
  delay(100); // Make sure everyone is alive and running on the 8266 to respond
  ser6.write("AT+GMR\r\n");
}

void loop() {
while (ser6.available()) {
    Serial.printf("%c", ser6.read());
  }
}
````

closed 2024-09-13 by earlephilhower

vslinuxdotnet, 2024-09-13:
yes that's one problem the slow response, and there's more that matter:

1º 8266 with AT Firmware takes 2000 to boot after reset.
2º 8266 is slow to answer, and read all data in serial port takes 3 or 4 cpu loops to read 5 lines.
3º must not be delays in loop or some data will be missed.
4º 8266 with AT Firmware don't accept  commands < 2 seconds, because takes some time do process one and ignore all the others.

I solved like that:
```
unsigned long currentMillis = millis();

 if (currentMillis - heartbeatMillis >= 3000)
{
    heartbeatMillis = currentMillis;
    ser6.write("AT+GMR\r\n");
}

  while (ser6.available()) {
    Serial.printf("%c", ser6.read());
  }
```
@earlephilhower  thanks for you time.
Regards VS

earlephilhower, 2024-09-13:
If you can rewire the ESP to a SPI port, look at https://github.com/earlephilhower/arduino-pico/tree/master/libraries/lwIP_ESPHost which lets you use all the built-in networking here at much higher speed than an AT-based serial port.

PR #2451 (2024-09-13): Support PIO2 for SoftwareSerial/SerialPIO
New IRQs and PIO pointers required for the 3rd PIO on the 2350.

Fixes #2447
commit aaaa99d9 (2024-09-13): Support PIO2 for SoftwareSerial/SerialPIO (#2451)

New IRQs and PIO pointers required for the 3rd PIO on the 2350.

Fixes #2447