earlephilhower/arduino-pico #2448: Bluetooth implementations don't remember devices they've paired to previously

avra-m3, 2024-09-13:
I've gone down a bit of a rabbit hole and if my understanding is correct, the cause of this appears to be a missing implementation of `btstack_link_key_db_t` as outlined (very poorly) in https://bluekitchen-gmbh.com/btstack/porting/#sec:persistentStoragePorting.

I'll be playing around with this independently, if I make any progress would you be open to accepting a PR?

earlephilhower, 2024-09-13:
Kudos for reading through the BTStack docs!  Now, extra points if you can find where the docs actually match the implementation. :laughing:

In my experience, BT pairing across reboots to the exact same device works OK most of the time.  However, trying to pair to another device after unpairing from a 1st one never seems to work until a reboot indicating a RAM-state issue, not a stable-store one IMO.

In any case, we do persist the TLV to flash using the SDK's own routine.  See https://github.com/raspberrypi/pico-sdk/blob/efe2103f9b28458a1615ff096054479743ade236/src/rp2_common/pico_btstack/btstack_flash_bank.c, slightly modified to not overwrite a fixed flash sector and avoid multicore crashes https://github.com/earlephilhower/arduino-pico/blob/master/cores/rp2040/sdkoverride/btstack_flash_bank.cpp

If you do get something working better, I'd absolutely be happy to incorporate it!  If that happens, you should really consider contributing upstream to the SDK so all apps will get the fix.

avra-m3, 2024-09-13:
Ahh interesting! So something else is causing my pico to not reconnect. Should have known this was too good to be true. I'll still attempt to follow this thread and see if it has any impact. Less optimistic now though.

earlephilhower, 2024-09-13:
#1817 is the thing I've run into most.  It was reported and reproduced upstream by others on the naked SDK.  Since we don't have any access to the BT stack running on the CYW43, and nobody here seems to be a low-level HCI expert, we left it as a "we'll pull the fix in from upstream, should one appear."

That said, if you do have a issues with reconnection between reboots then please go ahead and look into it if you're interested.  But I don't think it's as gross an error as not actually writing bits to stable store, unfortunately.

avra-m3, 2024-09-14:
Didn't meant to imply it was a gross error 😅 .

I'm very new to the embedded world but know how easy it is for a small line of documentation to be missed. I'll keep investigating, it's very odd and happening to both my Pico Ws using the BT implementation. That said #1817 is still applicable to me, I wasn't able to get the BLE implementation to send anything unfortunately (even doing the forget device / flash keyboard dance I've seen recommended).

Starting to dive into the lower layers the arduino-pico library but I don't expect to discover anything that someone hasn't noticed already.

If I find anything interesting I'll add it to this thread but feel free to go ahead and close this one off for now if you'd like to keep the issues clear 😄.

earlephilhower, 2024-10-29:
Maybe https://github.com/raspberrypi/pico-sdk/issues/2006 is the root

earlephilhower, 2024-11-20:
The SDK issue I opened was a goof, no real problem.  NOR flash can update 1->0 in any order, not just incrementally, so the TLV store was good.  

As far as I know, nothing here has changed related to this.  Possibly the SDK updates in the meanwhile have brought in a BTStack fix that helps with your case.

closed 2024-11-20 by earlephilhower