earlephilhower/arduino-pico #2457: Unable to upload using rpi debugprobe cmsis-dap from platformio

Yveaux, 2024-09-16:
Hi there! I'm trying to upload to rpi pico with a raspberry debugprobe.
Everything is connected according to the manuals and is working correctly in Arduino-IDE (board manager URL: https://github.com/earlephilhower/arduino-pico/releases/download/global/package_rp2040_index.json and Upload method: Picoprobe/Debugprobe (CMSIS_DAP)) with arduino-pico installed, on Windows 11.
debugprobe has WinUSB 6.1.7600.16385 driver installed with Zadig.

Arduino-IDE output:
```
Sketch uses 53068 bytes (2%) of program storage space. Maximum is 2093056 bytes.
Global variables use 9436 bytes (3%) of dynamic memory, leaving 252708 bytes for local variables. Maximum is 262144 bytes.
Open On-Chip Debugger 0.12.0+dev-gebec9504d (2024-08-27-18:27)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : Hardware thread awareness created
Info : Hardware thread awareness created
Warn : An adapter speed is not selected in the init scripts. OpenOCD will try to run the adapter at very low speed (100 kHz).
Warn : To remove this warnings and achieve reasonable communication speed with the target, set "adapter speed" or "jtag_rclk" in the init scripts.
Info : Using CMSIS-DAPv2 interface with VID:PID=0x2e8a:0x000c, serial=E6614103E72A9D24
Info : CMSIS-DAP: SWD supported
Info : CMSIS-DAP: Atomic commands supported
Info : CMSIS-DAP: Test domain timer supported
Info : CMSIS-DAP: FW Version = 2.0.0
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : SWCLK/TCK = 0 SWDIO/TMS = 0 TDI = 0 TDO = 0 nTRST = 0 nRESET = 0
Info : CMSIS-DAP: Interface ready
Info : clock speed 100 kHz
Info : SWD DPIDR 0x0bc12477, DLPIDR 0x00000001
Info : SWD DPIDR 0x0bc12477, DLPIDR 0x10000001
Info : [rp2040.core0] Cortex-M0+ r0p1 processor detected
Info : [rp2040.core0] target has 4 breakpoints, 2 watchpoints
Info : [rp2040.core0] Examination succeed
Info : [rp2040.core1] Cortex-M0+ r0p1 processor detected
Info : [rp2040.core1] target has 4 breakpoints, 2 watchpoints
Info : [rp2040.core1] Examination succeed
Info : starting gdb server for rp2040.core0 on 3333
Info : Listening on port 3333 for gdb connections
[rp2040.core0] halted due to breakpoint, current mode: Thread 
xPSR: 0xf1000000 pc: 0x000000ee msp: 0x20041f00
[rp2040.core1] halted due to debug-request, current mode: Thread 
xPSR: 0xf1000000 pc: 0x000000ee msp: 0x20041f00
** Programming Started **
Warn : Function FUNC_BOOTROM_STATE_RESET not found in RP2xxx ROM. (probably an RP2040 or an RP2350 A0)
Warn : Function FUNC_FLASH_RESET_ADDRESS_TRANS not found in RP2xxx ROM. (probably an RP2040 or an RP2350 A0)
Info : RP2040 Flash Probe: 33554432 bytes @0x10000000, in 8192 sectors

Info : Padding image section 1 at 0x100110e0 with 32 bytes (bank write end alignment)
Warn : Adding extra erase range, 0x10011100 .. 0x10011fff
** Programming Finished **
** Verify Started **
** Verified OK **
```

Just running openocd in a windows command prompt also discovers the debugprobe with connected rp2040 correctly:
```
c:\Users\Yveaux\.platformio\packages\tool-openocd-rp2040-earlephilhower\bin>openocd -f c:\Users\Yveaux\.platformio\packages\framework-arduinopico\lib\rp2040\picoprobe_cmsis_dap.tcl
Open On-Chip Debugger 0.12.0+dev-gebec9504d (2024-08-27-18:27)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : Hardware thread awareness created
Info : Hardware thread awareness created
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : Using CMSIS-DAPv2 interface with VID:PID=0x2e8a:0x000c, serial=E6614103E72A9D24
Info : CMSIS-DAP: SWD supported
Info : CMSIS-DAP: Atomic commands supported
Info : CMSIS-DAP: Test domain timer supported
Info : CMSIS-DAP: FW Version = 2.0.0
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : SWCLK/TCK = 0 SWDIO/TMS = 0 TDI = 0 TDO = 0 nTRST = 0 nRESET = 0
Info : CMSIS-DAP: Interface ready
Info : clock speed 5000 kHz
Info : SWD DPIDR 0x0bc12477, DLPIDR 0x00000001
Info : SWD DPIDR 0x0bc12477, DLPIDR 0x10000001
Info : [rp2040.core0] Cortex-M0+ r0p1 processor detected
Info : [rp2040.core0] target has 4 breakpoints, 2 watchpoints
Info : [rp2040.core0] Examination succeed
Info : [rp2040.core1] Cortex-M0+ r0p1 processor detected
Info : [rp2040.core1] target has 4 breakpoints, 2 watchpoints
Info : [rp2040.core1] Examination succeed
Info : starting gdb server for rp2040.core0 on 3333
Info : Listening on port 3333 for gdb connections
shutdown command invoked
```

However, a simple blink sketch in platformio consistently fails to upload with `embedded:startup.tcl:28: Error: Can't find cmsis-dap`:

platformio.ini
```
[env:rpipico]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
board = rpipico
framework = arduino
board_build.core = earlephilhower
upload_protocol = cmsis-dap
debug_tool = cmsis-dap
```

main.cpp
```
#include <Arduino.h>

void setup()
{
  pinMode(LED_BUILTIN, OUTPUT);
}

void loop()
{
  digitalWrite(LED_BUILTIN, HIGH);
  delay(200);
  digitalWrite(LED_BUILTIN, LOW);
  delay(200);
}
```

```
Executing task: C:\Users\Yveaux\.platformio\penv\Scripts\platformio.exe run --target upload --environment rpipico 

Processing rpipico (platform: https://github.com/maxgerhardt/platform-raspberrypi.git; board: rpipico; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------
Tool Manager: Installing https://github.com/earlephilhower/pico-quick-toolchain/releases/download/2.2.0/x86_64-w64-mingw32.openocd-ebec9504d.240827.zip
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: tool-openocd-rp2040-earlephilhower@5.120300.240827 has been installed!
Tool Manager: Installing earlephilhower/tool-mklittlefs-rp2040-earlephilhower @ ~5.100300.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: tool-mklittlefs-rp2040-earlephilhower@5.100300.230216 has been installed!
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/raspberrypi/rpipico.html
PLATFORM: Raspberry Pi RP2040 (1.14.0+sha.1282d20) > Pico
HARDWARE: RP2040 133MHz, 256KB RAM, 2MB Flash
DEBUG: Current (cmsis-dap) External (blackmagic, cmsis-dap, jlink, picoprobe, raspberrypi-swd)
PACKAGES:
 - framework-arduinopico @ 1.40002.0+sha.aaaa99d
 - tool-mklittlefs-rp2040-earlephilhower @ 5.100300.230216 (10.3.0)
 - tool-openocd-rp2040-earlephilhower @ 5.120300.240827 (12.3.0)
 - tool-picotool-rp2040-earlephilhower @ 5.120300.240827 (12.3.0)
 - toolchain-rp2040-earlephilhower @ 5.120300.240125 (12.3.0)
Flash size: 2.00MB
Sketch size: 2.00MB
Filesystem size: 0.00MB
Maximium Sketch size: 2093056 EEPROM start: 0x101ff000 Filesystem start: 0x101ff000 Filesystem end: 0x101ff000
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 64 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Retrieving maximum program size .pio\build\rpipico\firmware.elf
Flash size: 2.00MB
Sketch size: 2.00MB
Filesystem size: 0.00MB
Maximium Sketch size: 2093056 EEPROM start: 0x101ff000 Filesystem start: 0x101ff000 Filesystem end: 0x101ff000
Checking size .pio\build\rpipico\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [          ]   3.5% (used 9244 bytes from 262144 bytes)
Flash: [          ]   2.8% (used 57804 bytes from 2093056 bytes)
Configuring upload protocol...
AVAILABLE: blackmagic, cmsis-dap, jlink, picoprobe, picotool, raspberrypi-swd
CURRENT: upload_protocol = cmsis-dap
Uploading .pio\build\rpipico\firmware.elf
Open On-Chip Debugger 0.12.0+dev-gebec9504d (2024-08-27-18:27)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 1
embedded:startup.tcl:28: Error: Can't find cmsis-dap
in procedure 'script'
at file "embedded:startup.tcl", line 28
*** [upload] Error 1
============================================================ [FAILED] Took 11.23 seconds ============================================================
```

I went as far as uninstalling platformio from vscode, deleted the .platformio folder and the .pio and .vscode folders. Then restarted vscode, reinstalled platformio and platform-raspberrypi / arduino-pico.

About to loose my mind here... 🙄

Yveaux, 2024-09-16:
Progress! I tested various 'upload_protocols' and `picoprobe` works, including debugging!
```
[env:rpipico]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
board = rpipico
framework = arduino
board_build.core = earlephilhower
upload_protocol = picoprobe
debug_tool = picoprobe
```
```
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [          ]   3.5% (used 9244 bytes from 262144 bytes)
Flash: [          ]   2.8% (used 57804 bytes from 2093056 bytes)
Configuring upload protocol...
AVAILABLE: blackmagic, cmsis-dap, jlink, picoprobe, picotool, raspberrypi-swd
CURRENT: upload_protocol = picoprobe
Uploading .pio\build\rpipico\firmware.elf
Open On-Chip Debugger 0.12.0+dev-gebec9504d (2024-08-27-18:27)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 1
adapter speed: 5000 kHz
[rp2040.core0] halted due to breakpoint, current mode: Thread 
xPSR: 0xf1000000 pc: 0x000000ee msp: 0x20041f00
[rp2040.core1] halted due to debug-request, current mode: Thread
xPSR: 0xf1000000 pc: 0x000000ee msp: 0x20041f00
** Programming Started **
Warn : Function FUNC_BOOTROM_STATE_RESET not found in RP2xxx ROM. (probably an RP2040 or an RP2350 A0)
Warn : Function FUNC_FLASH_RESET_ADDRESS_TRANS not found in RP2xxx ROM. (probably an RP2040 or an RP2350 A0)
Warn : Adding extra erase range, 0x10011100 .. 0x10011fff
** Programming Finished **
** Verify Started **
** Verified OK **
[rp2040.core0] halted due to breakpoint, current mode: Thread 
xPSR: 0xf1000000 pc: 0x000000ee msp: 0x20041f00
[rp2040.core1] halted due to debug-request, current mode: Thread
xPSR: 0xf1000000 pc: 0x000000ee msp: 0x20041f00
shutdown command invoked
================================================================ [SUCCESS] Took 5.11 seconds ================================================================
```

Did this change recently and did I miss it in the release notes maybe?

earlephilhower, 2024-09-16:
You should [double check you're running the latest p.io git](https://github.com/earlephilhower/arduino-pico/blob/master/docs/platformio.rst) and follow up in @maxgerhardt 's [repo on this](https://github.com/maxgerhardt/platform-raspberrypi).  I believe P.IO uses different tool binaries for debugging than we do, and definitely uses different scripts.  The core here has a new version of `openocd` built for the RP2350 core, not sure if P.IO has moved up yet.

closed 2024-09-16 by earlephilhower

maxgerhardt, 2024-09-16:
Looking into it. I did a change yesterday to redirect `picropobe` into `cmsis-dap`, maybe I did a small mistake.

maxgerhardt, 2024-09-16:
Woopsie. That's a reproducable error. I was missing a bracket. You exactly cought the moment in which the bug was introduced.

[image]

Fixed in https://github.com/maxgerhardt/platform-raspberrypi/commit/9e55f6db5c56b9867c69fe473f388beea4546672.

Delete `C:\Users\<user>\.platformio\raspberrypi*` and rebuild.

Yveaux, 2024-09-16:
Thanks gents for the fast response!
As a user it's not always clear if an issue is in arduini-pico or in platformio support, buit I really appreciate both of you supporting where possible!

@maxgerhardt Do you know if there is a difference in functionality for `cmsis-dap` vs `picoprobe` with the rpi debugprobe?

maxgerhardt, 2024-09-16:
The debugprobe firmware implements the CMSIS-DAP protocol. Picoprobe was the old name before Debug Probe, *and* it implemented an entirely different protocol (not CMSIS-DAP), and was since removed from the RaspberryPi's OpenOCD code. The old Picoprobe firmware / debugging way is not supported. I just redirect the name to CMSIS-DAP (DebugProbe) for legacy's sake.

Yveaux, 2024-09-16:
Clear, thanks!