earlephilhower/arduino-pico #2460: RP2350B and serial

NollKollTroll, 2024-09-16:
It is possible to set serial functions on the pins >=32, but I can't get it to work. All pins <=31 works as expected.
Using a Pimoroni PGA2350, tested with 4.0.2, both Generic 2350 and PGA2350 profile.
The 4.0.2 release docs does not mention serial as working for the upper pins, the proper docs has no mention at all.

earlephilhower, 2024-09-16:
It should work per the code, but I've spent much more time on the PIO-based serial ports than the HW ones.  Can you post a small MCVE showing the problem?

NollKollTroll, 2024-09-16:
Guessing an MCVE means you want some code, here is what I've used so far.
SerialExtra needs to be Serial2 when using pins 36/37, but still does not work.
I have a simple serial-to-USB converter connected to the RS/TX pins.
Also, thanks for reminding me about the PIO-serial library which will solve my problem without reassigning any pins :D

`#define SerialExtra Serial1

void setup() {
  Serial.begin();

  SerialExtra.setTX(28); // works: 0/28, fails: 32/36
  SerialExtra.setRX(29); // works: 1/29, fails: 33/37
  SerialExtra.begin(115200);
}

void loop() {
  Serial.println("#");
  SerialExtra.println("E");
  delay(1000);
}`

earlephilhower, 2024-09-16:
Just looking at the code, was a simple typo in `#define` checks.  `git master` should be all good now.

commit 1f291482 (2024-09-16): Fix SerialUART RP2350B define checks

Fixes #2460
closed 2024-09-16 by earlephilhower