earlephilhower/arduino-pico #2462: EEPROM.commit without stopping other core

leobel96, 2024-09-16:
Hello, I'm trying to write an application structured in this way:

Core0: continuous generations of sounds
Core1: Doing other stuff, included EEPROM read and write

I noticed that, every time Core1 writes to the EEPROM, the Core0 lags causing a stuttering in the sound being generated. Looking the EEPROM library, it looks like the "problem" is at:
https://github.com/earlephilhower/arduino-pico/blob/1f291482cd01078b5f9f9fc6ad897cc89194a453/libraries/EEPROM/src/EEPROM.cpp#L122
Why is that line needed? Given that Core0 is never writing to the EEPROM, is it actually needed? If not, would it be possible to have an alternative/unsafe implementation of the `commit` method?

earlephilhower, 2024-09-16:
You cannot erase or write flash while running code from it.  The other core must be in a special RAM-only busy loop or it will crash since the flash can't erase and read at the same time.

closed 2024-09-16 by earlephilhower

leobel96, 2024-09-16:
Is there any other way I can achieve a similar result? As in saving to EEPROM without stopping the execution of the other Core

earlephilhower, 2024-09-16:
Use a larger buffer for audio output to give enough samples while stalled or, save your data to a SD card or external physical EEPROM (i.e. not the emulated `EEPROM` here in flash).

earlephilhower, 2024-09-16:
Yes, it's writing to flash to save your data so same issue.  It may be a little worse since there is a filesystem vs. just banging a single flash block.

leobel96, 2024-09-16:
Okay, I'll try exploring some options on my own. Thank you both!