earlephilhower/arduino-pico #2465: Wrong architecture when selecting "Raspberry Pi Pico 2"

jfseaman, 2024-09-16:
Code:
void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}

-------------------------------------------------------------------------
Output:
FQBN: rp2040:rp2040:rpipico2
Using board 'rpipico2' from platform in folder: C:\Users\fseam\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\4.0.2
Using core 'rp2040' from platform in folder: C:\Users\fseam\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\4.0.2

Detecting libraries used...
C:\Users\fseam\AppData\Local\Arduino15\packages\rp2040\tools\pqt-gcc\2.2.0-d04e724/bin/arm-none-eabi-g++ -I C:\Users\fseam\AppData\Local\Temp\arduino\sketches\0FE8079CF6AE71413987353FABBA5723/core -c -Werror=return-type -Wno-psabi -DUSBD_PID=0x000f -DUSBD_VID=0x2e8a -DUSBD_MAX_POWER_MA=250 -DUSB_MANUFACTURER="Raspberry Pi" -DUSB_PRODUCT="Pico 2" -DLWIP_IPV6=0 -DLWIP_IPV4=1 -DLWIP_IGMP=1 -DLWIP_CHECKSUM_CTRL_PER_NETIF=1 -DARDUINO_VARIANT="rpipico2" -DPICO_FLASH_SIZE_BYTES=4194304 @C:\Users\fseam\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\4.0.2/lib/rp2350/platform_def.txt -mcpu=cortex-m33 -mthumb -march=armv8-m.main+fp+dsp -mfloat-abi=softfp -mcmse -ffunction-sections -fdata-sections -fno-exceptions -iprefixC:\Users\fseam\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\4.0.2/ @C:\Users\fseam\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\4.0.2/lib/rp2350/platform_inc.txt @C:\Users\fseam\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\4.0.2/lib/core_inc.txt -IC:\Users\fseam\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\4.0.2/include -fno-rtti -std=gnu++17 -g -pipe -w -x c++ -E -CC -DF_CPU=150000000L -DARDUINO=10607 -DARDUINO_RASPBERRY_PI_PICO_2 -DBOARD_NAME="RASPBERRY_PI_PICO_2" -DARDUINO_ARCH_RP2040 -Os -DWIFICC=CYW43_COUNTRY_WORLDWIDE -IC:\Users\fseam\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\4.0.2\cores\rp2040 -IC:\Users\fseam\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\4.0.2\variants\rpipico2 C:\Users\fseam\AppData\Local\Temp\arduino\sketches\0FE8079CF6AE71413987353FABBA5723\sketch\sketch_sep16a.ino.cpp -o nul
Generating function prototypes...
C:\Users\fseam\AppData\Local\Arduino15\packages\rp2040\tools\pqt-gcc\2.2.0-d04e724/bin/arm-none-eabi-g++ -I C:\Users\fseam\AppData\Local\Temp\arduino\sketches\0FE8079CF6AE71413987353FABBA5723/core -c -Werror=return-type -Wno-psabi -DUSBD_PID=0x000f -DUSBD_VID=0x2e8a -DUSBD_MAX_POWER_MA=250 -DUSB_MANUFACTURER="Raspberry Pi" -DUSB_PRODUCT="Pico 2" -DLWIP_IPV6=0 -DLWIP_IPV4=1 -DLWIP_IGMP=1 -DLWIP_CHECKSUM_CTRL_PER_NETIF=1 -DARDUINO_VARIANT="rpipico2" -DPICO_FLASH_SIZE_BYTES=4194304 @C:\Users\fseam\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\4.0.2/lib/rp2350/platform_def.txt -mcpu=cortex-m33 -mthumb -march=armv8-m.main+fp+dsp -mfloat-abi=softfp -mcmse -ffunction-sections -fdata-sections -fno-exceptions -iprefixC:\Users\fseam\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\4.0.2/ @C:\Users\fseam\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\4.0.2/lib/rp2350/platform_inc.txt @C:\Users\fseam\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\4.0.2/lib/core_inc.txt -IC:\Users\fseam\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\4.0.2/include -fno-rtti -std=gnu++17 -g -pipe -w -x c++ -E -CC -DF_CPU=150000000L -DARDUINO=10607 -DARDUINO_RASPBERRY_PI_PICO_2 -DBOARD_NAME="RASPBERRY_PI_PICO_2" -DARDUINO_ARCH_RP2040 -Os -DWIFICC=CYW43_COUNTRY_WORLDWIDE -IC:\Users\fseam\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\4.0.2\cores\rp2040 -IC:\Users\fseam\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\4.0.2\variants\rpipico2 C:\Users\fseam\AppData\Local\Temp\arduino\sketches\0FE8079CF6AE71413987353FABBA5723\sketch\sketch_sep16a.ino.cpp -o C:\Users\fseam\AppData\Local\Temp\1277755053\sketch_merged.cpp
C:\Users\fseam\AppData\Local\Arduino15\packages\builtin\tools\ctags\5.8-arduino11/ctags -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives C:\Users\fseam\AppData\Local\Temp\1277755053\sketch_merged.cpp
Compiling sketch...
"C:\\Users\\fseam\\AppData\\Local\\Arduino15\\packages\\rp2040\\tools\\pqt-python3\\1.0.1-base-3a57aed/python3" -I "C:\\Users\\fseam\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\4.0.2/tools/signing.py" --mode header --publickey "C:\\Users\\fseam\\AppData\\Local\\Temp\\.arduinoIDE-unsaved2024816-62072-s5bhx.1zmaad\\sketch_sep16a/public.key" --out "C:\\Users\\fseam\\AppData\\Local\\Temp\\arduino\\sketches\\0FE8079CF6AE71413987353FABBA5723/core/Updater_Signing.h"
"C:\\Users\\fseam\\AppData\\Local\\Arduino15\\packages\\rp2040\\tools\\pqt-gcc\\2.2.0-d04e724/bin/arm-none-eabi-g++" -I "C:\\Users\\fseam\\AppData\\Local\\Temp\\arduino\\sketches\\0FE8079CF6AE71413987353FABBA5723/core" -c -Wall -Wextra -Werror=return-type -Wno-ignored-qualifiers -Wno-psabi -Wno-unused-parameter -Wno-missing-field-initializers -DUSBD_PID=0x000f -DUSBD_VID=0x2e8a -DUSBD_MAX_POWER_MA=250 "-DUSB_MANUFACTURER=\"Raspberry Pi\"" "-DUSB_PRODUCT=\"Pico 2\"" -DLWIP_IPV6=0 -DLWIP_IPV4=1 -DLWIP_IGMP=1 -DLWIP_CHECKSUM_CTRL_PER_NETIF=1 "-DARDUINO_VARIANT=\"rpipico2\"" -DPICO_FLASH_SIZE_BYTES=4194304 "@C:\\Users\\fseam\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\4.0.2/lib/rp2350/platform_def.txt" -mcpu=cortex-m33 -mthumb -march=armv8-m.main+fp+dsp -mfloat-abi=softfp -mcmse -ffunction-sections -fdata-sections -fno-exceptions -MMD "-iprefixC:\\Users\\fseam\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\4.0.2/" "@C:\\Users\\fseam\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\4.0.2/lib/rp2350/platform_inc.txt" "@C:\\Users\\fseam\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\4.0.2/lib/core_inc.txt" "-IC:\\Users\\fseam\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\4.0.2/include" -fno-rtti -std=gnu++17 -g -pipe -DF_CPU=150000000L -DARDUINO=10607 -DARDUINO_RASPBERRY_PI_PICO_2 "-DBOARD_NAME=\"RASPBERRY_PI_PICO_2\"" -DARDUINO_ARCH_RP2040 -Os -DWIFICC=CYW43_COUNTRY_WORLDWIDE "-IC:\\Users\\fseam\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\4.0.2\\cores\\rp2040" "-IC:\\Users\\fseam\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\4.0.2\\variants\\rpipico2" "C:\\Users\\fseam\\AppData\\Local\\Temp\\arduino\\sketches\\0FE8079CF6AE71413987353FABBA5723\\sketch\\sketch_sep16a.ino.cpp" -o "C:\\Users\\fseam\\AppData\\Local\\Temp\\arduino\\sketches\\0FE8079CF6AE71413987353FABBA5723\\sketch\\sketch_sep16a.ino.cpp.o"
Compiling libraries...
Compiling core...
Using precompiled core: C:\Users\fseam\AppData\Local\Temp\arduino\cores\dc026870b41e18a2ac980114b5fef9c3\core.a
Linking everything together...
"C:\\Users\\fseam\\AppData\\Local\\Arduino15\\packages\\rp2040\\tools\\pqt-python3\\1.0.1-base-3a57aed/python3" -I "C:\\Users\\fseam\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\4.0.2/tools/simplesub.py" --input "C:\\Users\\fseam\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\4.0.2/lib/rp2350/memmap_default.ld" --out "C:\\Users\\fseam\\AppData\\Local\\Temp\\arduino\\sketches\\0FE8079CF6AE71413987353FABBA5723/memmap_default.ld" --sub __FLASH_LENGTH__ 4186112 --sub __EEPROM_START__ 272621568 --sub __FS_START__ 272621568 --sub __FS_END__ 272621568 --sub __RAM_LENGTH__ 512k --sub __PSRAM_LENGTH__ 0x000000
"C:\\Users\\fseam\\AppData\\Local\\Arduino15\\packages\\rp2040\\tools\\pqt-gcc\\2.2.0-d04e724/bin/arm-none-eabi-gcc" -Werror=return-type -Wno-psabi -DUSBD_PID=0x000f -DUSBD_VID=0x2e8a -DUSBD_MAX_POWER_MA=250 "-DUSB_MANUFACTURER=\"Raspberry Pi\"" "-DUSB_PRODUCT=\"Pico 2\"" -DLWIP_IPV6=0 -DLWIP_IPV4=1 -DLWIP_IGMP=1 -DLWIP_CHECKSUM_CTRL_PER_NETIF=1 "-DARDUINO_VARIANT=\"rpipico2\"" -DPICO_FLASH_SIZE_BYTES=4194304 "@C:\\Users\\fseam\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\4.0.2/lib/rp2350/platform_def.txt" -mcpu=cortex-m33 -mthumb -march=armv8-m.main+fp+dsp -mfloat-abi=softfp -mcmse -ffunction-sections -fdata-sections -fno-exceptions -Os -u _printf_float -u _scanf_float -c "C:\\Users\\fseam\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\4.0.2/boot2/rp2350/boot2_generic_03h_2_padded_checksum.S" "-IC:\\Users\\fseam\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\4.0.2/pico-sdk/src/rp2350/hardware_regs/include/" "-IC:\\Users\\fseam\\AppData\\Local\\Arduino15\\packages\\rp2040\\hardware\\rp2040\\4.0.2/pico-sdk/src/common/pico_binary_info/include" -o "C:\\Users\\fseam\\AppData\\Loc [... cut ...]
"C:\\Users\\fseam\\AppData\\Local\\Arduino15\\packages\\rp2040\\tools\\pqt-gcc\\2.2.0-d04e724/bin/arm-none-eabi-size" -A "C:\\Users\\fseam\\AppData\\Local\\Temp\\arduino\\sketches\\0FE8079CF6AE71413987353FABBA5723/sketch_sep16a.ino.elf"
Sketch uses 50008 bytes (1%) of program storage space. Maximum is 4186112 bytes.
Global variables use 9736 bytes (1%) of dynamic memory, leaving 514552 bytes for local variables. Maximum is 524288 bytes.

--------------------------------------------------------------
Settiings

[image]

jfseaman, 2024-09-16:
But the code works. I'm confused....

and much faster than RP2040. At 150Mhz RP2350 is faster than RP2040 at 276Mhz (overclocked)

earlephilhower, 2024-09-16:
Sorry, not seeing anything wrong.  That's the right compiler architecture for the Pico 2, and it's an RP2350 M33 dual core which is faster clock-to-clock than the RP2040 dual M0 on the old Pico.

jfseaman, 2024-09-17:
All the printed text says RP2040 but it's compiling for RP2350 M33 

Just label confusion.

jfseaman, 2024-09-17:
I've been a programmer for over 50 years. 

C/C++ programmer for 40 years.

Details like text returned from compilers is important. 

But in this case, seeing RP2040 where I expect RP2350 seems fine.