earlephilhower/arduino-pico #2467: 4-bit SD / SDIO Support

DatanoiseTV, 2024-09-17:
Hello,

I would love to see 4-bit SD / SDIO support. According to [this](https://github.com/juico/pico-sdio-example) repository, reads up to 26MB/s would be possible.

DatanoiseTV, 2024-09-17:
On the PicoADK v2 (RP2350), due to pin usage for other components, the SPI pins are not available, so the SD Card needs to be handled using PIO, so using SDIO would be good.

[image]

earlephilhower, 2024-09-17:
You've got HW and some sample code, so I'll pencil in a PR from you later this month, right? :laughing:

Seriously, though, my reading of that thread when it came out was that the performance was great, but there were some timing gotchas that needed individual attention and everyone got busy on other things.  If you do make a PR, then I'd recommend allowing GPIO strengths/delays to be fine tuned thru `#define`s so individual boards can make it happen.

I'm not sure if there's a reason for SDIO other than SD cards, but if not then folding it into the `grieman` SDFat seems like the simplest thing to do.

DatanoiseTV, 2024-09-17:
Some benchmarking (using 200 MHz CPU Clock):

```
> bench
Type is exFAT
Card size: 127.93 GB (GB = 1E9 bytes)

Manufacturer ID: 0x3
OEM ID: SDProduct: SD128
Revision: 8.5
Serial number: 0xfeb5bf29
Manufacturing date: 10/2023

FILE_SIZE_MB = 128
BUF_SIZE = 65536
Starting write test, please wait.

write speed and latency
speed,max,min,avg
KB/Sec,usec,usec,usec
9483.3,28165,6552,6909
9690.1,30408,6551,6762
9482.7,30455,6551,6910
```

using following HW config:
```

#include "hw_config.h"

/* SDIO Interface */
static sd_sdio_if_t sdio_if = {
    /*
    Pins CLK_gpio, D1_gpio, D2_gpio, and D3_gpio are at offsets from pin D0_gpio.
    The offsets are determined by sd_driver\SDIO\rp2040_sdio.pio.
        CLK_gpio = (D0_gpio + SDIO_CLK_PIN_D0_OFFSET) % 32;
        As of this writing, SDIO_CLK_PIN_D0_OFFSET is 30,
            which is -2 in mod32 arithmetic, so:
        CLK_gpio = D0_gpio -2.
        D1_gpio = D0_gpio + 1;
        D2_gpio = D0_gpio + 2;
        D3_gpio = D0_gpio + 3;
    */

    .CLK_gpio_drive_strength = GPIO_DRIVE_STRENGTH_12MA,
    //.CLK_gpio_drive_strength = GPIO_DRIVE_STRENGTH_4MA,
    .CMD_gpio_drive_strength = GPIO_DRIVE_STRENGTH_4MA,
    .D0_gpio_drive_strength = GPIO_DRIVE_STRENGTH_4MA,
    .D1_gpio_drive_strength = GPIO_DRIVE_STRENGTH_4MA,
    .D2_gpio_drive_strength = GPIO_DRIVE_STRENGTH_4MA,
    .D3_gpio_drive_strength = GPIO_DRIVE_STRENGTH_4MA,
    .SDIO_PIO = pio1,
    .DMA_IRQ_num = DMA_IRQ_0,

    .CMD_gpio = 21,
    .D0_gpio = 22,
    .baud_rate = 21 * 1000 * 1000
};

/* Hardware Configuration of the SD Card socket "object" */
static sd_card_t sd_card = {
    .type = SD_IF_SDIO,
    .sdio_if_p = &sdio_if};

/* Callbacks used by the library: */
size_t sd_get_num() { return 1; }

sd_card_t *sd_get_by_num(size_t num)
{
    if (0 == num)
        return &sd_card;
    else
        return NULL;
}
/* [] END OF FILE */
```

DatanoiseTV, 2024-09-19:
[This](https://github.com/earlephilhower/arduino-pico/discussions/1560#discussioncomment-10681395) solves it and gives 26MB/s.

closed 2024-09-20 by DatanoiseTV