earlephilhower/arduino-pico #2484: FreeRTOS SMP Support and wrappers

DatanoiseTV, 2024-09-20:
While porting some ESP32 Audio Code, I was wondering if arduino-pico is using an SMP version of the FreeRTOS kernel.

Also, Espressif made some own non-standard wrappers for various functions, such as:
```
xTaskCreatePinnedToCore( task1, "Task1", 8000, NULL, (1 | portPRIVILEGE_BIT), &Task1, 0 );
portENTER_CRITICAL_ISR(x)...
portEXIT_CRITICAL_ISR(x)...
```

I think it would be practical if these were implemented as well, so code could be ported easier. I have already started writing some wrappers for other things like PSRAM allocation etc.

DatanoiseTV, 2024-09-20:
After some digging, I found out that indeed both cores are supported already.

Also found how to set core affinity, so I guess I will just write some wrappers around it.
 vTaskCoreAffinitySet(loop2Handle, CORE_0);

earlephilhower, 2024-09-20:
Yeah, we were SMP before it was cool, man!  :laughing:

The Espressif guys were there even earlier, probably one of the first folks doing real SMP work w/FreeRTOS.  They probably needed to write lots of these little helpers like your header to get things up to speed back in the day.

I'm not really keen on tracking their changes.  If they're just syntactic sugar there's a simple compatible way of doing the call which can be used by everyone.  If they're modifying the upstream FreeRTOS functionality/guts then it's on them to propose those changes upstream or maintain their own branch very separate from RPI's/ours.

closed 2024-09-26 by earlephilhower