earlephilhower/arduino-pico #2485: lfs_flash_erase hangs on idleOtherCore with FreeRTOS

Yveaux, 2024-09-20:
Formatting LittleFS hangs when FreeRTOS is enabled:

platformio.ini
```
[env]
; Created a fork of platform-raspberrypi just to get FreeRTOS support for arduino-pico
platform = https://github.com/Yveaux/platform-raspberrypi.git
framework = arduino
board = rpipico2
board_build.core = earlephilhower
board_build.filesystem = littlefs
board_build.filesystem_size = 0.5m
upload_protocol = picoprobe
debug_tool = picoprobe
debug_speed = 20000

debug_build_flags =
  -O0
  -g
  -ggdb3
```

main.cpp:
```
#include <Arduino.h>
#include <FreeRTOS.h>
#include "LittleFS.h"

void setup()
{
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, LOW);
  LittleFS.format();
}

void setup1()
{
}

void loop()
{
  digitalWrite(LED_BUILTIN, HIGH);
  delay(200);
}

void loop1()
{
  delay(100);
  digitalWrite(LED_BUILTIN, LOW);
  delay(100);
}
```

Running this code will hang on https://github.com/earlephilhower/arduino-pico/blob/b2ec6ecb09bd68ab8631b34a1cb327508d18c260/libraries/LittleFS/src/LittleFS.cpp#L199

while when removing `#include <FreeRTOS.h>` (and thereby disabling FreeRTOS), the `LittleFS.format()` call completes succesfully.

Verified with debugger and easily visble from the pico2 onboard led too.

earlephilhower, 2024-09-20:
Thanks for the concise bug and MCVE.  FreeRTOS SMP really does not want to be frozen out (and you can't actually stop the SYSTICK exceptions that happen @ 1KHz).

Yveaux, 2024-09-20:
Wow, thanks for the incredibly fast fix! Really appreciate it!

PR #2486 (2024-09-20): Fix FreeRTOS Flash Freeze, Fastly
Fixes #2485
commit a6ab6e1f (2024-09-20): Fix FreeRTOS Flash Freeze, Fastly (#2486)

Fixes #2485
closed 2024-09-20 by earlephilhower