earlephilhower/arduino-pico #2490: I2S SimpleTone Example not working.

ullibak, 2024-09-23:
Hi

I use a A2DP sink device programmed with an earlier version of the Pico SDK and an earlier version of the board implementation.
Now, I had to change some thongs in the program.  I am using arduino-pico version 4.0.2 and can get not audio output anymore.

To investigate further, I tired to run the SimpleTone I2S example supplied with the board implementation and I get the same error messages. Also, there is no sound coming out of my DAC. 

I adapted the example to my pinout:
```c
// GPIO pin numbers
#define pBCLK 10
#define pWS (pBCLK+1)
#define pDOUT 9 
```  

Then, I added
while(!Serial); after the line Serial.begin(115200) to avoid losing any error messages.

When starting up the device, I get this output:

13:06:57.579 -> I2S simple tone
13:06:57.579 -> PIOProgram 0x100200f0: Searching for high=0, pins 9-11
13:06:57.579 -> PIOProgram: Skipping PIO 0x50200000, wrong allocated/needhi
13:06:57.579 -> PIOProgram: Skipping PIO 0x50300000, wrong allocated/needhi
13:06:57.579 -> PIOProgram 0x100200f0: Allocating new PIO 0x50300000(high=0) for pins 9-11

The last line suggests that a new program has been allocated to the right pins, but there is no output. Is there something that I overlooked or is there something wrong with the PIO program that runs I2S?

Thanks!

earlephilhower, 2024-09-23:
Sorry, I can't reproduce any problem with the example using an RP2040 and `git master`.  CLK, LR, and DATA all shift out when captured by pulseview:
[image]

There were minor RP2350-only fixes after 4.0.2 (related to the 3rd PIO), but since you're talking about A2DP then I imagine you're still on the 2040 which didn't really have any changes.

Can you capture some waveforms from the sample and verify (GP20-22, different PCB pins depending on board of course)?

ullibak, 2024-09-23:
Thanks. Will try tomorrow on a fresh installation...

ullibak, 2024-09-24:
Tried a fresh installation in a Windows sandbox. Everything worked fine. 
Then, on my normal setup, I deleted the entire package directory and re-installed version 4.0.2.
Now it worked. Obviously, something got messed up on my computer. 

Thanks and sorry for bothering you. Have a nice day!

closed 2024-09-24 by ullibak