earlephilhower/arduino-pico #2494: ADC/DMA + PWMAudio Problem

MuhammedSaygili, 2024-09-26:
I am able to perform sampling with ADC and DMA, but when I try to generate sound with PWMAudio at a different time outside of sampling, a problem occurs. I am also using Wifi and AsyncUDP, but even after turning them off and completely deleting all the code, the problem persists. In summary, Wifi and ADC + DMA work without issues, and Wifi and PWMAudio also work without problems, but ADC + DMA and PWMAudio do not work together. When I try to run it, the code after `pwm.begin(samplerate)` does not execute. I wanted to debug this and found that nothing after the line` if (!_arb->begin(_pacer_dreq, (volatile void*)ccAddr)) { `inside begin is executed.

**ADC and DAC Configuration:
  Before the Setup function:**
```c++

uint16_t buffer[BUFFER_SIZE];
volatile bool dma_complete = false;
PWMAudio pwm(10);

int dma_chan = dma_claim_unused_channel(true);
dma_channel_config c = dma_channel_get_default_config(dma_chan);

void dma_handler() {
    dma_hw->ints0 = 1u << 0;
    dma_complete = true;
}

void restartDma(){
    dma_channel_configure(
        dma_chan,
        &c,
        buffer,        // Destination pointer
        &adc_hw->fifo, // Source pointer
        BUFFER_SIZE,   // Number of transfers
        true           // Start immediately!!
    );
    dma_complete = false;
}
```

**Setup Function:**
```c++
void setup(){
  Serial.begin(115200);
  setupUDP(); 
  Serial.println("UDP Setup Complete”);
  
  adc_init();
  adc_gpio_init(26); //ADC input GPIO 26 
  adc_select_input(0);
  Serial.println("ADC Init Complete");
  adc_set_clkdiv(48e6/SAMPLE_RATE);
  channel_config_set_transfer_data_size(&c, DMA_SIZE_16);
  channel_config_set_read_increment(&c, false);
  channel_config_set_write_increment(&c, true);
  channel_config_set_dreq(&c, DREQ_ADC);
  
  dma_channel_configure(
      dma_chan,
      &c,
      buffer,
      &adc_hw->fifo,
      BUFFER_SIZE,
      false 
  );
  Serial.println("DMA Setup Complete");
  
  irq_set_exclusive_handler(DMA_IRQ_0, dma_handler);
  Serial.println("irq1");
  irq_set_enabled(DMA_IRQ_0, true);
  Serial.println("irq2");
  dma_channel_set_irq0_enabled(dma_chan, true);
  Serial.println("DMA IRQ Setup Complete");
  
  adc_fifo_setup(true, true, 1, false, true);
  adc_run(true);
  dma_start_channel_mask(1u << dma_chan);
  while(!dma_complete){
  }
  restartDma();
  
  pwm.begin(10000); 
  Serial.print("setup finish");
}
```

MuhammedSaygili, 2024-09-26:
First, I suspected the dreq channels, but I believe the issue is caused by both sides using the same DMA_IRQ. When I examined `AudioBufferManager.cpp`, I saw that there is the same interrupt flag. I haven't tested the functions yet, but I will note here whether a conflict occurs during operation.

But currently, the processor does not hang and the setup completes successfully.

earlephilhower, 2024-09-26:
You're mixing calls to the raw SDK for low level access and the core.  If you can repro this with only the core, then we can look at it.  OTW, you may want to use the R.PI forums.

>   irq_set_exclusive_handler(DMA_IRQ_0, dma_handler);

That's not legal, the core uses that IRQ and you've just made the SDK `panic` when it tries to use it...