earlephilhower/arduino-pico #2495: Feature Request: Support for new Pimoroni Pico Plus 2 W

znmeb, 2024-09-26:
Pimoroni have announced a new board with an RP2350 and WiFi. I have two on order.

https://shop.pimoroni.com/products/pimoroni-pico-plus-2-w?variant=42182811942995

So far the only software support I see for this is the Pimoroni build of MicroPython - I don't even know if it's in the Raspberry Pi Pico SDK yet. I'll update this issue when I get boards and can throw my test scripts at it.

earlephilhower, 2024-09-27:
Looks like it's just the existing WiFi chip on a carrier for easier FCC certification:  https://www.tomshardware.com/raspberry-pi/pimoroni-pico-plus-2-w-features-mysterious-raspberry-pi-rm2-wi-fi-chip

znmeb, 2024-09-27:
Interesting - my boards have shipped and DHL sez they'll be here Monday. The Pimoroni website only calls out MicroPython and the Tom's Hardware article seems to back that up so I'll load the Pimoroni MicroPython on one board to verify that the WiFi works. Then I can look at the MicroPython code in their repo to see if it has any differences between their new board and the Pico W.

sillycowvalley, 2024-10-15:
Have 2x Pico Plus 2 W's now.
Tried and failed to get it to compile in the Arduino IDE:
Basically used Pico Plus 2 W as my board and added:
`#if defined(ARDUINO_RASPBERRY_PI_PICO_W) || defined(ARDUINO_RASPBERRY_PI_PICO_2)`
everywhere in `WifiClass.h` and `WifiClass.cpp` to make the Pico Plus 2 config pretend to be a Pico Plus 2 W (for now) and just use the same code as the old Pico W.
Looks like I'm missing adding `libraries\lwIP_CYW43` (etc) to the include paths somewhere ...
Clearly I'm out of my depth but I'm happy to test someone elses' attempt at this.

znmeb, 2024-10-15:
I'm not sure Pimoroni even have it working with MicroPython yet. <https://github.com/pimoroni/pimoroni-pico-rp2350/releases/tag/v0.0.7> And I don't think there's a board definition for it in the SDK either. There's no header file for it in <https://github.com/raspberrypi/pico-sdk/tree/master/src/boards/include/boards>. I'm going to open an issue over on the SDK site to have them add it.

earlephilhower, 2024-11-15:
Tracked under #2608.  Might end up building CYW43 libs for all boards defined in the SDK w/the chip if the pinouts differ...

closed 2024-11-15 by earlephilhower