earlephilhower/arduino-pico #2502: newlib update broken builds - how to pin toolchain version?

JonnyHaystack, 2024-09-28:
It seems like since this change to the toolchain https://github.com/earlephilhower/pico-quick-toolchain/pull/63, my builds have been broken (see job logs [here](https://pastebin.com/ShJQcptw)).

I have the core version pinned like so:
```
platform_packages =
    framework-arduinopico@https://github.com/earlephilhower/arduino-pico.git#3.6.3
```
This is because upstream changes (presumably in the Adafruit TinyUSB library) broke my [XInput code](https://github.com/JonnyHaystack/Adafruit_TinyUSB_XInput).

Anyways, so I had thought that pinning the core version also pinned the toolchain version, but that now appears to not be the case. Can the toolchain be explicitly pinned to a specific version instead? I would very much like my builds to be completely reproducible with all dependencies pinned to known working versions.

Thanks

maxgerhardt, 2024-09-29:
>Anyways, so I had thought that pinning the core version also pinned the toolchain version, but that now appears to not be the case. 

Correct. The platform picks the used tools, not the framework. They always have to be in sync.

Apply the fix from  https://github.com/maxgerhardt/platform-raspberrypi/issues/75#issuecomment-2381371700
```
platform = https://github.com/maxgerhardt/platform-raspberrypi.git#5e87ae34ca025274df25b3303e9e9cb6c120123c
```

JonnyHaystack, 2024-09-29:
Ah, that's good to know. This has solved my problem. Thanks as always, Max 🙂

closed 2024-09-29 by JonnyHaystack