earlephilhower/arduino-pico #2503: New version pqt-picotool 2.3.0 crushes Windows 7 through the COM port driver

Nicks7777, 2024-09-30:
After upgrading pqt-picotool from 2.2.0 to 2.3.0, the following happens:
- download any sketch with USB Stack "Pico SDK" installed
- the board reboots
- after that, when a COM port is detected, Windows 7 crashes into BSOD
- BAD_POOL_CALLER 0xC2, Parameter 1 = 0x7: attempt to free memory that is already free
- BSOD calls Windows driver usbser.sys installed on this COM port

If you use "Adafruit TinyUSB" instead of "Pico SDK" or replace pqt-picotool with the old version 2.2.0, then everything works fine.

earlephilhower, 2024-09-30:
Sorry, but Windows 7 went EOL almost 5 years ago so we're not going to be able to do anything on this end.  Having a USB device that can BSOD it instantly seems to be a problem with the OS, not the device...

You can try changing to Tools->Upload Method->Default (UF2) or keep running the older version of the core.

closed 2024-09-30 by earlephilhower

maxgerhardt, 2024-09-30:
>If you use "Adafruit TinyUSB" instead of "Pico SDK" or replace pqt-picotool with the old version 2.2.0, then everything works fine.

You could try to replicate this by compiling the native Pico-SDK USB hello world example from https://github.com/raspberrypi/pico-examples/blob/master/hello_world/usb/ and if the resulting binary still crashed Windows 7, you can get support from https://github.com/raspberrypi/pico-sdk/issues.

earlephilhower, 2024-10-01:
When built using the `picotool` upload method, a secondary vendor-specific USB interface is created on the RP2040 to support the special picotool reset tickle.  I suppose Windows could lose its mind if it didn't know what to make of that interface (it should ignore it, but who knows for Win 7).

By building and uploading UF2 versions of an app, only the serial CDC USB port will be created by the app (i.e. only CDC descriptor, not a combined CDC + Picotool interface one).  So it *might* help here without needing to dig into the SDK.

Nicks7777, 2024-10-01:
The loading process through the UF2 is extremely slow and clumsy. Therefore, for myself, I use loading via a CMSIS-DAP adapter, use UART (0/1) for debugging and completely abandoned the USB Pico SDK.

By the way, the latest release of pqt-picotool 2.0.0 (https://github.com/raspberrypi/pico-sdk-tools/releases/tag/v2.0.0-5) from raspberrypi causes the same error. But the Load command is already working in it.