earlephilhower/arduino-pico #2508: Is there a conflict between adc and bluetooth or more particularly ADCInput and SerialBT

crp500, 2024-10-01:
I wrote a simple oscilloscope program using ADCInput and serial over usb and had no problems receiving the data.
I then tried to send the data using serial over bluetooth with the SerailBT library.
I found that simply turning on the ADCInput object (at a frequency of 8000 originally) really screwed with the bluetooth to such and extent that I could no longer discover the device. Turning off the ADCInput object let me discover the device again.
I tried using analogRead instead of the ADCInput library and that sort of worked however the data rate was to low for me to use.

Is the some sort of connection between bluetooth/Wifi and ADC on the pico? 

Here is a simple sketch that shows what I am trying to do. 

//--------------------------------------------------

#include <ADCInput.h>
#include <SerialBT.h>

ADCInput adc(A0, A1, A2, A3);

void setup() {
  SerialBT.setName("PicoW SerialBT 00:00:00:00:00:00");
  SerialBT.begin();  
  delay(2000);
  adc.begin(8000);
}

void loop() {
  uint8_t buf[6];
  int r[3];
  r[0] = adc.read();
  r[1] = adc.read();
  r[2] = adc.read();
  adc.read();
  buf[0] = highByte(r[0]);
  buf[1] = lowByte(r[0]);
  buf[2] = highByte(r[1]);
  buf[3] = lowByte(r[1]);
  buf[4] = highByte(r[2]);
  buf[5] = lowByte(r[2]);

  if(SerialBT.availableForWrite()) {
    SerialBT.write(buf, 6);
    SerialBT.flush();
  }
}

earlephilhower, 2024-10-01:
Sorry, I can't reproduce that here using the [example in the 2nd comment](https://github.com/earlephilhower/arduino-pico/issues/2508#issuecomment-2385318885).  I can see the BT classic serial port on my phone and PC's USB BT adapter.

There should be no interaction between BT and ADC.  The ADC is only a periodic DMA + IRQ every buffer so there will be some CPU load but that's about it.

BT is mostly handled on the CYW chip over a PIO-SPI interface.  You might have more ADC noise when Wifi/BT is running, but that's about it.  The HCI interface does have some CPU load, but again something simple like just advertising is almost nothing.

crp500, 2024-10-01:
Thanks for looking at this Earle.
I didn't think of using my phone. I can see the pico on my phone and occasionally on my ubuntu laptop however I can't pair to it on either device. Did you try to pair ? If I comment out the adc part of the script I have no problem discovering or pairing.
I hope this is just me being dumb. It usually is.

earlephilhower, 2024-10-02:
The ADC DMA is eating too much CPU when combined with the Bluetooth stack.  You can increase the size of buffers by adding a `adc.setBuffers(8, 256)` which will reduce the pressure somewhat at the price of increased latency.  Each sample will take 4 words of space since you're recording 4 ADCs, so it'll go through the default size very quickly.  The BT stack and driver also needs CPU to work and disables IRQs in certain critical sections, eating into total available time.  (I'm also not sure the Pico's BT stack can successfully send data as quickly as you're trying to do, (~96KBaud) especially with the `flush` requiring transmission before continuing.)

A simple test dropping sampling rate to 100hz runs fine even on my phone with whatever BT serial terminal I ended up installing.

You can drop the sampling rate down, send fewer bytes, or look into doing multicore processing to run ADC DMA IRQs on core 1 with core 0 only doing BT.  There's unfortunately not an interaction other than out of CPU.  Maybe a better BT profile than Serial would help.

crp500, 2024-10-02:
Im not having any luck.
I tried another pico w without success.
I also tried this sketch which works for a short time and then stops.
If I comment out the adc.begin() line it works perfectly and seemingly forwever.

//---------------------------------------------

#include <ADCInput.h>
#include <SerialBT.h>

ADCInput adc(A0);

void setup() {
  SerialBT.setName("PicoW SerialBT 00:00:00:00:00:00");
  SerialBT.begin();  
}

void loop() {
  if(SerialBT.availableForWrite()) {
    SerialBT.println("hello world");
  }
}

void setup1() {
  adc.begin(100);
}

void loop1() {
  if(adc.available()) {
    adc.read();
  }
}

earlephilhower, 2024-10-02:
Interesting.  I didn't try on the 2nd core (but what you're doing should be OK since all `adc` accesses are from core 1).

Can you add in the buffer settings?  I came from the ESP8266 world where every byte was crucial and chose very conservative default buffers for the `AudioBufferManager`.  
````
adc.setBuffers(8, 256) ;
````

You can also get a dump of the 2 cores' states w/GDB when things seem to be frozen.  If it's a double-IRQ-miss (which is fatal) then that would indicate the BT subsystem has a really extended period w/IRQs locked..

earlephilhower, 2024-10-02:
A quick SO search seems to indicate the max serial port profile (SPP) baud is 128Kbit = 16KByte/sec which is 1 channel of your 16bit data @ 8KHZ.  A2DP uses a different protocol completely at a lower level (and massive compression) to get higher throughput, but I doubt that'd be useful for your app...

earlephilhower, 2024-10-03:
Enabling Serial debug on with Bluetooth debugging on gave the following interesting output from the BTStack driver and CYW43 chip:
````
....
[BT @17425] CMD => 35 0C 05 01 0B 00 01 00 
[BT @17426] LOG -- hci.c.666: ACL classic buffers: 3 used of 3
17.426: cybt_hci_read_packet: too much data len 598027
cyw43_bluetooth_hci_read: failed to read from shared bus
17.618: cybt_hci_read_packet: too much data len 4325381
cyw43_bluetooth_hci_read: failed to read from shared bus
22.211: cybt_hci_read_packet: too much data len 130827
cyw43_bluetooth_hci_read: failed to read from shared bus
````

It seems like the whole CYW43 chip has lost its mind, which seems to happen a lot with the BT on this board. :cry:   I've also seen crashes inside the CYW43 driver HCI transport using GDB under heavy loads.  And in a couple instances I saw it disconnect but not send the proper event back to the app, leaving the SerialBT object waiting on a write that will never complete.

There's nothing we can do here about any of those, unfortunately, so I would suggest chalking this up to experience and sticking with USB for high throughput transfers...

closed 2024-10-03 by earlephilhower