earlephilhower/arduino-pico #2511: address-of-packed-member warning generated when building

misaka19465, 2024-10-02:
When building my rp2040 project, I always get warning below:
```
/home/misaka19465/.platformio/packages/framework-arduinopico/cores/rp2040/RP2040Support.h: In member function 'void _MFIFO::begin(int)':
/home/misaka19465/.platformio/packages/framework-arduinopico/cores/rp2040/RP2040Support.h:53:20: warning: taking address of packed member of '_MFIFO' may result in an unaligned pointer value [-Waddress-of-packed-member]
   53 |         mutex_init(&_idleMutex);
      |                    ^~~~~~~~~~~
/home/misaka19465/.platformio/packages/framework-arduinopico/cores/rp2040/RP2040Support.h: In member function 'void _MFIFO::idleOtherCore()':
/home/misaka19465/.platformio/packages/framework-arduinopico/cores/rp2040/RP2040Support.h:99:34: warning: taking address of packed member of '_MFIFO' may result in an unaligned pointer value [-Waddress-of-packed-member]
   99 |             mutex_enter_blocking(&_idleMutex);
      |                                  ^~~~~~~~~~~
/home/misaka19465/.platformio/packages/framework-arduinopico/cores/rp2040/RP2040Support.h: In member function 'void _MFIFO::resumeOtherCore()':
/home/misaka19465/.platformio/packages/framework-arduinopico/cores/rp2040/RP2040Support.h:110:20: warning: taking address of packed member of '_MFIFO' may result in an unaligned pointer value [-Waddress-of-packed-member]
  110 |         mutex_exit(&_idleMutex);
```

earlephilhower, 2024-10-02:
Can you give a sample .CPP that causes this?  We build `-Werror` in CI so this is unexpected.  And I'm not seeing an `__attribute__((packed))` in the class definition...
https://github.com/earlephilhower/arduino-pico/blob/690293e7e1696c313714c3e5a15212e46191d090/cores/rp2040/RP2040Support.h#L154-L158

misaka19465, 2024-10-02:
After some test, I noticed `#pragma pack(1)` would cause this. But in my code, it is necessary to make `union` work as expected. So I still think it is something we should solve.

misaka19465, 2024-10-02:
By the way, [this question](https://stackoverflow.com/questions/73152859) explained it well.

earlephilhower, 2024-10-02:
There's nothing to solve, though, because it's an option you're setting manually in your code.  Setting that pragma means that the linker could align a pointer's address to a non-by-4 offset, which is going to go boom.  You're also going to change the structures used for certain precompiled API calls, possibly, which is going to cause some amazing weird errors.

If you've got some data structure in your own code you need packing on, there are pragmas to go back to normal packing after the definition.

closed 2024-10-02 by earlephilhower