earlephilhower/arduino-pico #2512: RP2350 PSRAM compile linker error

justin-luoma, 2024-10-02:
When compiling the PSRAMTest.ino example for the Adafruit Feather RP2350 HSTX the following linker error occurs.

```
/home/user/.arduino15/packages/rp2040/tools/pqt-gcc/2.3.0-dfd82b2/bin/../lib/gcc/arm-none-eabi/12.4.0/../../../../arm-none-eabi/bin/ld: /tmp/arduino/sketches/A56EB97C057FD169A75D23E89722D410/PSRAMTest.ino.elf section `.psram' will not fit in region `PSRAM'
/home/user/.arduino15/packages/rp2040/tools/pqt-gcc/2.3.0-dfd82b2/bin/../lib/gcc/arm-none-eabi/12.4.0/../../../../arm-none-eabi/bin/ld: region `PSRAM' overflowed by 1048576 bytes
collect2: error: ld returned 1 exit status

exit status 1

Compilation error: exit status 1
```

earlephilhower, 2024-10-02:
The board definition for the HSTX says it has no PSRAM, so this is expected (2nd from last parameter):
https://github.com/earlephilhower/arduino-pico/blob/690293e7e1696c313714c3e5a15212e46191d090/tools/makeboards.py#L492

If the boards all do have PSRAM, then if you could get the size and make a PR that would be great.  OTW, it would be expected not to fit 1MB in 0MB of space. :)

earlephilhower, 2024-10-02:
Quick check here https://www.adafruit.com/product/6000 shows no PSRAM from the factory.  You can hack your board and solder on a chip/jumpers, but in that case you need to use the `Generic RP2350` board to ensure the PSRAM_CS pin/etc. are set to match.

closed 2024-10-02 by earlephilhower

maxgerhardt, 2024-10-04:
The product page says it has unpopulated PSRAM here: 

https://learn.adafruit.com/adafruit-feather-rp2350/pinouts#optional-psram-3182502

Mhm, maybe we can still have the menu for selectable PSRAM size for `adafruit_feather_rp2350_hstx` like the Generic one does? With just the default being "0 MB".

earlephilhower, 2024-10-05:
Good and simple idea.  Probably won't be used by anyone (looks like a tight solder job...no coffee that morning! :laughing: ) but the CS and frequency were already in the variant so very minor change.