earlephilhower/arduino-pico #2518: Newest release of toolchain made Pico W project unbuildable

mwedmark, 2024-10-04:
Error about ipstack not an avaiable option in platform file.. Reverting from 4.1.1 to 4.1.0. sand it works again.

earlephilhower, 2024-10-04:
Sorry, you will need to provide a small sample code and output because we build for the PicoW in CI and no issues are seen.

The only difference from 4.1.0->4.1.1 was a minor addition of RP2350B pins to ADCInput and a fix to the platform.txt to allow the IDE to build RISCV.  Neither of which are used for the 2040 Pico W....

maxgerhardt, 2024-10-05:
Not sure what the exact issue is, but the "menu.ipstack" was renamed to "menu.ipbtstack" already more than **a year** ago, so it for sure can't have been broken between 4.1.0 and 4.1.1: https://github.com/earlephilhower/arduino-pico/commit/d92c1025baeadf4632eb7e3b14ba18d50454706b#diff-26e3bc544a9f6f7eaa298789ecac31a971ecd5e663dfdc6c5abc7709da38cb56

More likely the platform update resulted in a broken install? Try a clean remove -> install of the platform again.

earlephilhower, 2024-10-06:
Closing for now.  If needed info gets added we can look at it.

closed 2024-10-06 by earlephilhower

HamzaHajeir, 2024-10-08:
Here might be a similar issue? 

https://github.com/maxgerhardt/platform-raspberrypi/issues/82