earlephilhower/arduino-pico #2522: Best way to test for CPU Architecture selection?

technoblogy, 2024-10-05:
Thank you for the new **CPU Architecture** option allowing you to select **ARM** or **RISC-V** in version 4.1.0 of the core.

What is the recommended way to test for the option that's been selected, both at compile time (such as with `#if defined`) and at run time? I checked the documentation but couldn't find anything relevant.

Thanks, David

earlephilhower, 2024-10-05:
I use the compiler defined `__riscv`.  There is an ARM equivalent, but since it's one or the other you can just use
````
#ifdef __riscv
....r5 specific...
#else
...arm specific...
#endif
````

At run time, just use the defines to set some constant to true or false...

closed 2024-10-05 by earlephilhower

technoblogy, 2024-10-05:
Thank you - perfect!
David