earlephilhower/arduino-pico #2528: Feature Request: support deep sleep for the RP2350

sillycowvalley, 2024-10-07:
I wouldn't know where to start so I asked the One Who Is Wrong With Confidence (ChatGPT) for an example and got the example below as a starting point. Would we want to add some RISC-V specific functionality to the RP2040 class for this?

---

Here's an example of how to put the Pico 2 into deep sleep and wake it up using a GPIO pin (You can also use the RTC to wake it up after a specific time interval):

### C++ Example for Deep Sleep on RP2350 (Pico 2)

```cpp
#include "pico/stdlib.h"
#include "hardware/gpio.h"
#include "hardware/clocks.h"
#include "hardware/vreg.h"
#include "hardware/regs/io_bank0.h"

// Set the GPIO pin to wake the system
#define WAKE_PIN 15

void setup_wake_pin() {
    // Enable the wake pin
    gpio_set_dir(WAKE_PIN, GPIO_IN);
    gpio_pull_down(WAKE_PIN);

    // Enable wakeup by pin
    gpio_set_irq_enabled(WAKE_PIN, GPIO_IRQ_EDGE_RISE, true);
}

int main() {
    // Optional: Lower the voltage to reduce power consumption in sleep
    vreg_set_voltage(VREG_VOLTAGE_0_90);

    // Initialize stdio, not needed for actual deep sleep but for testing/debugging
    stdio_init_all();

    // Configure the GPIO pin for waking up
    setup_wake_pin();

    // Print message before going to deep sleep
    printf("Going to sleep now...\n");

    // Enable deep sleep (Note: the Pico 2 allows real deep sleep with RTC wakeup)
    sleep_goto_dormant();

    // This point will only be reached when the system is woken up
    printf("Woke up from sleep!\n");

    return 0;
}
```

### Key Steps in the Example:
1. **GPIO Wake Pin**: Set a GPIO pin (e.g., pin 15) to wake up the device from deep sleep when a rising edge is detected.
2. **Voltage Reduction**: Optionally, you can reduce the core voltage to 0.9V using `vreg_set_voltage()` to further lower power consumption.
3. **GPIO Interrupts**: The code configures the GPIO interrupts to handle the wake-up.
4. **sleep_goto_dormant()**: This is the key function that puts the Pico into deep sleep, cutting power to most peripherals while maintaining the ability to wake up via external signals.

This example ensures the Pico 2 goes into a low-power deep sleep and wakes up based on a GPIO event.

earlephilhower, 2024-10-07:
I'm pretty sure the code sample you got isn't close to correct, but there are good examples from the RPI folks directly in the `pico-examples` and the SDK and datasheet docs.

I still haven't found out where my USB power monitor migrated to so don't really have the equipment to do this myself.

If someone wants to give ti a go, I'd be happy to get it in here.  The interesting bit will be getting the system and core back into a consistent state after restart.  Things like which clocks are on, what the USB state is, which blocks are enabled (ADC, PWM, timer, etc.).  Requiring the user to assume the board is equivalent to power on (but with valid RAM) might be fine, but the core internal state will need to be updated to reflect whatever is chosen...

maxgerhardt, 2024-10-09:
For the RP2350 there is at least the powman (power manager) API that can switch off certain things. And the CMSIS headers for the ARM Cortex-M33 and the RISC-V Hazard3 probably also have functions to go into a sleep state (wait for interrupt, etc.).

Also related to https://github.com/adafruit/circuitpython/issues/9491 and https://github.com/raspberrypi/pico-playground/pull/48

seamusdemora, 2024-11-07:
I need x-low power consumption from a uC for an off-grid project I've undertaken. I'd like to use the Pico for this as I've done another project (controlling an INA260) using a Pico 1. _**However:**_ I have tried the [`dormant` example from the Pico Playground](https://github.com/raspberrypi/pico-playground/tree/master/sleep) on an Pico 1, and it simply did not work for me. I'm not sure it works at all... 

I'm posting to ask about the status of "deep sleep" here: does it appear feasible/do-able, any thoughts on the timeline??

My alternative for the Pico is a TI MSP430 - they have examples that actually work :)

maxgerhardt, 2024-11-10:
I just tested some code found in the RP2040 datasheet and ran it on my Pico (1). The DORMANT sleep with  GPIO wakeup worked just fine.

Here's the sketch:
```cpp
#include <Arduino.h>
#include <hardware/xosc.h>

#define GPIO_EXIT_DORMANT_MODE 3 /* connect GP3 to GND once in DORMANT mode */

void setup() {
  Serial1.begin(9600);
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, HIGH);
  pinMode(3, INPUT_PULLUP); // pull pin that will get us out of sleep mode
}

void loop() {
  if (Serial1.available() > 0) {
    (void) Serial1.read();
    digitalWrite(LED_BUILTIN, LOW);
    // this interrupt source will get us out of the DORMANT mode agin
    gpio_set_dormant_irq_enabled(GPIO_EXIT_DORMANT_MODE, IO_BANK0_DORMANT_WAKE_INTE0_GPIO0_EDGE_LOW_BITS, true);
    // disable systick now so that no milisecond interrupts will occur
    systick_hw->csr &= ~1;
    // we will get out of sleep when an interrupt occurs.
    // this will shutdown the crystal oscillator until an interrupt occurs.
    xosc_dormant();
    // this will only be reached after wakeup.
    gpio_acknowledge_irq(GPIO_EXIT_DORMANT_MODE, IO_BANK0_DORMANT_WAKE_INTE0_GPIO0_EDGE_LOW_BITS); // acknowledge GPIO IRQ
    systick_hw->csr |= 1; // enable systick again, hope we survived this
    // we don't actually know the time duration during which we were dormant.
    // so, the absolute value ofmillis() will be messed up.
    digitalWrite(LED_BUILTIN, HIGH);
  }
  Serial1.println("hello, world");
  delay(500);
}
```
note that this was compiled with the USB stack **off**, so `Seriral1` is the hardware UART on GP0, GP1.

The sketch waits on a character to be received via UART, then it will set the crystal oscillator into dormant mode. Before that, GP3 is registered as a wakeup source, expecting a falling edge.

With a Nordic PPK2, I was able to measure the current going into the 5V "VBUS" line. So, this measures the **entire** current draw of the Pico board at 5V, and not just that of the RP2040 processor.

I can clearly see that, once I send the UART character, the current draw drops dramatially, from an average of 23.54mA to 2.10mA (at 5V).

[image]

maxgerhardt, 2024-11-10:
The `pico_sleep` component was also recently updated: https://github.com/raspberrypi/pico-extras/tree/master/src/rp2_common/pico_sleep

maxgerhardt, 2024-11-10:
Mhm, the Pico datasheet claims 0.8mA @ 5V in DORMANT mode ([source](https://datasheets.raspberrypi.com/pico/pico-datasheet.pdf)).. If I manually copy in the pico_sleep and hardware_rosc component from above (with a few mods to get rid of stdio reinit and enable a GPIO pullup) and do a 
```cpp
    sleep_run_from_dormant_source(DORMANT_SOURCE_XOSC);
    sleep_goto_dormant_until_pin(GPIO_EXIT_DORMANT_MODE, true, false); 
```
I do get down to 1.88mA. Something is still missing here.. maybe a perpiheral is on, or some other component on the board is drawing power. In any case, enough experiments for today.

[image]

earlephilhower, 2024-11-10:
Off the cuff, maybe try setting all GPIOs to `INPUT_PULLUP` to avoid wasting any pad power on floating pins?

Also probably want to `Serial1.end()` to put the UART0 into reset.  It's still a digital device running off of some clock even when idle, with input protection FFs/etc. always doing something...

seamusdemora, 2024-11-12:
> I just tested some code found in the RP2040 datasheet and ran it on my Pico (1). The DORMANT sleep with GPIO wakeup worked just fine.
> 
> Here's the sketch:

@maxgerhardt : 

That's a big help - thank you! Uh... where is the "RP2040 datasheet" where you found this?

maxgerhardt, 2024-11-12:
The pico_sleep component works on both RP2040 and RP2350 chips, it differentiates between them using `#if (!)PICO_RP2040` at compile time.

seamusdemora, 2024-11-12:
@maxgerhardt 
> I just tested some code found in the RP2040 datasheet and ran it on my Pico (1). The DORMANT sleep with GPIO wakeup worked just fine. 

Excuse me please, but I still don't understand what you mean by "code found in the RP2040 datasheet".  Can you provide some clarification - or a link to that code? I have looked in the [Raspberry Pi Pico Datasheet](https://datasheets.raspberrypi.com/pico/pico-datasheet.pdf), but I find no code in there. 

I'm also confused by the source of the [code you posted in your earlier comment](https://github.com/earlephilhower/arduino-pico/issues/2528#issuecomment-2466951767)... It's not the same as this ['hello_dormant' example in the Pico Playground repo](https://github.com/raspberrypi/pico-playground/blob/master/sleep/hello_dormant/hello_dormant_gpio.c). Where do I find this code? 

My apologies for acting as a nincompoop, but I'm not really familiar at all with this `arduino-pico` repository. Can you provide me with a brief pointer to - or explanation - on how to get started with this??

seamusdemora, 2024-11-13:
@maxgerhardt 

I've opened a discussion on an [issue I'm having getting your example above to compile](https://github.com/earlephilhower/arduino-pico/discussions/2601#discussioncomment-11234970)... I'd surely appreciate you taking a look at it.

makejimmi, 2025-05-20:
Hi, 
try deinitializing the cyw43-module (cyw43_arch_deinit()) before calling sleep_run_from_dormant_source().
Reached ~380 uA @ 5V with both cores running on the Pico 2W using the same PPK2 measurement tool. Make sure to initialize it (cyw43_arch_init()) again after sleep_power_up() in order to restore all the Wireless functionalities.

As far as I know, the module gets initialized during the boot sequence before any core gets up and running which is why the module will always draw current. I would not be able to pinpoint the location in which it gets initialized but probably here: [variants init](https://github.com/earlephilhower/arduino-pico/blob/7c51742522b54b1dbf5492fe2b513461c3cf1246/variants/rpipico2w/init.cpp#L23).

This consequently comes with the disadvantage of having all Wireless functionalities completely stopped when going and staying DORMANT but if it is about saving as much power as possible then it surely is worth considering.

[image]

Settings: cyw43_arch_deinit, Brown-Out Detection Off, Pin 4 Pull Up as wake up source

seamusdemora, 2025-05-23:
> Hi, try deinitializing the cyw43-module (cyw43_arch_deinit()) before calling sleep_run_from_dormant_source(). Reached ~380 uA @ 5V with both cores running on the Pico 2W using the same PPK2 measurement tool and the Low Power Oscillator as DORMANT source. Make sure to initialize it (cyw43_arch_init()) again after sleep_power_up() in order to restore all the Wireless functionalities.

@makejimmi : 

Thanks for the update... I'm fighting Linux audio battles at present, but will get back & try this at first opportunity; I'm sure I'll have more questions then.

seamusdemora, 2026-03-14:
Like the Phoenix, this project may be rising from the ashes?  But I think I'll wait a while to see how things pan out!

maxgerhardt, 2026-03-14:
Using a different PowMan sleeping technique (https://github.com/mamba2410/rp2350-powman-sleep/) that actually powers off the entire CPU core but needs a more difficult register / state restore code, I get it down to 475µA at 5V, but still much higher than the claimed "<150µA". GPIO pin initialization has a major effect on current draw: Different floating input vs. driving high / low vs pullup / pull down settings are needed per-pin.

[image]

seamusdemora, 2026-03-16:
@radek-turecki 

> [@seamusdemora](https://github.com/seamusdemora) Does that mean the topic has been brought up many times but never actually implemented?
> 
> Darn, that’s too bad, because I have a Pico 2 board, but with that much power consumption, I’ll have to use a different one to save energy, and I’ll use the Pico 2 for another project.
> 
> Too bad :-(

You can see the October, 2024 dates at the top of this thread... there have been other (related) threads.  IIRC, there were similar threads for the RP2040.

radek-turecki, 2026-04-25:
I got it after many tests and few hours later....

Based on my tests it seems that the problem with the resume is when the WiFi connection is established. So before the calling `Watchdog.goToSleepUntil(ms);`  there is necessary to put:
`WiFi.disconnect();`

and after resuming the module there is necessare to connect to WiFi again.

jeenode, 2026-11-08:
Picking this back up with a concrete proposal. The SDK now has `pico_low_power` (sleep, dormant, GPIO/timer wake, low-leakage pin setup), and the core already builds it into `libpico.a`. Instead of reviving #2620's copy of `sleep.c`, I wrote a thin Arduino wrapper over the SDK:

```cpp
#include <LowPower.h>
LowPower.sleepFor(ms);
LowPower.sleepUntilPin(pin, FALLING);
LowPower.sleepUntilPin(pin, FALLING, timeoutMs);   // pin or timeout; LowPower.wakeReason() says which
LowPower.dormantUntilPin(pin, FALLING);
LowPower.dormantFor(ms);        // RP2350, or RP2040 with an external 32 kHz clock
LowPower.keepPins(mask);        // exclude pins from low-leakage parking
```

The SDK wakes into its default clocks, so the core-specific part is re-applying what `main()` sets up (`F_CPU`, the VREG bump, PSRAM timing) after wake; the library also parks and restores pins around dormant and idles core 1. The SDK's light sleep leaves both PLLs and the ring oscillator running (about 8 mA on a Pico), so on RP2040 the library runs the system from the crystal and stops them around `sleepFor`/`sleepUntilPin`; in USB builds the USB clock is kept only while a host is actually using the port. Light sleep also leaves `attachInterrupt()` handlers working (the SDK's GPIO-wake helper clears the shared GPIO callback, so the library uses its own wake path).

**What is done and tested** (RP2040 Pico, one board): about 6,700 wakes across all modes, with and without USB (the last soaks on the current code: 2,400 mixed and 2,760 light-sleep wakes), no failures from the library, and SPI/I2C/PIO/UART/ADC/PWM keep working after wake without `begin()`. Wake latency: dormant about 73 ms (crystal restart), light sleep on a pin a few ms. The examples build for RP2040 and RP2350 (ARM and RISC-V, default stack, `NO_USB`, Adafruit TinyUSB, 200 MHz, PSRAM, FreeRTOS), astyle clean.

Whole-board current on a Pico at VSYS (INA226, uncalibrated, ±0.1 mA):

| | |
|---|---|
| running, 133 MHz | 19.7 mA |
| SDK light sleep alone | 7.9 mA |
| light sleep, USB build with no host (default sketches on battery) | 1.1 mA |
| light sleep, `NO_USB` build | 1.3 mA |
| light sleep, host using USB | 3.1 mA |
| dormant | 0.76 mA |

**Not done:** no RP2350 hardware run, no chip-level (PPK2) current numbers yet.

**Proposed PR order** (each is small and builds on its own; PRs to follow):
1. RP2350: provide empty `__persistent_data` linker symbols (any `low_power_*()` call fails to link on RP2350 today)
2. `stdio_flush()` links with `NO_USB` and `USE_TINYUSB` (same failure for any `low_power_*()` call)
3. `__applyCoreClockConfig()`: the clock setup from `main()` as a function, no change at boot
4. the LowPower library, examples and docs page

**One known problem**, not caused by the library: on early RP2040 chips (B0/B1) with USB on, the SDK's RP2040-E5 enumeration workaround can `panic()` when the host resets the bus. A dormant wake forces a re-enumeration, so it shows up there (about 1 in 100 wakes on my board); a plain USB disconnect/connect with no low-power code gives 1 in 400, and with USB disabled I saw none in 400 dormant wakes. The workaround is off by default in the SDK and enabled by this core's `libpico`. I'll file this separately and documented it in the library docs as a known issue.

**Disclosure:** this was done with the help of Claude (Anthropic's AI assistant). It wrote most of the code, tests and docs and drove the bench runs under my direction; the hardware, the measurements and the decisions are mine, and I'll answer for the PRs.

Questions: would you take this as `libraries/LowPower` in the core, or would you rather have it as a separate library? Happy to change names or scope before the PRs go out. And is the order above fine, or would you prefer (3) folded into (4)?

maxgerhardt, 2026-11-08:
Wouldn't it make more sense to merge improvements into Adafruit's SleepyDog library?

jeenode, 2026-11-08:
Good question, and I did look at that first. A few reasons I went for the core instead:

1. **Most of the work has to be in the core anyway.** The SDK's low-power calls wake up on the SDK's default clocks, so whatever library calls them needs the core to re-apply its own setup afterwards (`F_CPU`, the VREG bump, PSRAM timing). That's PR (3), `__applyCoreClockConfig()`. PRs (1) and (2) are link fixes: today any call to `low_power_*()` fails to link on RP2350, or with `NO_USB` / Adafruit TinyUSB on RP2040. SleepyDog would need those three fixes too.
2. **The library depends on core internals.** It has to know which USB stack is in use and whether a host is attached (to decide whether to keep the USB clock), which pins the core has muxed to UART/I2C (dormant must not park them), how to idle core 1, and it has to stay consistent with `attachInterrupt()` (the SDK's GPIO-wake helper clears the shared GPIO callback). From outside the core that means relying on private symbols that can change between releases.
3. **The API is a different shape.** SleepyDog's portable `sleep(ms)` maps to light sleep with a timer. Pin wake, pin-or-timeout with `wakeReason()`, dormant with pin parking, and `keepPins()` don't have an equivalent there. And SleepyDog's current RP2040 `sleep()` (1.8.4) is `sleep_ms()`, i.e. no actual low-power mode; on RP2350 the portable `sleep()` is also `sleep_ms()`, with separate RP2350-only sleep/dormant helpers based on a vendored copy of the old pico-extras sleep code rather than the SDK's `pico_low_power`.

WIth all above said, I don't see it as either/or. Once the core pieces are in, SleepyDog's RP2040/RP2350 backend could become a thin call into `LowPower.sleepFor()`, and SleepyDog users would get the real low-power mode for free. I'm happy to send that adapter to Adafruit as a follow-up.

I want to say  thanks for #2620 as  it was the starting point for this, and the clock-restore issue it ran into is exactly what PR (3) addresses.

@earlephilhower: is in-core (`libraries/LowPower`) OK with you, or would you prefer it as a separate library on top of the core fixes?

earlephilhower, 2026-11-13:
@jeenode, I'm okay with an in-core operation.

As you said, on the more complicated setups the generic Adafruit library needs tons of hooks in the core anyway.  And looking at the Adafruit library it seems more focused on watchdog (i.e. the "Dog" in the name) than differing low power sleep modes.

jeenode, 2026-11-14:
Thanks — great. I’ll send it as a short series so each piece is easy to review: first the small core prerequisites (RP2350 .persistent_data symbols, a stdio flush fix, and re-applying the core clock config after wake), then the LowPower library itself (light sleep, pin/timer/pin-or-timeout wakes, dormant), with follow-ups after that for keeping selected peripherals running, WiFi.powerOff()/powerOn(), and RP2350 deep sleep with retained data. All of it has been bench-tested on Pico, Pico W, Pico 2 and Pico 2 W with current measurements; numbers will be in each PR.

earlephilhower, 2026-11-15:
I believe this was closed as a new feature in #3624.  Any bugs found should be tracked with another issue to keep it clean (and shorter!).  Thanks, @jeenode !

closed 2026-11-15 by earlephilhower