earlephilhower/arduino-pico #2534: RP2350 ADC INL/DNL

greiman, 2024-10-10:
I am looking for other users experience with the Pico 2 ADC.

I found lots of noise when using USB power.  I followed a suggestion and put a LM4040C30 3.0 volt reference from the Vref pin to ground.  That improved the noise problem.

The INL problem is better but still not great.  the DNL is also not good.

I used a 16-bit  SD5693 DAC to generate ADC input. I stepped from zero to 3.0 V in one millivolt steps.  I get this error chart for the ADC at 12-bit resolution.  I subtracted a large zero offset and adjusted for the 3.0 V reference.

[image]
 The three big jumps are at these multiples of 512:  1536, 2048, 2560.

This is better than the RP2040.  Here is an early example for a Pico 1.  It has noise since I didn't use a voltage reference chip.  It goes to 3.3v since 3V3 is the reference.

[image]

I checked the DAC performance with a 6 1/2 digit DMM.  Here is the result:

[image]

earlephilhower, 2024-10-11:
Very interesting results.   There's a minimum of 9 ENOB per the dataset, but that looks more 7.5 if my back of the envelope math is correct.

Have you opened up a discussion on the RPI forums?  The actual designers are active there and might be able to offer assistance.

greiman, 2024-10-12:
> Have you opened up a discussion on the RPI forums? 

I see the claim of 9 ENOB and I looked at the forums but found nothing.

The latest datasheet still has this so I am waiting for more info.

> 12.4.5. INL and DNL
> Details to follow.

I really like the RP2350 but often need an ADC for my projects but it looks like strike two for the ADC.

greiman, 2024-10-13:
Here Is a SAMD21 ADC with Vref the internal 3.3v.

[image]

Better accuracy between 0.3 and 3.0 V.

earlephilhower, 2024-10-13:
Microchip has had a few more years than the RPI folks getting the analog portion right.  Analog ASIC design is always a dark art practiced by wizards. :laughing:

Just spitballing, but is it possible to design a transfer function to get you closer to what you need?  If multiple chips are repeatably off in the same way, is there a 4K entry lookup table that can map from ADC readout to actual physical value?  I know you can't add information but maybe you can shift the errors around?

greiman, 2024-10-14:
> Just spitballing, but is it possible to design a transfer function to get you closer to what you need?

Probably a quadratic fit would be pretty good for the RP2350.  I don't like the DNL jumps that many CPU chips have.

I have lots of experience with ADC chips so if I need more accuracy I add a chip.  I have written drivers for a number chips.

For high accuracy I like the MCP342X family of 18 bit ADCs.  They have a builtin reference and INL: 10 ppm of Full Scale Range.

There are lots of good SAR 12 bit chips.  For many years I have used the MCP320X family.

greiman, 2024-10-15:
Here is an example of a 12-bit MCP3202 with a Pico 2.  I just put the ADC and DAC chips on a bread board with no caps so there is some noise.

Most reading are within a millivolt.  Since the least significant bit is 3000/4095 = 0.73 millivolt this is about as good as you can expect.

[image]

greiman, 2024-10-16:
I have been trying to reduce the noise on Pico 2 3V3.  The schematic has a note that I can run the RT6150B Buck/Boost converter in the default  PFM mode or the alternate PWM mode.

I loaded this program:
```
void setup() {
  pinMode(23, OUTPUT);
  // digitalWrite(23, HIGH);  // PWM Mode
  digitalWrite(23, LOW);  // PFM Mode
}
void loop() {
}
```
And I get this trace on 3V3 note the time scale is 20 μs/division.
[image]

I get this in PWM mode at 500 ns/division.
[image]

Still not great.

I tried an Adafruit Feather RP2350 with this program:
```
void setup() {}
void loop() {}
```
And get this on 3V3:
[image]

Looks like I should use an Adafruit board for analog apps.

Anyone checked other new RP2350 boards?

retsifp, 2024-10-28:
The relevant difference is, that the pico 2 comes with a buck-boost regulator, which is more efficient and is able to power the pico down to 1.8V. That is great for battery powered applications!
The Adafruit Feather RP2350 comes with an Low-dropout-regulator (LDO), which just burnes the difference between 5V and 3.3V. Very inefficient, but due to the linear regulation great for analog applications! :)

greiman, 2024-10-29:
> The relevant difference is, that the pico 2 comes with a buck-boost regulator, which is more efficient and is able to power the pico down to 1.8V. That is great for battery powered applications!

I understand all of this.  For my applications the Adafruit approach is better.  For USB power I don't care about burning a little power.

For most battery power applications I like the builtin LiPoly charger.

There are options for disabling the LDO regulator and the LiPoly charger to use the JST-PH battery port with non-LiPoly batteries, such as AA or AAA battery packs.

Here are pages about power for the RP2040 and RP2350 Feather Boards that discuss power options.

https://learn.adafruit.com/adafruit-feather-rp2040-pico/power-management

https://learn.adafruit.com/adafruit-feather-rp2350/pinouts