earlephilhower/arduino-pico #2535: RP2350 FPU

DatanoiseTV, 2024-10-10:
According to [this](https://github.com/earlephilhower/arduino-pico/blob/edba2faa3424737b1b24d5583a330d1c54f0e7df/tools/makeboards.py#L311) it seems like it is using softmp mode for floating point operation for the RP2350.

SoftFP (soft floating point): Use hardware floating-point operations, but function calls pass parameters via general-purpose registers (compatible with non-FPU systems).

HardFP (hard floating point): Pass floating-point values in floating-point registers during function calls (which is faster but requires hardware FPU support).

Would it make sense to switch it to
```
-mfloat-abi=hard
```

earlephilhower, 2024-10-11:
We're using the same options as the SDK.  I'm sure there was a reason for them choosing it. 

Possibly because the m33 only has sp float support and half the fp registers as the full fat implementation?

You should open up an issue on the pico-sdk repo or the RPI forums.  If it could work, and it improves performance, that would help all apps, not just this core.

earlephilhower, 2024-10-24:
Looks like the SDK folks aren't really interested in this due to some ROM and assembly optimized functions they provide using `softfp`.  If they do change, we'll follow (will need new library build, I think).

closed 2024-10-24 by earlephilhower