earlephilhower/arduino-pico #2538: Delay breaking serial communication

GuiltySpark7, 2024-10-12:
I came across a weird bug (I think, maybe I am missing something)  where a delay() greater than 2030-2090ms causes the serial to stop printing.  It took me a long time to figure this out as I was using the Serial to debug

IDE 2.3.3
Pico 2 (RP2350) && Pico (RP2040)
Linux Mint
I wrote the following code to isolate and replicate the problem:

`
unsigned long timer=0;
int delay_ms;

void setup() {
  Serial.begin(9600);
  pinMode(25, OUTPUT);
 

}

void loop() {
  Serial.println(millis()-timer);
  timer = millis();
  delay(delay_ms);
  delay_ms += 30;
  digitalWrite(25, !digitalRead(25));
  
  

}
`

The serial output reliably gets its last transmission between 2000 and 2100 (near 2^11?).  The LED however continues to blink so the code is still running. Just something about the Serial.print() no longer works
Example of terminal:
...1960
1980
2000
2020
2040
2060
2080
2100
2120

It does not reliable stop at the same value. a range from 2020-2200

I wrote the following code "delay()" free to see if maybe it was a problem with the nature of delay function it's self:

`unsigned long timer=0;

unsigned long printLoopTimer = 0;
int printLoopInterval = 1800;
int printLoopIntervalInc = 20;

void setup() {
  Serial.begin(9600);
  pinMode(25, OUTPUT);
}

void loop() {
  if (printLoopTimer < millis()){
    Serial.println(millis()-timer);
    timer = millis();
    digitalWrite(25, !digitalRead(25));
    printLoopTimer += printLoopInterval;
    printLoopInterval += printLoopIntervalInc;
  }
}
` 
but again the last sucessful transmit was at 2130.  So it seems that the serial needs to get called every 2 seconds or it breaks but the code keeps running...

I can reproduce this with 
pico 1 and 2
Alternative serial capture software (CoolTerm linux 64bit in this case) 

I am happy to just not use delay as I don't like blocking code anyway but this is causing me problems with some libraries that do. Right now that is Stuart Robinson's LoRa library.

Any thoughts?

GuiltySpark7, 2024-10-12:
I am yet to try this on a different computer entirely in case it is something to do with my OS but that seems unlikely..

earlephilhower, 2024-10-12:
Err, there is absolutely no guarantee with USB timings.  You're writing things, so depending on when the OS polls the Pico you can sit and wait multiple milliseconds as a random variable.

If you need to synchronize something at exactly some number of milliseconds, you can't just open-loop delay() and expect it to be synchronized.  You'll need to use an active feedback loop, looking at millis() instead of using a hardcoded delay(), or something like the `Ticker` library.

So this isn't an issue, just the way a system with USB will work.

closed 2024-10-12 by earlephilhower

maxgerhardt, 2024-10-12:
You might be able to make the timing more deterministic by forcing a `Serial.flush();` after you printed some data.

GuiltySpark7, 2024-10-13:
Sorry I talked to much about timing which gave you the wrong end of the stick.

The problem was that the serial communication would stop working if the Arduino did not print anything for an extended period.  All the timing talk was demonstrating and determining exactly how long it took for the comms to break down (no problems when for frequent prints were called.)

the problem does not present its self when talking to an old raspberry Pi 4b i had lieing around, which means that actually the problem was a new problem that has developed on my USB drivers on my laptop...  Nothing to do with the RPcore and this issue can stay closed

Sorry for the false alarm!

The USB driver thing really was unexpected. Has thrown me big time, looking in all the wrong places

earlephilhower, 2024-10-13:
Great, thanks for the update.