earlephilhower/arduino-pico #2547: RP2040  Raspberry Pico W issues with KeyboardBLE

nkoske, 2024-10-16:
Im having Trouble to get KeyboardBLE Running and staying Connected.

Compiling my Code and running it is no issue, but when i connect to my Win11 Laptop it connects and disconnects after few seconds.
Windows 11 tells me its an driver issue, i tryed to install new drivers (other BT devices work just fine).

My gut tells me its either my code or the libs fault.
What im doing wrong here?

`#include <KeyboardBLE.h>

// Define keyboard inputs and outputs
byte inputs[] = {6, 7, 8}; // Minimal 3 inputs
byte outputs[] = {0, 1};   // Minimal 2 outputs

// Simple keyboard layout with a 2x3 matrix
char layout[2][3] = {
    {'a', 'b', 'c'},
    {'d', 'e', 'f'}
};

int keyDown[2][3] = {0};   // Key status

void setup() {
  // Initialize outputs
  for (int i = 0; i < 2; i++) {
    pinMode(outputs[i], OUTPUT);
    digitalWrite(outputs[i], HIGH);
  }
  
  // Initialize inputs
  for (int i = 0; i < 3; i++) {
    pinMode(inputs[i], INPUT_PULLUP);
  }

  // Start Bluetooth HID
  KeyboardBLE.begin();
  delay(5000);  // Allow time for Bluetooth to initialize
}

void loop() {
  for (int i = 0; i < 2; i++) {    
    digitalWrite(outputs[i], LOW);  // Set one row low
    
    for (int j = 0; j < 3; j++) {
      if (digitalRead(inputs[j]) == LOW) {
        keyPressed(i, j);  // Call function when key is pressed
      } else if (keyDown[i][j] != 0) {
        resetKey(i, j);    // Reset key when released
      }
    }
    
    digitalWrite(outputs[i], HIGH);  // Set the row back high
  }
}

void keyPressed(int row, int col) {
  if (keyDown[row][col] == 0) {
    KeyboardBLE.write(layout[row][col]);  // Send keypress
  }
  keyDown[row][col]++;
}

void resetKey(int row, int col) {
  keyDown[row][col] = 0;
  KeyboardBLE.release(layout[row][col]);  // Release the key
}
`

maxgerhardt, 2024-10-19:
How is your Pico W board powered? Maybe by a power bank, that shuts off if not enough current is pulled after some time?

nkoske, 2024-10-19:
Im Testing on a Pc Port or on a  USB-PSU

maxgerhardt, 2024-10-19:
Interesting, I'm replicating this error on a Pico W on Windows 10 with a much simpler sketch:

[image]

```cpp
#include <Arduino.h>
#include <KeyboardBLE.h>

void setup() {
  KeyboardBLE.begin();  // Start Bluetooth HID
  delay(5000);  // Allow time for Bluetooth to initialize
}

void loop() {
  if (BOOTSEL) {
   KeyboardBLE.press('A');
  } else {
   KeyboardBLE.release('A');
  }
}
```
(doesn't need any external buttons, uses the BOOTSEL button itself). Let's see if the verbatim examples work..

maxgerhardt, 2024-10-19:
Nope, https://github.com/earlephilhower/arduino-pico/blob/master/libraries/rp2040/examples/Bootsel/Bootsel.ino runs into the same error. Let's what the nRF Connect app says-

maxgerhardt, 2024-10-19:
Mhm. nRF Conenct shows an "Incorrect data length (16 bit expected): (0x) 00" error when trying to get the Boot Keyboard Input Report.. not exactly sure what the means yet.

[image]

earlephilhower, 2024-10-21:
Re the [dump you posted](https://github.com/earlephilhower/arduino-pico/issues/2547#issuecomment-2423908944), @maxgerhardt ...`Boot Keyboard Input Report`... may be the issue.  This core does not implement the boot protocol mode of operation.

`HID_Keyboard` is the backend and is common for USB, BT, and BLE (since HID reports are the same for all of them).  There's a bit in the HID descriptor the Pico supplies which toggles between boot and normal modes.  BOOT protocol mode also only allows 1 device, not composite devices and the "keyboard" is really a composite device of a keyboards and a consumer_keys (mute, pause, volup/down keys, etc.).

There also should not be a mouse descriptor at all.  There is logic to combine things allowing for, say, a keyboard and mouse to be emulated.  But you should need to actually instantiate the MouseBLE to have the HID report descriptor report that.

earlephilhower, 2024-10-21:
The HID descriptor is good.  This is dumped from the KeyboardBLE example and decoded online as a USB HID.  No mouse. present from inside the core

````
0x05, 0x01,        // Usage Page (Generic Desktop Ctrls)
0x09, 0x06,        // Usage (Keyboard)
0xA1, 0x01,        // Collection (Application)
0x85, 0x01,        //   Report ID (1)
0x05, 0x07,        //   Usage Page (Kbrd/Keypad)
0x19, 0xE0,        //   Usage Minimum (0xE0)
0x29, 0xE7,        //   Usage Maximum (0xE7)
0x15, 0x00,        //   Logical Minimum (0)
0x25, 0x01,        //   Logical Maximum (1)
0x95, 0x08,        //   Report Count (8)
0x75, 0x01,        //   Report Size (1)
0x81, 0x02,        //   Input (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x95, 0x01,        //   Report Count (1)
0x75, 0x08,        //   Report Size (8)
0x81, 0x01,        //   Input (Const,Array,Abs,No Wrap,Linear,Preferred State,No Null Position)
0x05, 0x08,        //   Usage Page (LEDs)
0x19, 0x01,        //   Usage Minimum (Num Lock)
0x29, 0x05,        //   Usage Maximum (Kana)
0x95, 0x05,        //   Report Count (5)
0x75, 0x01,        //   Report Size (1)
0x91, 0x02,        //   Output (Data,Var,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
0x95, 0x01,        //   Report Count (1)
0x75, 0x03,        //   Report Size (3)
0x91, 0x01,        //   Output (Const,Array,Abs,No Wrap,Linear,Preferred State,No Null Position,Non-volatile)
0x05, 0x07,        //   Usage Page (Kbrd/Keypad)
0x19, 0x00,        //   Usage Minimum (0x00)
0x2A, 0xFF, 0x00,  //   Usage Maximum (0xFF)
0x15, 0x00,        //   Logical Minimum (0)
0x26, 0xFF, 0x00,  //   Logical Maximum (255)
0x95, 0x06,        //   Report Count (6)
0x75, 0x08,        //   Report Size (8)
0x81, 0x00,        //   Input (Data,Array,Abs,No Wrap,Linear,Preferred State,No Null Position)
0xC0,              // End Collection
0x05, 0x0C,        // Usage Page (Consumer)
0x09, 0x01,        // Usage (Consumer Control)
0xA1, 0x01,        // Collection (Application)
0x85, 0x02,        //   Report ID (2)
0x15, 0x00,        //   Logical Minimum (0)
0x26, 0xFF, 0x03,  //   Logical Maximum (1023)
0x19, 0x00,        //   Usage Minimum (Unassigned)
0x2A, 0xFF, 0x03,  //   Usage Maximum (0x03FF)
0x95, 0x01,        //   Report Count (1)
0x75, 0x10,        //   Report Size (16)
0x81, 0x00,        //   Input (Data,Array,Abs,No Wrap,Linear,Preferred State,No Null Position)
0xC0,              // End Collection

// 92 bytes
````

Report ID 0 is the feature report in this BLE implementation, so the report ID looks good.

earlephilhower, 2024-10-22:
Version 3.9.5 (pre 2.0 SDK) works fine with the examples I've tried for BLE.  This is a BTStack change in the 2.0 SDK release.

If I build and run the example I see the correct protocol mode and no errors in the "boot keyboard input report" section....
[image]

earlephilhower, 2024-10-26:
No diffs w/the 2.0.0SDK in the core for BT other than removing the manual absolute mouse include (now included in TinyUSB.
````
earle@amd:~/Arduino/hardware/pico/rp2040$ git diff dd1c9095e8e3f37eb38fc4d11f9aec9214d1cfba libraries/HID_Bluetooth/
diff --git a/libraries/HID_Bluetooth/src/HID_Bluetooth.h b/libraries/HID_Bluetooth/src/HID_Bluetooth.h
index ddb90bf08..e8e6e0b42 100644
--- a/libraries/HID_Bluetooth/src/HID_Bluetooth.h
+++ b/libraries/HID_Bluetooth/src/HID_Bluetooth.h
@@ -8,11 +8,6 @@
 
 #pragma once
 
-//necessary to implement the absolute mouse descriptor define,
-//remove if merged into TinyUSB
-#include <sdkoverride/tusb_absmouse.h>
-
-
 //override weak declarations to include HID report to report map.
 //done in each library (KeyboardBT,...)
 extern void __BTInstallKeyboard() __attribute__((weak));
earle@amd:~/Arduino/hardware/pico/rp2040$ git diff dd1c9095e8e3f37eb38fc4d11f9aec9214d1cfba libraries/KeyboardBLE/
earle@amd:~/Arduino/hardware/pico/rp2040$ 
````

No changes in the BTStack HOG (HID-over-GATT aka BLE HID) keyboard demo sources other than a minor pairing tweak
````
earle@amd:~/Arduino/hardware/pico/rp2040/pico-sdk/lib/btstack/example$ git diff 0d212321a995ed2f9a80988f73ede854c7ad23b8 hog_keyboard_demo.c
diff --git a/example/hog_keyboard_demo.c b/example/hog_keyboard_demo.c
index 43397b791..90d59b391 100644
--- a/example/hog_keyboard_demo.c
+++ b/example/hog_keyboard_demo.c
@@ -188,7 +188,7 @@ static void le_keyboard_setup(void){
 
     // setup SM: Display only
     sm_init();
-    sm_set_io_capabilities(IO_CAPABILITY_DISPLAY_ONLY);
+    sm_set_io_capabilities(IO_CAPABILITY_NO_INPUT_NO_OUTPUT);
     sm_set_authentication_requirements(SM_AUTHREQ_SECURE_CONNECTION | SM_AUTHREQ_BONDING);
 
     // setup ATT server
earle@amd:~/Arduino/hardware/pico/rp2040/pico-sdk/lib/btstack/example$ git diff 0d212321a995ed2f9a80988f73ede854c7ad23b8 hog_keyboard_demo.gatt
earle@amd:~/Arduino/hardware/pico/rp2040/pico-sdk/lib/btstack/example$ 
````

earlephilhower, 2024-10-26:
Even building and running the BTStack included `hog_keyboard_demo.c` no longer works with `git` `master`.

Running it under 3.9.5 works perfectly.

Running it under 4.0.1 fails, so this is not related to the GCC12->14 bump (thankfully).

So, at this point, the 2.0 SDK sources seem to be failing without anything on our part.  I'll see if I can get a raw SDK build for the hog_keyboard demo and see if that fails.  If so, then this is out of our hands and I'll report in the SDK repo.

earlephilhower, 2024-10-26:
Nope, Using the [official ARM toolchain](https://developer.arm.com/downloads/-/arm-gnu-toolchain-downloads)  and `pico-examples`, the `hog_keyboard_demo` no longer works with the latest SDK.  I'll open an issue there, but this means it's out of our control here, unfortunately...

closed 2024-10-27 by earlephilhower

maxgerhardt, 2024-10-29:
Just retested this with the latest Arduino-Pico 4.2.0 release and indeed it's working, no issues.

commit 5b135fab (2024-10-27): Fix 2.0 SDK BLE devices (#2560)

Remove old pre-release BTStack files from sdkoverrides (were needed for
compound BLE devices, but the latest BTStack now incorporates these
changes).

Fixes #2547