earlephilhower/arduino-pico #2568: Disabling interrupts and then re-enabling them seems to crash the MCU

MProx, 2024-10-30:
Apologies, I'm new to the Pico device. I've hit a weird issue where disabling and then re-enabling interrupts seems to crash the MCU. I'm using the latest version of this core (4.2.0). Is this expected behaviour, or a known issue? Minimum sketch to reproduce:

```
void setup()
{
  Serial.begin(115200);
}

void loop()
{
  delay(1000);
  static uint8_t count = 0;
  Serial.println(count++);
  if (count == 5)
  {
    noInterrupts();
    delay(1);
    interrupts();
    // ----> CRASHES HERE  <----
    Serial.println("This line never prints");
  }
}
```

I'm not sure if this is actually related, but I encountered this while investigating a wider issue where I can't seem to get two devices to share an SPI bus on the device. I have a RFM95W Lora module and a GC9A01 TFT display. They each work individually, but attempting to use them together breaks the LoRa module (which uses interrupts to know when new messages have arrived). This doesn't crash the MCU, but it does break the functionality of the radio (it can't receive messages any more). Minimal reproduction is here (note the last few lines specifically): https://pastebin.com/CqszEU2x

The RadioHead library specifically notes only partial support for the Pico was introduced back in 2021, due to problems with interrupts. I'm not sure if this was ever resolved. From the RadioHead change notes:

> 1.120 2021-11-13 Added intial support for Raspberry Pi Pico, using the Raspberry Pi Pico/RP2040 core by Earle F. Philhower, version 1.93, installed per https://arduino-pico.readthedocs.io/_/downloads/en/latest/pdf/ [RH_ASK](http://www.airspayce.com/mikem/arduino/RadioHead/classRH__ASK.html) works with default pins (11 and 12). But it seems that interrupt callbacks (needed by most SPI based radio drivers in RadioHead) are not working (yet) in this core, at least when compiled on our Linux platform. It appears there is some bug in std::map that causes the interrupt handling to not work correctly (which I have been able to reproduce in a simple sketch that only uses std::map). This is very strange. Interrupts are supported and work in the alternative Arduino MBED OS RP2040 core version 2.4.1, but there are other mysterious crashes RadioHead when a radio driver reads SPI inside an interrupt. But [RH_ASK](http://www.airspayce.com/mikem/arduino/RadioHead/classRH__ASK.html) also works with this core. So the result (so far) is partial support of RadioHead using either core (so far). We will revisit this when either core is updated

earlephilhower, 2024-10-30:
Interrupts get disabled/re-enabled many times per second (especially with Ethernet or WiFi!) in the core.

However, it's illegal to `delay` while interrupts are disabled.  That's how the core gets woken up again.  It uses the SDK's low-power alarm logic and `wfe` to await the timeout.

closed 2024-10-30 by earlephilhower

earlephilhower, 2024-10-30:
Oh, if you really need IRQs disabled for some fixed time (but in doing so you may break other things like the USB port), you can look at the cycle count since SYSCLK exception every 16M clocks should ignore any IRQ disabled bits.  Just calculate when you want to end and busy-read it.

MProx, 2024-10-31:
Thanks. I wasn't aware of this! Apologies for the rookie question.

Do you have any comment about the other point from my post, about the RadioHead library and their issues with interrupts in the Pico core? If not, that's totally fine, but their issues seem to go back many years so I'm hoping that there has been some progress. If you'd rather that be a separate issue, I can also make a new one.

earlephilhower, 2024-10-31:
Sorry, I have no info on the RadioHead library.  We do SPI and IRQs just fine, as far as I know, and have for quite some time.  If you have a specific issue with MCVE we can look at it in another issue, sure!