earlephilhower/arduino-pico #2569: WebServer with AM2302 Sensor

hasenradball, 2024-10-31:
Dear @earlephilhower,

thank you for providing this library.

I checked this with the RaspberryPi PicoW in the following way, and observed a not expected behaviour.

- a) take as basis the example of "HelloServer"
- b) I added a single Sensor AM2302 (aka DHT22) by cpp library.
- c) changed the Website in this way that it shows the temperature and humidity values. 

When you use the sensor lib alone on the Pico/PicoW everything is fine:
sensor.read() ==> returns always Status == OK.

But when you use the Sensor lib in combination with the Webserver the 
sensor.read() ==> returns mostly not Status == OK.

What means Webserver + Sensor readings is too much for the PicoW, my assumption.
But it is difficult to believe for me.

Wanted to analyse more in detail.
Can you give some hints to this issue?

AM2302 Lib: [AM2302-Sensor](https://github.com/hasenradball/AM2302-Sensor)

dquadros, 2024-10-31:
The response from DHT sensors is "time coded": the signal stays high for different times to indicate a 0 or a 1. The AM3202-Sensor library measure these times using a loop and delays, so it will be sensitive to interrupts. I observed similar problems with CircuitPython and a Adafruit library. The RP2040 has ways to do this more efficiently, by doing the reading in the other core or using the PIO.

hasenradball, 2024-10-31:
I will check this at ESP8266 but expectation is, this does not occur.

hasenradball, 2024-10-31:
@dquadros have you examples shifting to other core?

earlephilhower, 2024-10-31:
Run your sensor update in `setup1` and `loop1` like in [the example](https://github.com/earlephilhower/arduino-pico/blob/master/libraries/rp2040/examples/Multicore/Multicore.ino), updating a `volatile` global state variable, for example.

As @dquadros said, we may have more interrupts than the 8266 especially since the IP stack runs without needing `yield` to process.  If the bitrate is high enough, you might be able to get away with disabling interrupts around the reading process.  But if it's a multi-millisecond procedure, I second his suggestion of using the PIO or another method to get the more precise timing.

hasenradball, 2024-10-31:
Great Thanks for your feedback @earlephilhower + @dquadros .
Is there also an example of using the PIO?