earlephilhower/arduino-pico #2572: Upload of package_rp2040_index.json hangs

djaus2, 2024-11-01:
When I open the Arduino IDE I continually get  the **Downloading index: package_rp2040_index.json** upload message but it jumps to about 2/3rd done and doesn't progress. At least for a long time.

earlephilhower, 2024-11-01:
The JSON file is served from GitHub, so there's not anything we can do here.  You may have a network issue, but that's something you might be able to workaround by using `git` and not the Arduino IDE to install the core.  See the [docs](https://github.com/earlephilhower/arduino-pico?tab=readme-ov-file#installing-via-git) for more info on installing via git.

closed 2024-11-01 by earlephilhower

djaus2, 2024-11-01:
It happens when I start a build.

djaus2, 2024-11-01:
Also I can directly download https://github.com/earlephilhower/arduino-pico/releases/download/global/package_rp2040_index.json vias browser
so network is working.

djaus2, 2024-11-01:
Did the Git install, no change.
>NB: The BSP was insatlled previously, guess it is trying to just do an update.

djaus2, 2024-11-01:
[image]

djaus2, 2024-11-01:
Problem seems to have gone way, for now. Not sure why.

djaus2, 2024-11-15:
Still happens when I  reload the IDE.
[image]
Backgraph doesn't move.
Note no or little netwrk load (o or 0.1)
Workaround is 
- to open a new Sketch from the IDE.
- Close the current one then reopen it.
?????????????????