earlephilhower/arduino-pico #2574: WIZnet EVB boards

sstaub, 2024-11-02:
Can you also add the WIZnet W5500-EVB-Pico2 and WIZnet W5100S-EVB-Pico2 boards?

earlephilhower, 2024-11-02:
Do you have them in hand to test?  Are they the same pinout as the RP2040 versions?  If so, I can make a sample and you can try.  OTW, I don't have anything to verify with so we'll probably have to wait for WizNet or some user to add.

sstaub, 2024-11-02:
I have a WIZnet W5500-EVB-Pico2, it is the same pinout like the others WIZnet board, the only different between the original Pico2 is the smaller Flash (2MB).

earlephilhower, 2024-11-02:
@sstaub if you can use `git`, can you try the PR attached #2575 and verify the example W5500 works?  Per the schematic it looks good.

I'm just surprised they cheaped out on the flash.  2MB is pretty tight if you want to do OTA...

sstaub, 2024-11-02:
I found that there is a Quad SPI Flash on board with 16MB for the boards with RP2350, the W55RP20-EVB-Pico has 2MB like the other RP2040 boards.
It was hard to find out because it is not documented but you can find the Flash in the schematics (W25Q16JVUXIQ)

earlephilhower, 2024-11-02:
W25Q16JVUXIQ = 16 mega**bit**, so only 2 mega**byte**, like the earlier boards and the old Pico 1.  Official RPi Pico 2 boards moved to 4MB (32Mbit) chips.

sstaub, 2024-11-03:
How to test the PR with the Arduino IDE? PIO doesn't work because the board is missing.
On Arduino IDE I tried Generic 2350 with 2 MB and it works with Ethernet.

closed 2024-11-03 by earlephilhower

earlephilhower, 2024-11-03:
If you can run the `git` copy of the core (not IDE installed), just `git pull` and you'll get the new boards.  Otherwise, it's probably just fine to wait for a release and let me know if there's something amiss afterwards.

FWIW, the implementation here doesn't need to use the `Ethernet` library.  The `lwip_w5500` library hooks the chip directly into the LWIP stack so it works like the WiFi or other adapters and OTA, WebServer, etc. should all "just work."

maxgerhardt, 2024-11-06:
>PIO doesn't work because the board is missing.

Updated per https://github.com/maxgerhardt/platform-raspberrypi/commit/b3e82da57ba00d6e4e574adcb3d95af75f8da735

PR #2575 (2024-11-03): Add wiznet_5100s_evb_pico2, wiznet_5500_evb_pico2
Add 2 new WIZNet Pico2 boards, same pinouts as Pico revs.

Fixes #2574
commit c4b65218 (2024-11-03): Add wiznet_5100s_evb_pico2, wiznet_5500_evb_pico2 (#2575)

Add 2 new WIZNet Pico2 boards, same pinouts as Pico revs.

Fixes #2574