earlephilhower/arduino-pico #2577: Initial connection to an MQTT server (and possibly other WLAN activities) clears attached interupts

jaguilar, 2024-11-03:
In the below test program, if I set up the GPIO IRQs before setting up the MQTT client (i.e. `DINTERRUPT_SECOND=0`), the IRQs get cleared. If I set up the GPIO IRQs after, then they work fine. Through testing with another program, I've found that the internet still works fine with my IRQs in place. So there is likely some bug that is causing them to be cleared.

I tried to debug this but eventually I got into places where only machine code was visible, because it looks like some of the libraries in this package are precompiled. I would love to debug further, but I don't see a clear path to diving into that prepackaged libraries. Please let me know if there's a reasonably straightforward way to do that..

```
#include <Arduino.h>
#include <FreeRTOS.h>
#include <HADevice.h>
#include <HAMqtt.h>
#include <WiFi.h>
#include <device-types/HASensor.h>
#include <semphr.h>
#include <task.h>
#include <timers.h>

#include <array>
#include <atomic>
#include <cmath>
#include <functional>
#include <string_view>
#include <utility>

WiFiClient client;
HADevice ha_device("unused_test_device");
HAMqtt mqtt(client, ha_device);

#define SSTR(s) (#s)
#define STR(s) SSTR(s)

void setup_mqtt() {
  Serial1.println("mqtt.begin");
  IPAddress mqtt_addr;
  if (!mqtt_addr.fromString(STR(MQTT_HOST))) {
    Serial1.printf("%s is not a valid IP address\n", STR(MQTT_HOST));
    return;
  }
  Serial1.printf("mqtt.begin %s\n", mqtt_addr.toString().c_str());
  mqtt.begin(mqtt_addr, STR(MQTT_USER), STR(MQTT_PASSWORD));

  mqtt.begin(mqtt_addr, STR(MQTT_USER), STR(MQTT_PASSWORD));
  static auto mqtt_lock = xSemaphoreCreateMutex();
  xTaskCreate(
      [](void*) {
        while (true) {
          xSemaphoreTake(mqtt_lock, portMAX_DELAY);
          mqtt.loop();
          xSemaphoreGive(mqtt_lock);
          delay(5);
        }
      },
      "mqtt_loop", 1024, nullptr, 1, nullptr);

  xSemaphoreTake(mqtt_lock, portMAX_DELAY);
  while (!mqtt.isConnected()) {
    xSemaphoreGive(mqtt_lock);
    Serial1.println("Waiting for MQTT to connect");
    delay(1000);
    xSemaphoreTake(mqtt_lock, portMAX_DELAY);
  }
  xSemaphoreGive(mqtt_lock);
  Serial1.println("mqtt ready");
}

void setup_interrupt() {
  pinMode(15, INPUT_PULLUP);
  attachInterrupt(15, []() { Serial1.println("Interrupted!"); }, FALLING);
  Serial1.println("set up interrupt");
}

void setup() {
  Serial1.begin();
  Serial1.println("Wifi.begin");
  WiFi.begin(STR(WIFI_SSID), STR(WIFI_PASSWORD));
  while (!WiFi.isConnected()) {
    delay(100);
  }
  Serial1.println("Wifi ready");

  Serial1.println("Setting up test.");
#if INTERRUPT_SECOND != 0
  setup_mqtt();
  setup_interrupt();
#else
  setup_interrupt();
  setup_mqtt();
#endif
  pinMode(14, OUTPUT_4MA);
  digitalWrite(14, HIGH);
};

bool on = false;

void loop() {
  delay(1000);
  if (!on) {
    Serial1.println("Sending signal that should cause interrupt . . .");
  }

  digitalWrite(14, on);
  on = !on;
}
```

Here is the required platformio.ini to build it:

```
[common]

[env:pico]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
board = rpipicow
framework = arduino
board_build.core = earlephilhower
upload_protocol = cmsis-dap
debug_tool = cmsis-dap
lib_ldf_mode = chain+
build_flags = 
	-DPIO_FRAMEWORK_ARDUINO_ENABLE_RTTI
	-fstack-protector
	-std=c++20
	-DconfigTIMER_SERVICE_TASK_CORE_AFFINITY=0b1
	-DconfigUSE_TIMERS=1
	-DINTERRUPT_SECOND=0
	-DMQTT_HOST="${sysenv.HOMEASSISTANT_MQTT_HOST}"
	-DMQTT_USER="${sysenv.HOMEASSISTANT_MQTT_USERNAME}"
	-DMQTT_PASSWORD="${sysenv.HOMEASSISTANT_MQTT_PASSWORD}"
	-DWIFI_SSID="${sysenv.UCONTROLLER_WIFI_SSID}"
	-DWIFI_PASSWORD="${sysenv.UCONTROLLER_WIFI_PASSWORD}"
lib_deps = 
	knolleary/PubSubClient@^2.8
	dawidchyrzynski/home-assistant-integration@^2.1.0
	FreeRTOS
	SPI
	WiFi
```

earlephilhower, 2024-11-03:
The code looks illegal. You've made a task which can run on core1, but only core0 can do any Wi-Fi calls.   Please fix the code and try again.

 If after pinning the core irqs still stop, please make an MCVE without any third party libraries so it can be looked at.

jaguilar, 2024-11-03:
Thank you. It appears that the issue was caused by not setting the core affinity on the task that is doing the mqtt loop! Changing the code as follows resolved the issue:

```
  xTaskCreate(
      [](void*) {
        vTaskCoreAffinitySet(xTaskGetCurrentTaskHandle(), 0b1);  // Here.
        while (true) {
          xSemaphoreTake(mqtt_lock, portMAX_DELAY);
          mqtt.loop();
          xSemaphoreGive(mqtt_lock);
          delay(5);
        }
      },
      "mqtt_loop", 1024, nullptr, 1, nullptr);
```

Is there a debug setting I can turn on to proactively detect mistakes like this?

earlephilhower, 2024-11-03:
Great to hear, thanks for the update.

There presently is no checking, because on bare metal it's pretty obvious which core is running what.  On FreeRTOS, not so much.  I think it's possible to add.  All the `LWIP` calls are `gcc` wrapped in `cores/rp2040/lwip_wrap.cpp` so it may be a simple matter of adding an `assert(rp2040.cpuid == 0)` in every function.  But it wouldn't really do much to help folks pinpoint things.

Core1 would crash with a `panic`...but hard crashes on this chip are interesting.  CPU1 can panic and be in the `hardfault` state, but CPU0 can continue going on its merry way.  So, still a very odd partially working state even with the assert.

There's no real way to force the other core to `hardfault`, either.  You can use the intercore FIFO to make an IRQ which it can (and will) ignore (and which is used by FreeRTOS for task switching signaling).

I'll close this, but if you want to see if the `assert` or something else could help debug your case please do try your hand at a PR.  Thx!

closed 2024-11-03 by earlephilhower

jaguilar, 2024-11-03:
What if we made configASSERT something like

```
if (!(X)) {
  // Print error.
  vTaskSuspendAll();
  while(true) {}
}
```

That should make the problem very noticeable when a violation occurs. And a attaching with a debugger will easily reveal the source of the problem.

earlephilhower, 2024-11-03:
Unfortunately the core doesn't use FreeRTOS, so defining anything in there won't work.

You could create a new `__freertos_panic(const char *error)` call and explicitly call it, as long as you do the `weak` definitions the other FreeRTOS wrapper functions use.  Then call it under the usual guards in the wrapper functions.