earlephilhower/arduino-pico #2581: digitalRead() returning 0 on analoge Pins

mblees, 2024-11-04:
Hello there,

I have ran into an issue where I assumed that I can use analoge pins as digital pins as usual on other controllers. However when using the digitalRead function on an analog pin while there is a voltage that should be concidered as high it returns 0.

controller: W5100S-EVB-Pico
pins: 26, 27, 28
voltage: ~3,2V

expected return: 1
actual return: 0

I personally fixed this by manually implementing a threshold like this:
```cpp
if (!(pin == 26 || pin == 27 || pin == 28))
    {
        return digitalRead(pin);
    }
    else
    {
        if (analogRead(pin) > 512)
        {
            return 1;
        }
        else
        {
            return 0;
        }
    }
```

earlephilhower, 2024-11-04:
Analog pins read just fine digitally, but make sure you use `pinMode(A0, INPUT)` on them before `digitalRead`ing.  The digital portion of the pad is identical to the other pads.

If that doesn't fix it, we need a complete small MCVE that shows the problem.

mblees, 2024-11-05:
I tried but was unable to reproduce the issue using a simple script:

```
#include <Arduino.h>

#include "my_ethernet.h"

void setup()
{
  Serial.begin(115200);

  init_ethernet();
  while (!check_for_ethernet_link);
  while (!check_for_ethernet_connection);

  pinMode(27, INPUT); // analog Pin
  pinMode(13, INPUT); // digital Pin
}

void loop()
{
  Serial.print("Analog Pin: ");
  Serial.println(digitalRead(27));
  Serial.print("Digital Pin: ");
  Serial.println(digitalRead(13));
  Serial.println("");
  delay(1000);
}

```

It must have something to do with the way my electronics are set up or some other function I implemented is using the analog Pins. Therefore it seams to work as intended and my fix works for me .

closed 2024-11-05 by mblees