earlephilhower/arduino-pico #2583: `#include "pico/flash.h"` cannot find the file

jaguilar, 2024-11-05:
I can `#include <hardware/flash.h>`, but not `pico/flash.h`. It would be useful to have access to this API so I can use `flash_safe_execute`.

Test program:

```
#include <hardware/flash.h>
#include <pico/flash.h>

void setup() {}

void loop() {}
```

jaguilar, 2024-11-05:
From reading the comments in flash.c, maybe it's not that simple. The way they implemented it, it knows at compile time whether FreeRTOS is going to be running or not. And we don't know that in libpico.a. It looks like y'all have put together a dynamic version of the same concept in RP2040Support.h -- I'm going to see if I can just use that.

jaguilar, 2024-11-05:
I ended up with this:

```
struct FlashSafeHelper {
  FlashSafeHelper() {
    rp2040.idleOtherCore();
    noInterrupts();
  }
  ~FlashSafeHelper() {
    interrupts();
    rp2040.resumeOtherCore();
  }
};
```

And then I can do stuff like this:

```
  FlashSafeHelper helper;
  flash_range_erase(flash_offset, num_bytes);
```

If this seems sensible, I'm happy. If I'm barking up the wrong tree, please let me know.

earlephilhower, 2024-11-05:
I'd reverse the order of interrupt and freeze, I think

But at a higher level, why would you want to do this manually?  There are two different file systems and an eeprom emulator that "just work" already to persist data in flash safely.  Calling bare flash functions isn't going to make it faster because the bottleneck is the flash erase and write.

earlephilhower, 2024-11-05:
Oh, and the lack of flash helper is intended.  I actually implemented the core freeze before there was any sdk support so the whole infrastructure was built here before they went their way.  It's a fine way of doing things, but it's more general than we need in the core and not compatible with the way we do FreeRTIS and bare metal as you noted.

jaguilar, 2024-11-05:
I have a thingy that wants to write a counter value to the flash maybe a dozen times a day, and it only needs to write one bit at a time. In theory, I might want to support maintaining 10-50 counters, so maybe like 500 writes per day? From what I read of e.g. the EEPROM, it would erase the flash sector each time I commit, which would go through the 100k flash cycles available in less than a year. I don't know what sort of wear leveling the filesystems do.

But also, it's partly just cause I can, you know? :)

earlephilhower, 2024-11-05:
Yes, the emulator will erase and rewrite the whole sector.  It's not recommended and just there for really old sketches ported here. 

Little FS allocates at 256byte chunks and wear levels across the whole region.  If you have more than a basic flag you're updating, it's really useful and simple.

jaguilar, 2024-11-05:
I will definitely reach for LittleFS next time I'm not just incrementing a counter. This is to store the rolling code of some shades that move automatically, which is updated each time they are sent a command. It's a pain -- I mean that's what the manufacturers designed it to be. Thank you so much for the feedback.

closed 2024-11-05 by jaguilar

jaguilar, 2024-11-05:
> I'd reverse the order of interrupt and freeze, I think

For posterity -- I'm not sure this works with FreeRTOS? Not sure why but it seems like we hang if we do it. I think in FreeRTOS mode, interrupts() afterward is a noop since __freertos_idle_other_core calls portDISABLE_INTERRUPTS.