earlephilhower/arduino-pico #2603: esphome failes to compile with 4.2.0 due to symbol mangling in lock.cpp

afflux, 2024-11-13:
I get this linking error:

```
~/.platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/12.3.0/../../../../arm-none-eabi/bin/ld: ~/.platformio/packages/toolchain-rp2040-earlephilhower/bin/../lib/gcc/arm-none-eabi/12.3.0/../../../../arm-none-eabi/lib/thumb/libc.a(lib_a-findfp.o): in function `__sfp':
/workdir/repo/newlib/newlib/libc/stdio/findfp.c:175: undefined reference to `__retarget_lock_init_recursive'
```

Not sure which scenario is causing this exactly, but looking at the object files, I see these symbols defined:

```
00000001 T _Z20__retarget_lock_initP6__lock
[...]
00000001 T _Z30__retarget_lock_init_recursiveP6__lock
```

I tried compiling with extern C on these symbols, but I get this error:

```
~/.platformio/packages/framework-arduinopico/cores/rp2040/lock.cpp:103:6: error: conflicting declaration of C function 'void __retarget_lock_init(_LOCK_T)'
  103 | void __retarget_lock_init(_LOCK_T lock) {
      |      ^~~~~~~~~~~~~~~~~~~~
In file included from ~/.platformio/packages/framework-arduinopico/cores/rp2040/lock.cpp:26:
~/.platformio/packages/toolchain-rp2040-earlephilhower/arm-none-eabi/include/sys/lock.h:41:13: note: previous declaration 'void __retarget_lock_init(__lock**)'
   41 | extern void __retarget_lock_init(_LOCK_T *lock);
      |             ^~~~~~~~~~~~~~~~~~~~
~/.platformio/packages/framework-arduinopico/cores/rp2040/lock.cpp:117:6: error: conflicting declaration of C function 'void __retarget_lock_init_recursive(_LOCK_T)'
  117 | void __retarget_lock_init_recursive(_LOCK_T lock) {
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
~/.platformio/packages/toolchain-rp2040-earlephilhower/arm-none-eabi/include/sys/lock.h:43:13: note: previous declaration 'void __retarget_lock_init_recursive(__lock**)'
   43 | extern void __retarget_lock_init_recursive(_LOCK_T *lock);
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

It compiles fine when I change the function signatures to accept `_LOCK_T*`, but I haven't had the chance to test yet. Specifically, I'm not sure if the casts to `mutex_t*` and `recursive_mutex_t*` are still ok or if we need to dereference, because I'm unfamiliar with this API.

earlephilhower, 2024-11-13:
You are not running the latest p.io integration and using the wrong compiler.  Please update @maxgerhardt 's integration script to get the proper tool chain.

closed 2024-11-13 by earlephilhower

afflux, 2024-11-13:
Sorry about that - you are completely right. Platform.IO is still a mystery to me :-(

earlephilhower, 2024-11-13:
No worries! 

> ....Platform.IO is still a mystery to me :-(

To you and me both!  😆