earlephilhower/arduino-pico #2608: Support multiple CYW43 driver configurations for RP2350

earlephilhower, 2024-11-14:
I did some investigation and there are several things that need to fall into place to get CYW43 running on non-RP2040-PicoW boards.

`libpico` needs to be updated on the RP2350 builds to set the proper config and libraries.  Right now only the RP2040 will include the cyw43 driver stack.  `tools/libpico/Cmakelists.txt` and rebuild, which leads to the first issue.

The 2.0.0 SDK has hardcoded defines for the pinout of the CYW43 chip.  On the RP2040/PicoW that's no big deal, there is only one physical setup to worry about.  On the RP2350 with multiple Wifi PCBs, that will need to be abstracted (we need to build the SDK using Cmake outside of Arduino, so only build a single .A for all CPU archs, not for all PCBs).  Defines like `CYW43_PIN_WL_HOST_WAKE` and `CYW43_PIN_WL_SDIO_1`, etc.  This can probably be handled with some careful `cores/rp2040/sdkoverrides` but I'd need to think about how to safely implement it...

Once that's running, then an additional define, `PICO_CYW43_SUPPORTED` needs to be added to the `extra` list in `tools/makeboards.py` for the PicoW and RP2350 boards w/the CYW43 chip.  Maybe including the physical `CYW43_PIN_xxxx` from above.  And then an almost search/replace of `s/ifdef ARDUINO_RASPBERRY_PI_PICO_W/ifdef PICO_CYW43_SUPPORTED/` through the libraries and you're within spitting distance...

So, I see a couple options.  You may want to consider going radio-free to start, and then I can work on my own to get the CYW43 stuff up later.  Or, I can start a new dev branch/PR with the changes above and you can rebase this PR to that branch.  When it's good, we can merge this Pr into that branch, then merge that branch to `master`.  Whatever makes sense for you, just let me know...

--edit-- One other option would be to split out the cyw43 Cmake into its own library and pull that in on an individual basis.  If the constant definitions become too much of a hangup, then a hardcoded list of pinouts in the `CMakefile.txt` could be built and then linked in via `makeboards.py` adding the appropriate flags.

_Originally posted by @earlephilhower in https://github.com/earlephilhower/arduino-pico/issues/2605#issuecomment-2476743384_

maxgerhardt, 2024-11-15:
Related to https://github.com/earlephilhower/arduino-pico/issues/2495

earlephilhower, 2024-11-15:
Looks like there is a new define in the `pico-sdk` `develop` branch, `CYW43_PIN_WL_DYNAMIC`, which will let the pins be passed in before the `cyw43_init` call.  With luck, that lets us keep a single binary with the variant files passing in the appropriate pins at startup.  :crossed_fingers:

commit 1780acd9 (2024-11-15): WIP - Support CYW43 boards on RP2350

Using pico-sdk develop branch with minor patch, add in support for the
CYW43-based WiFi boards on the RP2350.

Fixes #2608

Rolls in dynamic SPI divider #2600
PR #2616 (2024-11-19): Support CYW43 boards on RP2350
Using pico-sdk develop branch with minor patch, add in support for the CYW43-based WiFi boards on the RP2350.

Fixes #2608