earlephilhower/arduino-pico #2611: Issue with I2S Input from V4220M/CS4220 Audio Codec

kurogedelic, 2024-11-15:
Hello.
First, I should mention that I'm not an I2S expert, so I might be missing something fundamental. I'm trying to receive audio data from a V4220M/CS4220 codec using the Arduino-Pico I2S library, but I'm experiencing issues with the input data. Interestingly, I2S output to the same codec works perfectly fine.

Hardware Setup:
- Raspberry Pi Pico (RP2040)
- V4220M/CS4220 Audio Codec (I've tested multiple modules, so I don't believe it's a hardware failure)
- Connections:
  - GPIO20 -> SCLK (pin5)
  - GPIO21 -> LRCK (pin4)
  - GPIO15 -> SDIN (pin9)
  - GPIO18 -> MCLK (pin3)
  - GPIO19 -> RST (pin27)

Clock Configuration:
- Sample Rate: 48kHz
- BCLK: 3.072MHz 
- MCLK: 12.288MHz (256Fs)
- LRCK: 48kHz
(confirmed with oscilloscope)

Code:
```cpp
I2S i2s(INPUT);
const int32_t sampleRate = 48000;

void setup() {
    // V4220M reset sequence
    pinMode(PIN_I2S_RST, OUTPUT);
    digitalWrite(PIN_I2S_RST, LOW);
    delay(10);
    digitalWrite(PIN_I2S_RST, HIGH);
    delay(50);

    // I2S configuration
    i2s.setBCLK(PIN_I2S_BCLK);
    i2s.setDATA(PIN_I2S_DIN);
    i2s.setBitsPerSample(32);
    i2s.setBuffers(4, 32);
    i2s.setMCLK(PIN_I2S_MCLK);
    i2s.setMCLKmult(256);

    if (!i2s.setSysClk(sampleRate)) {
        Serial.println("Failed to set system clock!");
        while(1);
    }

    if (!i2s.begin(sampleRate)) {
        Serial.println("Failed to start I2S!");
        while(1);
    }
}
```

The received data shows an unusual pattern:
```
Time,Raw,Hex,Voltage,Debug
82087,14,0x00000E,0.000,0x00000ED0
82087,262141,0x03FFFD,0.062,0x03FFFD74
82087,7,0x000007,0.000,0x00000728
82088,262121,0x03FFE9,0.062,0x03FFE9A8
```

The data seems to alternate between very small values (0x00000X) and larger values (0x03FFXX), which doesn't look like normal audio data. I've tried:
1. Different bit depths (24/32)
2. Different I2S formats (DIF0/DIF1 configurations on V4220M)
3. Checking clock signals with oscilloscope
4. Multiple codec chips to rule out hardware issues

The fact that output works perfectly suggests the clock setup is correct. Am I missing something important in the I2S input configuration? Since I'm relatively new to I2S, there might be some fundamental concept I'm overlooking.

Any help or suggestions would be greatly appreciated!

earlephilhower, 2024-11-15:
Please try the `git master` version, not the official release.  There was an I2S input timing change to better match the spec which affects the left/right word clock (which would possibly cause data shifts).  See the docs for full info, it's simple to get `git` running for most folks.

#2592 is the specific PR.

kurogedelic, 2024-11-18:
I have tested with the latest , but the issue remains unchanged. 

Key findings:
1. Testing with multiple V4220M/CS4220 chips shows the same behavior across all devices 
2. DAC functionality works perfectly fine successfully.
3. DIF settings and I2S format configuration should be correct according to the datasheet (DIF0=DIF1=0 for I2S mode).

Current conclusion:
The issue might not be related to the Arduino-Pico I2S library itself. The incomming I2S data is random.

Next steps:
1. Will test with a different I2S ADC (PCM1808 and other) to verify if I2S reception works correctly with other devices.
2. Will also test master mode operation once I obtain the appropriate crystal.

Tested code:
```cpp
#include <I2S.h>

#define PIN_I2S_BCLK 20
#define PIN_I2S_LRCLK (PIN_I2S_BCLK + 1)
#define PIN_I2S_DIN 15
#define PIN_I2S_MCLK 18
#define PIN_I2S_RST 19

I2S i2s(INPUT);

void setup() {
  Serial.begin(115200);
  while (!Serial) delay(100);

  // Reset V4220M
  pinMode(PIN_I2S_RST, OUTPUT);
  digitalWrite(PIN_I2S_RST, LOW);
  delay(10);
  digitalWrite(PIN_I2S_RST, HIGH);
  delay(50);

  // I2S setup
  i2s.setSysClk(48000);
  i2s.setMCLK(PIN_I2S_MCLK);
  i2s.setBCLK(PIN_I2S_BCLK);
  i2s.setDATA(PIN_I2S_DIN);
  i2s.setBitsPerSample(32);
  i2s.setMCLKmult(256);
  i2s.begin(48000);
}

void loop() {

  if (i2s.available() > 0) {
    // L and R
    Serial.printf("0x%08X,0x%08X,\n",
                  i2s.read(), i2s.read());
  }
}
```

Serial monitor:
```
15:29:54.776 -> 0xFF86E01F,0xFF02801F,
15:29:54.776 -> 0xFF1B801F,0xFFA4A01F,
15:29:54.776 -> 0xFFA6C01F,0xFEF7601F,
15:29:54.776 -> 0xFF22001F,0xFFA2C01F,
15:29:54.776 -> 0xFF96001F,0xFF0E401F,
15:29:54.776 -> 0xFEED001F,0xFFAA201F,
15:29:54.776 -> 0xFFC0601F,0xFF1E801F,
15:29:54.776 -> 0xFF47C01F,0xFF95801F,
15:29:54.776 -> 0xFFCB601F,0xFF17601F,
15:29:54.776 -> 0xFF0B601F,0xFF95C01F,
15:29:54.776 -> 0xFFC1401F,0xFF59A01F,
15:29:54.776 -> 0xFF49401F,0xFF94401F,
15:29:54.776 -> 0xFF83801F,0xFF48801F,
15:29:54.776 -> 0xFF5FC01F,0xFFBCC01F,
15:29:54.776 -> 0xFFBA801F,0xFF3B401F,
15:29:54.776 -> 0xFF08C01F,0xFFB9E01F,
15:29:54.776 -> 0xFFB0A01F,0xFF40001F,
15:29:54.776 -> 0xFF3D001F,0xFFBD401F,
15:29:54.776 -> 0xFFB0601F,0xFF72001F,
15:29:54.776 -> 0xFF89601F,0xFF98E01F,
15:29:54.776 -> 0xFFBFA01F,0xFF61201F,
15:29:54.776 -> 0xFF6F601F,0xFFB3601F,
15:29:54.776 -> 0x01FE601F,0xFFB40000,
15:29:54.776 -> 0xFFB42000,0x01D2C01F,
```

kurogedelic, 2024-11-21:
I got more V4220Ms, but all had the same results.
All still had the DAC working, but the ADC not working.
I gave up on thinking that the Pico was to blame.
I tried to catch it with Digital Analyzer, but it was too difficult for me.
The screenshot below is the V4220M -> Pico data line, and it seems that only the LRCLK is stable. Even if there is jitter in the other clocks, the Data Signal is not synchronized with anyone!
Even if there is jitter in the clock, the DAC (output) is not a problem at all.
I think I'll stop pursuing this any further.
I look forward to someone trying to use V4220M/CS4220 with Arduino-Pico in the future.

[image]
[image]

closed 2024-11-21 by earlephilhower