earlephilhower/arduino-pico #2612: Please reopen #2572

djaus2, 2024-11-15:
Still happening.

closed 2024-11-15 by earlephilhower

earlephilhower, 2024-11-15:
Also, when opening something there, the IDE is really verbose on the console.  If you can grab the logs it dumps that would probably be helpful and pinpoint what's going on there.

earlephilhower, 2024-11-15:
Over here, I get something like this when I start up and it auto-d/l's the multiple JSONs I've added...
````
....
2024-11-15T15:15:04.964Z daemon INFO time="2024-11-15T07:15:04-08:00" level=info msg="Updating index" url="https://espressif.github.io/arduino-esp32/package_esp32_index.json"
2024-11-15T15:15:04.964Z daemon INFO time="2024-11-15T07:15:04-08:00" level=info msg="Starting download" url="https://espressif.github.io/arduino-esp32/package_esp32_index.json"
2024-11-15T15:15:04.965Z root INFO core-client-provider [platform-index] Downloading index: package_esp32_index.json

2024-11-15T15:15:05.149Z root INFO core-client-provider [platform-index] Downloading index: package_rp2040_index.json

2024-11-15T15:15:05.261Z daemon INFO time="2024-11-15T07:15:05-08:00" level=info msg="Updating index" url="https://github.com/earlephilhower/arduino-pico/releases/download/global/package_rp2040_index.json"
time="2024-11-15T07:15:05-08:00" level=info msg="Starting download" url="https://github.com/earlephilhower/arduino-pico/releases/download/global/package_rp2040_index.json"
2024-11-15T15:15:05.276Z daemon INFO time="2024-11-15T07:15:05-08:00" level=info msg="Adding libraries dir" dir=/home/earle/.arduino15/libraries location=ide
time="2024-11-15T07:15:05-08:00" level=info msg="Adding libraries dir" dir=/home/earle/.arduino15/packages/esp32/hardware/esp32/3.0.1/libraries location=platform
time="2024-11-15T07:15:05-08:00" level=info msg="Adding libraries dir" dir=/home/earle/Arduino/libraries location=user
2024-11-15T15:15:05.767Z daemon INFO time="2024-11-15T07:15:05-08:00" level=info msg="Checking signature" index=/home/earle/.arduino15/package_index.json signatureFile=/home/earle/.arduino15/package_index.json.sig trusted=true
2024-11-15T15:15:05.769Z daemon INFO time="2024-11-15T07:15:05-08:00" level=info msg="Missing signature file" index=/home/earle/.arduino15/package_esp8266com_index.json
2024-11-15T15:15:05.769Z daemon INFO time="2024-11-15T07:15:05-08:00" level=info msg="Missing signature file" index=/home/earle/.arduino15/package_esp32_index.json
2024-11-15T15:15:05.772Z daemon INFO time="2024-11-15T07:15:05-08:00" level=info msg="Missing signature file" index=/home/earle/.arduino15/package_rp2040_index.json
2024-11-15T15:15:05.772Z daemon INFO time="2024-11-15T07:15:05-08:00" level=info msg="Loading hardware from: /home/earle/.arduino15/packages"
2024-11-15T15:15:05.772Z daemon INFO time="2024-11-15T07:15:05-08:00" level=info msg="Loading package arduino from: /home/earle/.arduino15/packages/arduino/hardware"
2024-11-15T15:15:05.773Z daemon INFO time="2024-11-15T07:15:05-08:00" level=info msg="Adding monitor tool" protocol=serial tool="builtin:serial-monitor"
time="2024-11-15T07:15:05-08:00" level=info msg="Loaded platform" platform="arduino:avr@1.8.6"
2024-11-15T15:15:05.774Z daemon INFO time="2024-11-15T07:15:05-08:00" level=info msg="Adding monitor tool" protocol=serial tool="builtin:serial-monitor"
....
````