earlephilhower/arduino-pico #2614: Custom PIO program with arduino-pico i2s buffer write

myreauks, 2024-11-15:
Hi!

I have a program with a pio program used to drive some led strips. The program I use can be found here:
[https://github.com/ErichStyger/mcuoneclipse/blob/master/Examples/RaspberryPiPico/pico_LedCube/src/ws2812.pio.h](url)

I set the pio program up with DMA like this:
```
void ledSetup() {

  // Initialize the PIO program for WS2812 parallel output
  uint offset = pio_add_program(pio, &ws2812_parallel_program);
  ws2812_parallel_program_init(pio, sm, offset, PIO_PIN_BASE, NUM_STRIPS, LED_DATA_RATE);

  // Set up DMA for each strip
  dma_channel = dma_claim_unused_channel(true);
  dma_channel_config c = dma_channel_get_default_config(dma_channel);
  channel_config_set_transfer_data_size(&c, DMA_SIZE_32);  // 32-bit data size
  channel_config_set_read_increment(&c, true);             // Increment source address
  channel_config_set_write_increment(&c, false);           // Do not increment destination
  channel_config_set_dreq(&c, pio_get_dreq(pio, sm, true)); // Use the PIO's data request

  dma_channel_configure(
    dma_channel,
    &c,
    &pio->txf[sm],              // PIO TX FIFO as destination
    pixel_buffer,       // Source buffer for this strip
    NUM_LEDS_PER_STRIP * 8,         // Number of transfers
    false                       // Do not start yet
  );

  irq_add_shared_handler(DMA_IRQ_0, dma_complete_handler, PICO_SHARED_IRQ_HANDLER_DEFAULT_ORDER_PRIORITY);
  irq_set_enabled(DMA_IRQ_0, true);

}

void dma_complete_handler() {
    // Check which DMA channel triggered the interrupt
    uint32_t interrupt_flags = dma_hw->ints0;  // Get the interrupt flags

    // If the interrupt was triggered by our channel, clear it and reset the status
    if (interrupt_flags & (1u << dma_channel)) {
        dma_hw->ints0 = 1u << dma_channel;  // Clear the interrupt flag for our channel

        // Mark the DMA as no longer busy, allowing the next transfer to start
        dma_busy = false;
    }
}
```

It works fine on its own and with the i2s 'size_t write(uint8_t/int8_t/int16_t/int32_t)' like so:

```
void updateAudio() {
  if(count < length) {
    short s = (gongu[count]); // Grab sample value.
    if(count < 200) s = s * i/200;  // Create fade in
    for(int i=0;i<4;i++) i2s.write(s);
    count++;
  }

  if(count >= length) {
    count = 0;
  }
}
```

But the previous setup causes crackle in the audio when the program is doing other stuff and the updateAudio() cannot be called often enough. I thought that this would be fixed by using the 'size_t write(const uint8_t *buffer, size_t size)'. This works fine on its own, but when I try to use it with the ws2812 pio program the pico crashes as soon as the write function is called here:

```
void updateAudio() {
  if(!playing) { return; }

  size_t amount = i2s.write(ptr, sizeof(ptr));
  ptr += amount;
  toPlay -= amount;

  if(toPlay == 0) {
    i2s.end();
    playing = false;
  }

}
```

Any ideas what could be happening?
Best,
Miro

myreauks, 2024-11-15:
Hi! Thanks for the fast reply! 

Here is a repository with a complete example: [https://github.com/myreauks/arduino-pico-i2s-ws2812](url)
Compiled with arduino-pico 4.2.0 and the Generic RP2040. Hardware is a custom board with RP2040 and a MAX98357 amp.
In the example, updateAudio() works if ledSetup() and updateLights() is commented out and vice versa. With all active, a crash happens.

I just got a PicoProbe yesterday and tried to debug my setup with it. It was causing a crash even with a working version of the program so I'm not sure what's going on with that one.

earlephilhower, 2024-11-21:
Any update, @myreauks ?

myreauks, 2024-11-22:
I never managed to figure out the cause of the crash. I got my program working by copying the principle with which the buffer write works. I am filling the buffer with my own while loop and for whatever reason, this is not causing a crash.
```
void updateAudio() {
  if(!playing) return;

  while(i2s.availableForWrite() > 1) {
    unsigned int s = audioData[pos]; // Grab sample value.
    if (pos < 200) s = s * pos/200;  // Create fade in
    i2s.write32((int32_t)s,(int32_t)s); pos++;
  }

  if(pos >= toPlay) {
    playing = false; 
    i2s.end();
  }
}
```

For now, I'll go with this solution. For me it is fine to close this issue for now. At some point it would be nice to figure out the reason for the crash though.

closed 2024-11-22 by earlephilhower