earlephilhower/arduino-pico #2617: WiFiUdp flush() purpose

metanav, 2024-11-16:
What is the intended behavior of the flush() function in the WiFiUdp? It calls endPacket() function. 

https://github.com/earlephilhower/arduino-pico/blob/master/libraries/WiFi/src/WiFiUdp.cpp#L226

I think it should be clearing up the buffer instead of calling endPacket() function.

I find this as an issue because I was getting an empty packet in a code that calls endPacket() and then flush().

maxgerhardt, 2024-11-16:
Maybe the API is intended for in-between flushes?
```cpp
udp_client.beginPacket(locator->address, locator->port);
size_t sent = udp_client.write(buf, len);
udp_client.flush(); // send this important data right now
size_t sent2 = udp_client.write("other data", 10); // continue writing
udp_client.endPacket(); // actual end of packet 
```

earlephilhower, 2024-11-16:
`flush` exists on the IP stack because it's part of `Stream` (IIRC) which requires an implementation of it.   "`flush` pushes out whatever data is in the pipe to the wire" is the general idea here.  `Serial1.flush()` makes sure anything in the HW FIFO has moved to the transmitter, for example.

For UDP, `endPacket` also guarantees it's sent to the Ethernet stack.  So a `flush` isn't really needed but should be a no-op, not a 0-byte UDP ping, after an `endPacket`.  So I think that may be a bug.

But conceptually, especially with UDP not having any advanced packet recombination like TCP, I am not sure what the use case of calling it would be.  We don't have a connection you can partially send data down, UDP is always built packet-by-packet here.

commit bda12cd7 (2024-11-17): WiFiUDP only emit a packet on flush if not ended (#2618)

A flush() on a packet that's already been sent should be a no-op, not
send a 0-byte UDP packet.  Track when an outgoing packet is dirty and
only endPacket()/transmit it when so.

Fixes #2617
closed 2024-11-17 by earlephilhower