earlephilhower/arduino-pico #2622: `detachInterrupt()` within ISR leads to FreeRTOS fatal error

GUVWAF, 2024-11-19:
After updating to version 4.2.1 of the core, we’re hitting a sporadic freeze. It looks like it is a side-effect of #2558, since when undoing that, it doesn’t happen.

It happens when calling `detachInterrupt()` from within an ISR while using `xSemaphoreTake()`. The issue seems to be caused by a memory deallocation performed by `std::map::erase()`, see the call stack below: 
```
sleep_until@0x10008f46 (/sleep_until.dbgasm:43)
sleep_us@0x10008fa0 (/sleep_us.dbgasm:13)
sleep_ms@0x10008fba (/sleep_ms.dbgasm:8)
rtosFatalError@0x10005e90 (/home/thijs/.platformio/packages/framework-arduinopico/libraries/FreeRTOS/src/variantHooks.cpp:356)
xQueueSemaphoreTake@0x10003bda (/home/thijs/.platformio/packages/framework-arduinopico/libraries/FreeRTOS/lib/FreeRTOS-Kernel/queue.c:1675)
xQueueTakeMutexRecursive@0x10003d12 (/home/thijs/.platformio/packages/framework-arduinopico/libraries/FreeRTOS/lib/FreeRTOS-Kernel/queue.c:837)
__freertos_recursive_mutex_take@0x10005c72 (/home/thijs/.platformio/packages/framework-arduinopico/libraries/FreeRTOS/src/variantHooks.cpp:81)
__retarget_lock_acquire_recursive@0x1000707e (/home/thijs/.platformio/packages/framework-arduinopico/cores/rp2040/lock.cpp:150)
__malloc_lock@0x10018fd4 (/__malloc_lock.dbgasm:4)
_free_r@0x10017f2e (/_free_r.dbgasm:11)
free@0x10018356 (/free.dbgasm:6)
__wrap_free@0x100075c6 (/home/thijs/.platformio/packages/framework-arduinopico/cores/rp2040/malloc-lock.cpp:112)
operator delete@0x10009896 (Unknown Source:28)
std::__new_allocator<std::_Rb_tree_node<std::pair<unsigned char const, CBInfo> > >::deallocate@0x1000780a (/home/thijs/.platformio/packages/toolchain-rp2040-earlephilhower/arm-none-eabi/include/c++/14.2.0/bits/new_allocator.h:156)
std::allocator_traits<std::allocator<std::_Rb_tree_node<std::pair<unsigned char const, CBInfo> > > >::deallocate@0x1000780a (/home/thijs/.platformio/packages/toolchain-rp2040-earlephilhower/arm-none-eabi/include/c++/14.2.0/bits/alloc_traits.h:513)
std::_Rb_tree<unsigned char, std::pair<unsigned char const, CBInfo>, std::_Select1st<std::pair<unsigned char const, CBInfo> >, std::less<unsigned char>, std::allocator<std::pair<unsigned char const, CBInfo> > >::_M_put_node@0x1000780a (/home/thijs/.platformio/packages/toolchain-rp2040-earlephilhower/arm-none-eabi/include/c++/14.2.0/bits/stl_tree.h:563)
std::_Rb_tree<unsigned char, std::pair<unsigned char const, CBInfo>, std::_Select1st<std::pair<unsigned char const, CBInfo> >, std::less<unsigned char>, std::allocator<std::pair<unsigned char const, CBInfo> > >::_M_drop_node@0x1000780a (/home/thijs/.platformio/packages/toolchain-rp2040-earlephilhower/arm-none-eabi/include/c++/14.2.0/bits/stl_tree.h:630)
std::_Rb_tree<unsigned char, std::pair<unsigned char const, CBInfo>, std::_Select1st<std::pair<unsigned char const, CBInfo> >, std::less<unsigned char>, std::allocator<std::pair<unsigned char const, CBInfo> > >::_M_erase@0x1000780a (/home/thijs/.platformio/packages/toolchain-rp2040-earlephilhower/arm-none-eabi/include/c++/14.2.0/bits/stl_tree.h:1934)
std::_Rb_tree<unsigned char, std::pair<unsigned char const, CBInfo>, std::_Select1st<std::pair<unsigned char const, CBInfo> >, std::less<unsigned char>, std::allocator<std::pair<unsigned char const, CBInfo> > >::clear@0x10007986 (/home/thijs/.platformio/packages/toolchain-rp2040-earlephilhower/arm-none-eabi/include/c++/14.2.0/bits/stl_tree.h:1251)
std::_Rb_tree<unsigned char, std::pair<unsigned char const, CBInfo>, std::_Select1st<std::pair<unsigned char const, CBInfo> >, std::less<unsigned char>, std::allocator<std::pair<unsigned char const, CBInfo> > >::_M_erase_aux@0x100079da (/home/thijs/.platformio/packages/toolchain-rp2040-earlephilhower/arm-none-eabi/include/c++/14.2.0/bits/stl_tree.h:2503)
std::_Rb_tree<unsigned char, std::pair<unsigned char const, CBInfo>, std::_Select1st<std::pair<unsigned char const, CBInfo> >, std::less<unsigned char>, std::allocator<std::pair<unsigned char const, CBInfo> > >::erase@0x10007a12 (/home/thijs/.platformio/packages/toolchain-rp2040-earlephilhower/arm-none-eabi/include/c++/14.2.0/bits/stl_tree.h:2517)
std::map<unsigned char, CBInfo, std::less<unsigned char>, std::allocator<std::pair<unsigned char const, CBInfo> > >::erase (this=0x20001910 <_map>, __x=@0x20041f27: 9 '\t') at /home/thijs/.platformio/packages/toolchain-rp2040-earlephilhower/arm-none-eabi/include/c++/14.2.0/bits/stl_map.h:1117
_detachInterruptInternal (pin=<optimized out>, pin@entry=9 '\t') at /home/thijs/.platformio/packages/framework-arduinopico/cores/rp2040/wiring_private.cpp:113
detachInterrupt (pin=9 '\t') at /home/thijs/.platformio/packages/framework-arduinopico/cores/rp2040/wiring_private.cpp:170
isr () at src/main.cpp:9
CBInfo::callback (this=this@entry=0x2001565c) at /home/thijs/.platformio/packages/framework-arduinopico/cores/rp2040/wiring_private.cpp:73
_gpioInterruptDispatcher (gpio=<optimized out>, events=<optimized out>) at /home/thijs/.platformio/packages/framework-arduinopico/cores/rp2040/wiring_private.cpp:105
gpio_default_irq_handler () at /home/earle/Arduino/hardware/pico/rp2040/pico-sdk/src/rp2_common/hardware_gpio/gpio.c:165
irq_handler_chain_slots ()
```
Here’s a MCVE. When touching GPIO 9 a couple times using a jumper from GPIO 8, it will run into the error above:
```cpp
#include <Arduino.h>
#include <FreeRTOS.h>
#include <semphr.h>

SemaphoreHandle_t semaphore;
int x = 0;
int last_x = 0;
void isr() {
  detachInterrupt(9);
  xSemaphoreGiveFromISR(semaphore, NULL);
  x++;
}

void setup() {
  Serial.begin(115200);
  semaphore = xSemaphoreCreateBinary();
  pinMode(9, INPUT_PULLDOWN);
  pinMode(8, OUTPUT);
  digitalWrite(8, HIGH);  
  attachInterrupt(9, isr, RISING);
}

void loop() {
  xSemaphoreTake(semaphore, pdMS_TO_TICKS(1000));
  attachInterrupt(9, isr, RISING);
  if (last_x != x) {
    Serial.printf("%d", x);
    Serial.println();
    last_x = x;
  }
}
```

While undoing #2558 fixes it, a more proper solution would probably be to avoid the dynamic memory (de)allocation of `std::map`, but that would require turning `_map` into a static array, and I’m not sure if that’s a feasible/suitable solution.

earlephilhower, 2024-11-19:
Thanks for the MCVE, that helps a lot.

`std::map` was used because there will probably only be a few elements, but given the fact that it is an interrupt handler doing memory (de)allocations really isn't a good idea no matter what.

With rejiggering it comes to <~256 bytes total including housekeeping to make it static and sane, so I'll clean it up and call it good enough.

GUVWAF, 2024-11-19:
Thanks again for the very quick fix! 
I confirmed that the MCVE works fine now.

commit 315c1358 (2024-11-19): Avoid memory (de)allocation in GPIO ISR handler

Replace std::map, which can `new` and `delete` elements of the tree,
with a statically allocated array and housekeeping.

Fixes #2622
PR #2623 (2024-11-19): Avoid memory (de)allocation in GPIO ISR handler
Replace std::map, which can `new` and `delete` elements of the tree, with a statically allocated array and housekeeping.

Fixes #2622
commit 322a1af6 (2024-11-19): Avoid memory (de)allocation in GPIO ISR handler (#2623)

Replace std::map, which can `new` and `delete` elements of the tree,
with a statically allocated array and housekeeping.

Fixes #2622
closed 2024-11-19 by earlephilhower