earlephilhower/arduino-pico #2632: WiFi.localIP() hangs in Nano RP2040 Connect when analogRead calls are present

danluca, 2024-11-25:
Environment:
* Ubuntu 24.04
* VSCode with PlatformIO
* Nano RP2040 Connect
* WiFiNINA 1.8.14
* Arduino-Pico 4.2.1

Invoking either `WiFi.localIP()` or `WiFi.gatewayIP()` (same SPI command - 33) when somewhere else in the code there is a `analogRead(A0)` hangs-up the board.

Simplified code structure:
```c++
#define IP_DNS 8,8,8,8                  // Google DNS
#define IP_GW 192,168,0,1           // default gateway (router)
#define IP_SUBNET 255,255,255,0     // usual subnet mask

void setup() {
  WiFi.config({IP_ADDR}, {IP_DNS}, {IP_GW}, {IP_SUBNET});
  while (wifiStatus != WL_CONNECTED) {
      // Connect to WPA/WPA2 network
      wifiStatus = WiFi.begin(ssid, pass);
      // wait 10 seconds for connection to succeed:
      delay(10000);
    }
  Serial.printf("IP Address               : %s\n", WiFi.localIP().toString().c_str());    // <<-- hangs here
}

void loop() {
  Serial.printf("Voltage at A0 = %d", analogRead(A0));
  delay(500);
}
```

Note that execution hasn't reached the analogRead call when it hangs.

Attached is a complete example with the analogRead command enabled (file net_setup.cpp line 68) - the program hangs; comment out the statement and it will work (edit the secrets.h file for proper WiFi credentials). [pico-nina-hangs.zip](https://github.com/user-attachments/files/17898048/pico-nina-hangs.zip)

I was able to avoid the `localIP` command and confirmed (in the larger app where this example is derived from) that the WiFi system **works well otherwise** - UDP calls to NTP, hosting a plain http web server and processing static files, JSON, etc. I have not tried any TLS setup or calls.

This is related to issue #973 - not sure I fully understand the recommended solution in there, if there was one.

Ideas? Thank you for looking into this.

dquadros, 2024-11-25:
Not sure if this has anything to do with ADC. The following code (which, by the way, runs fine with the Arduino Mbed OS core) will halt the second time WiFi.begin() is called (with the Arduino core it connects on the first try). 

```
#include <WiFi.h>

#define ssid "xxxxx" 
#define pass "xxxx"

void setup() {
  while (!Serial) {
    delay(100);
  }
  Serial.begin(115200);
  Serial.println("\n\nStarting");
  delay(3000);

  uint8_t wifiStatus = WiFi.status();
  while (wifiStatus != WL_CONNECTED) {
      // Connect to WPA/WPA2 network
      Serial.println("Trying to connect...");
      wifiStatus = WiFi.begin(ssid, pass);
      // wait 10 seconds for connection to succeed:
      delay(10000);
  }
  Serial.println("Connected");
  delay(3000);
  if (WiFi.localIP()) {
    Serial.print ("IP: ");
    Serial.println (WiFi.localIP().toString().c_str());
  }
}

void loop() {
  //Serial.print("Voltage at A0 = ");
  //Serial.println (analogRead(A0));
  delay(500);
}
```

earlephilhower, 2024-11-25:
While I have heard others say `WiFiNINA` "just works" for most things, I don't own one nor have I ever tried one.

FWIW, I find it unlikely that it would work because of basic things, like LWIP locking, which here follow the PicoW and SDK norms.  For example, I assume it's got an SPI or UART interface that gets an interrupt when a packet comes in.  The IRQ handler then reads the packet from the ESP32 and sends the packet to LWIP's processing queue.  But, if the IRQ comes in while another LWIP call is running, that will cause corruption and eventual failure (LWIP does not re-enter or multithread).

What you might look into is converting the ESP32's code from whatever Arduino is running to the ESPHost version.  If that's possible then you can use the `lwip_ESPHost` driver here to work with the ecosystem.  See the readme for more info:  https://github.com/Networking-for-Arduino/ESPHost/tree/80c0fa89b15724490f8bacfee5ac1d40c46e8683

JAndrassy, 2024-11-25:
@dquadros I guess you want `#include <WiFiNINA.h>` as on the Mbed core

dquadros, 2024-11-25:
> @dquadros I guess you want `#include <WiFiNINA.h>` as on the Mbed core. `#include <WiFi.h>` is the WiFi library bundled with the core. that library can't support NINA firmware.

Makes sense, I may try that later. For now the Nano RP2040 is back in the drawer.  Got tired to have to jumper two pins to get into the bootloader.