earlephilhower/arduino-pico #2644: LED_BUILTIN not working on Pico2 W

skpang, 2024-11-27:
I'm trying to blink the on board LED on the Pico2 W but the LED is not coming on.

pinMode(LED_BUILTIN, OUTPUT);
digitalWrite(LED_BUILTIN, HIGH);

I'm using library v4.3.0

Does it need updating for the Pico2 W ?

earlephilhower, 2024-11-27:
Please double check you have selected the Pico 2 W board from the menus, and also please run the `blink.ino` example.  I forget the exact polarity of the LED (high=on or high=off) on that board.

skpang, 2024-11-27:
This is the code I'm using : \void setup() {
  // initialize digital pin LED_BUILTIN as an output.
  pinMode(LED_BUILTIN, OUTPUT);
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(LED_BUILTIN, HIGH);  // turn the LED on (HIGH is the voltage level)
  delay(1000);                      // wait for a second
  digitalWrite(LED_BUILTIN, LOW);   // turn the LED off by making the voltage LOW
  delay(1000);                      // wait for a second
}

It works on a Pico W but when I swap the board to a Pico 2W and changed the board to Pico 2W, upload the same sketch the LED on board does not come on or blink. I've also tried on another Pico 2W with the same result.

earlephilhower, 2024-11-27:
Thanks for checking.  It's one of the teething problems in the Pico2W release.  Replace `LED_BUILTIN` with `64` and it should work.

I'll be updating this line to the right pin...
https://github.com/earlephilhower/arduino-pico/blob/e16c459598b86ce67d8b83ac5b1384dcb8c982ae/variants/rpipico2w/pins_arduino.h#L10

skpang, 2024-11-27:
Ok, changed to 64 and the LED now blinks.
Thanks.

PR #2646 (2024-11-27): Fix LED pin for Pico 2W
Fixes #2644
commit 90d4841b (2024-11-27): Fix LED pin for Pico 2W (#2646)

Fixes #2644
closed 2024-11-27 by skpang