earlephilhower/arduino-pico #2650: SimpleMDNS in combination with WiFiClient leads to panic

GUVWAF, 2024-11-30:
Using a Pico W (RP2040), SimpleMDNS in combination with `WiFiClient` leads to a panic due to an assert in LWIP:

`sys_timeout: timeout != NULL, pool MEMP_SYS_TIMEOUT is empty`

Here’s a MCVE:
```cpp
#include <Arduino.h>
#include <WiFi.h>
#include <SimpleMDNS.h>

void setup() {
  Serial.begin(115200);
  delay(5000);
  WiFi.begin("ssid", "psk");
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
  }
  Serial.println("Starting MDNS service");
  MDNS.begin("test");
  MDNS.addService("http", "tcp", 80);
  
  WiFiClient client;
  Serial.println("Connecting...");
  client.connect("www.google.com", 80);
}

void loop() {
  delay(1000);
  Serial.println("...");
}
```

It happens both for core version 4.2.1 and 4.3.0. When replacing SimpleMDNS with LEAmDNS, this works.

I found a reference to this error here: https://github.com/raspberrypi/pico-sdk/issues/1281. So I bumped `LWIP_DNS` from 1 to 2 in `lwipopts.h` and recompiled `libpico`, and then the MCVE works. However, I'm not sure if this is a suitable fix or we should find out why SimpleMDNS needs more timers than LEAmDNS.

earlephilhower, 2024-11-30:
Thanks for the MCVE and report.  I actually posted in that thread because we had this kind of issue when SDK 1.5.0 came out.  

The [SimpleMDNS/.local PR](https://github.com/earlephilhower/arduino-pico/pull/2582/files)  tried to increase the needed buffers, but evidently it's not enough.  Let me read up and see if there's a more direct way of increasing the timeouts than bumping that DNS value.  Worst case, we can use that.

What's going on is the while LWIP stack uses its own internal scheduling system (MEMP_TIMEOUT) and not any external timers, and space for them all needs to be statically allocated.  LeaMDNS is just another LWIP client and no different than any other connection so it's not using those internal timers directly.

In my testing I used ArduinoOTA (since I didn't have anything else that needed MDNS) but it uses UDP only.  So, no need for a TCP retransmit timeout and there were enough timers.  Here, you have a TCP connection as well as MDNS so we need add'l timeouts...

earlephilhower, 2024-11-30:
In the PR I just bumped up the placeholder we were using to increase timeouts for other reasons (1.5.0 SDK, etc.).  It should be a little lower memory than setting up 2 DNS request buffers (which in the Arduino case can't ever actually use but one) but it's the same idea, increase LWIP's internal `MEMP_NUM_SYS_TIMEOUT` without actually changing any LWIP files...

PR #2653 (2024-11-30): Increase LWIP timeouts to avoid MDNS panics
Bump up the LWIP_ARP setting to increase the LWIP timeout pool to avoid a panic when using SimpleMDNS and TCP clients.

Fixes #2650
commit 6024e9a7 (2024-11-30): Increase LWIP timeouts to avoid MDNS panics (#2653)

Bump up the LWIP_ARP setting to increase the LWIP timeout pool to
avoid a panic when using SimpleMDNS and TCP clients.

Fixes #2650
closed 2024-11-30 by earlephilhower