earlephilhower/arduino-pico #2668: Can watchdog just reboot one processor (e.g. core1) leaving core 0 running

Gavin-Perry, 2024-12-04:
For whatever reason I can't track down, it seems that my core 1 process is quitting/crashing. So I tried adding a watchdog  called for it like this:

void setup1() {
// init some vars
  Serial.printf( "Reset %s", rp2040.getResetReason());
  rp2040.wdt_begin(16000);  // Restart at 8 sec (double decrement issue still there?
}

And then periodically do
rp2040.wdt_reset();
in loop1

I was hoping to get a reason when it restarted, but I guess I don't have it reporting correctly, nothing prints.
And the USB resets (PC beeps on reconnect)

RP2040::resetReason_t rp2040.getResetReason()
[](https://arduino-pico.readthedocs.io/en/latest/rp2040.html#rp2040-resetreason-t-rp2040-getresetreason)
Returns the reason for the last reset, defined in enum RP2040::resetReason_t. 
See example `ResetReason` for some details.

red  " enum RP2040::resetReason_t. "  and `ResetReason` aren't links; where do I find examples?

Also it halts both cores. Core 0 is still working (along with USB stuff) so I'd rather leave it alone
Should I just write my own WD and use 
 rp2040.restartCore1();
to do the restart?
Probably answered my own question but I thought I'd put it out there in case there's a way to do it in SDK that I can't find.

earlephilhower, 2024-12-04:
Sorry but no, the watchdog isn't built that way.  More info is in the [datasheet](https://datasheets.raspberrypi.com/rp2040/rp2040-datasheet.pdf).

You also can't really just reset one core and keep things consistent.  This is a shared memory multiprocessor system and things like initializing `.data` and `.bss` are done only once at power up (i.e. static and global variables).  So even if you somehow got core1 to stop and reset to a known address/state, the global memory would not reset.

closed 2024-12-04 by earlephilhower

Gavin-Perry, 2024-12-04:
Global memory not resetting is fine. Mostly I don't want to lose var values. I'll deal with what needs to re-initialize in the setup1 when it restarts.
Thanks for your help.